aaronwmorris/indi-allsky #936: Cannot Get External Temperature Script Working

kbrown73, 2023-09-09:
Hi,

I cannot seem to get the external temperature script working properly on my RPi3. I created an external temperature script based on one of the examples in the Wiki. Currently it just outputs a static value into a json file that looks like this:

```
{
    "temp": -12.345
}
```

I have set the env var:

`TEMP_JSON=/home/pi/indi-allsky/tmp/temp.json`

I've pointed the config to my script file and made sure it has executable permissions.

Looking at the log I can see it runs the script twice but then for some reason stops running it ever again. It also looks like it's not really trying to use the temperature from the script. It uses 0.0c instead. Not sure whether I'm doing something wrong or could this be a bug?

Here's some of the log file:
```
Sep  9 23:08:01 titan [INFO] Capture-19/MainThread capture.getExternalTemperature() #1041: Running external script for temperature: /home/pi/indi-allsky/temperature.py
Sep  9 23:08:02 titan [INFO] Capture-19/MainThread capture.shoot() #1429: Taking 15.00000000 s exposure (gain 16)
Sep  9 23:08:02 titan [INFO] Capture-19/MainThread libcamera.setCcdExposure() #235: image command: libcamera-still --immediate --nopreview --raw --denoise off --gain 16 --shutter 15000000 --metadata /home/pi/indi-allsky/tmp/tmpirjrr7t5.json --metadata-format json --awbgains 1,1 --output /home/pi/indi-allsky/tmp/tmpy7i9gv48.dng
Sep  9 23:08:02 titan [INFO] Capture-19/MainThread capture.saferun() #414: Total time since last exposure 0.4268 s
Sep  9 23:08:05 titan [INFO] MainProcess/MainThread allsky._queueManualTasks() #1115: Checking for manually submitted tasks
Sep  9 23:08:13 titan [INFO] Capture-19/MainThread capture.saferun() #207: Camera last ready: 11.0s
Sep  9 23:08:13 titan [INFO] Capture-19/MainThread capture.saferun() #208: Exposure state: BUSY
Sep  9 23:08:13 titan [INFO] Capture-19/MainThread capture.detectNight() #1264: Sun altitude: -28:48:16.9
Sep  9 23:08:13 titan [INFO] Capture-19/MainThread capture.detectMoonMode() #1281: Moon altitude: -8:55:44.7, phase 22.3%
Sep  9 23:08:13 titan [INFO] Capture-19/MainThread capture.getExternalTemperature() #1041: Running external script for temperature: /home/pi/indi-allsky/temperature.py
Sep  9 23:08:19 titan [INFO] MainProcess/MainThread allsky._queueManualTasks() #1115: Checking for manually submitted tasks
Sep  9 23:08:20 titan [INFO] Capture-19/MainThread capture.saferun() #326: Exposure received in 18.5564 s (3.5564)
Sep  9 23:08:20 titan [INFO] Capture-19/MainThread capture.shoot() #1429: Taking 15.00000000 s exposure (gain 16)
Sep  9 23:08:20 titan [INFO] Capture-19/MainThread libcamera.setCcdExposure() #235: image command: libcamera-still --immediate --nopreview --raw --denoise off --gain 16 --shutter 15000000 --metadata /home/pi/indi-allsky/tmp/tmpi57lch_i.json --metadata-format json --awbgains 1,1 --output /home/pi/indi-allsky/tmp/tmpuuv4jnwq.dng
Sep  9 23:08:20 titan [INFO] Capture-19/MainThread capture.saferun() #414: Total time since last exposure 18.5564 s
Sep  9 23:08:24 titan [INFO] Capture-19/MainThread capture.saferun() #207: Camera last ready: 3.8s
Sep  9 23:08:24 titan [INFO] Capture-19/MainThread capture.saferun() #208: Exposure state: BUSY
Sep  9 23:08:24 titan [INFO] Capture-19/MainThread capture.detectNight() #1264: Sun altitude: -28:49:12.6
Sep  9 23:08:24 titan [INFO] Capture-19/MainThread capture.detectMoonMode() #1281: Moon altitude: -8:55:02.9, phase 22.3%
Sep  9 23:08:25 titan [INFO] Image-20/MainThread image.add() #1766: Image bits: 16, cfa: BGGR
Sep  9 23:08:25 titan [INFO] Image-20/MainThread image._detectBitDepth() #1848: Image max value: 4095
Sep  9 23:08:25 titan [WARNING] Image-20/MainThread image._detectBitDepth() #1887: Updated default bit depth: 12
Sep  9 23:08:25 titan [INFO] Image-20/MainThread image._calibrate() #1927: Searching for bad pixel map: gain 16, exposure >= 15.0, temp >= 0.0c
Sep  9 23:08:25 titan [WARNING] Image-20/MainThread image._calibrate() #1944: Temperature matched bad pixel map not found: 0.00c
Sep  9 23:08:25 titan [WARNING] Image-20/MainThread image._calibrate() #1962: Bad Pixel Map not found: ccd1 16bit 15.0000000s gain 16 bin 1 0.00c
Sep  9 23:08:25 titan [INFO] Image-20/MainThread image._calibrate() #1974: Searching for dark frame: gain 16, exposure >= 15.0, temp >= 0.0c
Sep  9 23:08:25 titan [WARNING] Image-20/MainThread image._calibrate() #1991: Temperature matched dark not found: 0.00c
Sep  9 23:08:25 titan [WARNING] Image-20/MainThread image._calibrate() #2009: Dark not found: ccd1 16bit 15.0000000s gain 16 bin 1 0.00c
```

aaronwmorris, 2023-09-10:
Can you post the contents of the script here?

The `TEMP_JSON` variable is set by indi-allsky when executing the script and is a random file name every time.

kbrown73, 2023-09-10:
Ah ok. I've now removed my manually defined TEMP_JSON variable and that did have a minor effect. It still only seems to evaluate my script just twice but from the log I can see it actually did get the value and tried to use it with darks and bad pixel maps etc:

```
Sep 10 02:42:33 titan [INFO] Capture-2/MainThread capture.getExternalTemperature() #1041: Running external script for temperature: /home/pi/indi-allsky/temperature.py
...
...
...
Sep 10 02:42:44 titan [INFO] Image-2/MainThread image._calibrate() #1927: Searching for bad pixel map: gain 16, exposure >= 15.0, temp >= -12.3c   <=== AS EXPECTED!
```

The script is pretty much exactly as one of the examples in the Wiki.
```
#!/usr/bin/env python3

# Example of an external temperature script for indi-allsky
# STDOUT and STDERR are ignored
#
# The json output file is set in the environment variable TEMP_JSON

import os
import sys
import io
import json
import logging

logging.basicConfig(level=logging.INFO)
logger = logging

temp_c = -12.345
try:
    # data file is communicated via environment variable
    temp_json = os.environ['TEMP_JSON']
except KeyError:
    logger.error('TEMP_JSON environment variable is not defined')
    sys.exit(1)

# dict to be used for json data
temp_data = {
    'temp' : temp_c,
}

# write json data
with io.open(temp_json, 'w') as f_temp_json:
    json.dump(temp_data, f_temp_json, indent=4)

# script must exist with exit code 0 for success
sys.exit(0)
```

Not sure why it would just evaluate it twice and then never again? On my image labels I never see anything else other than 0.0C when it should be -12.3C

aaronwmorris, 2023-09-10:
I see the issue.  The code only calls the external temperature script if the camera does not return a valid temperature.  libcamera must not be returning a valid temperature until a couple of exposures in.

I will get this fixed.

aaronwmorris, 2023-09-10:
Merged #939 .  You may have to re-run setup.sh because I just released an unrelated update.

kbrown73, 2023-09-10:
Brilliant. It works now. Thank you for the speedy fix!

closed 2023-09-10 by kbrown73