aaronwmorris/indi-allsky #950: Have image processing algiorithms changed recently?

vastmeridian, 2023-09-18:
In updating to #937 I have noticed that the images seem to be more 'processed' as the two shots below indicate.  Maybe I missed an announcement.

This image was typical of the pre #937 build
[image]

Whereas this image has become more prominent since upgrading to #937.
[image]

Have the image processing algorithms changed recently?  It looks like I will need to re-think my settings.

aaronwmorris, 2023-09-19:
I did change the default Field of View used for the SQM calculations and processing.  I changed the default to the central 50% of the image where it used to be 66%.  If you change the SQM FoV to 66% on the Processing Tab, it should function like it used to for you.

Another workaround is creating a detection mask, which will use the mask for the borders of the calculations.

This most heavily affects the stretching algorithm.

vastmeridian, 2023-09-25:
Following on from this and my post on the AllSky group on Facebook, I'll  try the new release and report my findings here.  The 'false colouring' does seem related to SCNR as disabling this does stop the colouring.  I will also try setting the colour saturation setting back to 1.0, as 1.5 does seem to over-colour my stars.

[image]

aaronwmorris, 2023-09-25:
Those colors actually look very good.

I have learned quite a bit more about numpy since originally implementing the SCNR.  I now know how to deal with the integer overflows.

vastmeridian, 2023-09-25:
I've updated the software following your note and I'll see how things go, so thank you as always for the quick response.

To me the colours look too magenta, if you follow.  I almost want to rebalance the colours.  Maybe I will do that once things have settled down with the processing.

vastmeridian, 2023-09-26:
Well, I'm still getting very odd colours...

I've lowered the gains in case I'm getting a lot of large-value pixels and I've got CLAHE and SCNR disabled; stretching on, but not excessively.  My settings are:

Night Gain | 1800
Moon Mode Gain | 1400
CLAHE Clip Limit | off
CLAHE Grid Size | off
16-bit Contrast   Enhance | off
Enable Stretching | on
Stretching Gamma | 1.5
Stretching Std   Deviations | 1.0
Moon Mode   Stretching | on
Daytime Stretching | off
Stretching split   screen | Off
Bayer Pattern | RGGB
SCNR (green   reduction) |  
Red Balance Factor | 0.6
Green Balance   Factor | 0.8
Blue Balance   Factor | 1.0
Auto White Balance | on
Saturation Factor | 1.5

[image]

The histogram is
[image]

Is there a setting I can use to get the raw out-of-camera FITS file so that I can see what the pixels are before processing?

aaronwmorris, 2023-09-26:
The histogram may be a bit deceiving, too.  I just remembered that the histogram is generated from a crop of the center of the image (trying to exclude things like trees on the edges).

vastmeridian, 2023-10-12:
I'm still bemused by the incredibly strong colouring I am getting.  I've reduced the gain of the camera; turned off CLAHE and have no colour saturation enabled - and I am still getting very strong colours in my images which are not present in the FITS versions of the images.

Is there a way of setting all processing settings to their defaults in case I've missed a setting?  I can then start my processing set up from scratch again.

aaronwmorris, 2023-10-12:
There is a new feature in the latest release where you can test processing settings.  If you have a FITS version saved of an image (in the database), you can select `FITS Processing` in the image viewer (at the bottom) to bring up a page where you can test different processing settings.  That might help you track it down.

vastmeridian, 2023-12-29:
I think this can be shut down as an issue now.

Thanks for your help

closed 2023-12-29 by vastmeridian