aaronwmorris/indi-allsky #952: Resolution resets when using V4L2 camera

dlemmink, 2023-09-19:
Using an SVBONY SV205 camera with limited success.  Found that running INDI-SERVER independently from the program seems to be more stable.  Using the included INDI-SERVER, the camera times out after a few images.  Starting the newly installed INDI-SERVER and then manually starting INDI-ALLSKY seems to be stable.

Unfortunately, every time INDI-ALLSKY starts, it returns the resolution back to 640x480 even though the resolution is saved in the INDISERVER Profile.

Is there a way to expose resolution as an option in the INDI-ALLSKY parameters to correct this?

aaronwmorris, 2023-09-20:
I am not sure of the exact setting you need, but it is likely related to one of these two:

https://github.com/aaronwmorris/indi-allsky/wiki/INDI-custom-config#webcam-resolution

dlemmink, 2023-09-20:
Thanks....This did the trick....just need to get indiserver stable now...

{
    "PROPERTIES": {},
    "SWITCHES": {
        "V4L2_SIZE_DISCRETE": {
            "on": [
                "3264x2448"
            ]
        }
    }
}

aaronwmorris, 2023-10-31:
Closing this for now.  Let me know if I can help.

closed 2023-10-31 by aaronwmorris