aaronwmorris/indi-allsky #955: ValueError: operands could not be broadcast together with shapes

StarGeezerPhil, 2023-09-26:
Hi @aaronwmorris 

Sorry to bug you again. Experiencing a new error having moved the camera and now trying to get the image rotated and cropped to tidy. Initially I thought it had to do with the logo mask I'd applied, so have disabled that. Then thought it might be the darks/calibration frames, so disabled them as well, but still getting this error - meaning all images fail.

Up until night mode kicked in everything was working with the rotation, crop and logo mask applied. Any ideas what's causing this?

```
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception: 
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception: ValueError: operands could not be broadcast together with shapes (3930,4670,3) (3040,4056,3) 
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:     masked_img = (original_img * self._sqm_gradient_mask).astype(numpy.uint8)
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/detectLines.py", line 46, in detectLines
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:     i_ref['lines'] = self._lineDetect.detectLines(self.image)
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 2421, in detectLines
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:     self.image_processor.detectLines()
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 536, in processImage
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:     self.processImage(i_dict)
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 284, in saferun
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:     self.saferun()
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 255, in run
Sep 26 19:40:35 SkyCam [ERROR] MainProcess/MainThread allsky._startImageWorker() #350: Image worker exception: Traceback (most recent call last):
Sep 26 19:40:28 SkyCam [INFO] Capture-1/MainThread indi.getTelescopeRaDec() #842: Telescope Coord: RA 18.93, Dec 53.51
```

StarGeezerPhil, 2023-09-26:
UPDATE: I've disabled rotation (by arbitrary angle) - and I'm back to getting captures, so suspect the rotation is the cause.

aaronwmorris, 2023-09-26:
If you have a detection mask defined, it would not match the dimensions of the rotated image.

StarGeezerPhil, 2023-09-26:
Thanks @aaronwmorris - sorry, yes probably - I'd thought detection etc was all done before any of the final tidying...

But yes - in your image order of operations Detect stars and Line detection are #14 and #15, so later in the order than rotate at #11. 

I don't know if you're open to it (or if this would be a nightmare in the code), but the following might make more sense - so that all calibration and detection mask frames are always from the reference/as-delivered image, essentially moving any flip/crop/rotate to after any meaningful detection (ADU, stars, lines, etc.).

1. Dark frame calibration
2. Save FITS
3. Calculate jSQM
4. Stack
5. Debayer
6. Save RAW (non-stacked)
7. Calculate ADU
8. Stretch (16-bit)
9. 16-bit Contrast enhance (CLAHE)
10. Downsample to 8-bit
11. Detect stars
12. Line detection

[all detection done by this point]

14. White balance
15. SCNR
16. Manual WB
17. Auto WB
18. Saturation
19. Contrast enhance (CLAHE)
20. Colorize (if grayscale)
21. Rotate (90 degree or arbitrary) 
22. Flip vertical
23. Flip horizontal
24. Crop
26. Image Circle Mask
27. Logo overlay
28. Scale image
29. Write text
30. Save image

Actually, come to think of it - wouldn't it make sense to move 11 (Detect Stars) and 12 (Line Detection) to after 19 (Contract enhance) as well - as you're likely to get better pickup?

So in the order of operations you essentially have:
1. Acquisition Settings
2. Image colour and contrast processing
3. Detection of stars and meteors
4. Output settings

StarGeezerPhil, 2023-09-27:
I understand the logic, but the problem is it's really confounded - and I'm only just realising - if I'm flipping my image, I need to do that and create the _detection_ mask flipped, **but not cropped or rotated!** however, for my logo mask I need to _flip, rotate and crop it_. (I should have doubled checked, but had assumed it was the original (unprocessed) image - so all my detection masks have not been flipped.       ...I _had_ also thought this was used to denote the area used for the ADU calculation).

I'd save a copy of the original image every X hundred captures as camera_output_reference.jpg or similar and make this available alongside the relevant settings in the config (e.g. something like Detection Mask | "Mask to set the area used for meteor and star detection - black areas are excluded, white areas used for detection. Must be PNG file format. See {camera output file} - the mask must match this image as output by your camera")

Similar for the logo mask (Logo Mask | "Optional overlay applied to final image to mask out areas for privacy or add a logo/other content. Must use an alpha transparency layer and match output image dimensions - {final output image example link}")

aaronwmorris, 2023-09-27:
I think you have convinced me.  I was just thinking I should probably write an external utility to export the base image for the mask, unflipped/uncropped/unrotated/etc.

Since this would be a breaking change for people using an existing mask, I will probably deploy this as part of the next major release.  I am in the middle of moving, as well, so it may be 1-2 months before I can really focus on this.

aaronwmorris, 2023-10-25:
I just merged #1001 to change the order of operations regarding the mask.

Masks are now processed before any geometry changes, so changing the image orientation will no longer need a new mask.  There is also a helper view that generates a base image that may be used for the detection masks.

aaronwmorris, 2023-12-08:
Closing this issue for now.  Let me know if you find any problems!

closed 2023-12-08 by aaronwmorris