aaronwmorris/indi-allsky #960: SQM data to InfluxDB

kbrown73, 2023-09-29:
Hi,

So this is not really an issue with indi-allsky but a request for help to understand something. I was hacking into the sqm.py in order to get the calculated sqm value inserted into an external time series database (InfluxDB) I have running on my NAS. This requires importing the [influxdb_client](https://pypi.org/project/influx-client/) library which I've installed using pip. However as indi-allsky is running in a virtual env, it is not able to import that library. Here's where my understanding ends hence these two (probably naive) questions:

1. Is there a way to get indi-allsky to see my pip installed influx_client library?
2. If not, can I somehow add this library to the virtual environment?

I know this is all very hacky but any help would be appreciated.

aaronwmorris, 2023-09-30:
In order for indi-allsky to see the module it just needs to be installed while the virtualenv is acttive

```
source virtualenv/indi-allsky/bin/activate

pip install influxdb_client
```

I have an example file in `examples/telegraf/indi_file.conf` that configures telegraf to import JSON data generated by indi-allsky into influxdb.  There is also a Grafana dashboard template in the folder.

kbrown73, 2023-09-30:
Brilliant as always! Thank you for pointing me to the right direction. I have Grafana running on my NAS as well so I'll definitely check out your dashboard template!

closed 2023-09-30 by kbrown73