aaronwmorris/indi-allsky #963: Backup DB prior to migration failing

tonycolo, 2023-10-01:
Following the Updating instructions, I am upgrading via setup.sh to the latest via git pull origin main and encounter the error below, regardless of whether I choose to keep the same indi-server or not.

```
**** Backup DB prior to migration ****
Error: While importing 'app', an ImportError was raised:

Traceback (most recent call last):
  File "/home/tonyb/sw/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 219, in locate_app
    __import__(module_name)
  File "/home/tonyb/sw/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/tonyb/sw/indi-allsky/indi_allsky/flask/__init__.py", line 9, in <module>
    from flask_wtf.csrf import CSRFProtect
  File "/home/tonyb/sw/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/__init__.py", line 4, in <module>
    from .recaptcha import Recaptcha
  File "/home/tonyb/sw/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/__init__.py", line 1, in <module>
    from .fields import RecaptchaField
  File "/home/tonyb/sw/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/fields.py", line 3, in <module>
    from . import widgets
  File "/home/tonyb/sw/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/widgets.py", line 3, in <module>
    from werkzeug.urls import url_encode
ImportError: cannot import name 'url_encode' from 'werkzeug.urls' (/home/tonyb/sw/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/werkzeug/urls.py)

Usage: flask [OPTIONS] COMMAND [ARGS]...
Try 'flask --help' for help.
```

After sourcing the virtualenv to indi-allsky's python, I tried reverting Flask to 2.3.3 (it was updated to 3.0.0 yesterday) but that didn't change anything.

Running on Raspberry Pi 4
Linux astro 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

Thanks for any suggestions.

aaronwmorris, 2023-10-01:
The problem appears to be in Flask-login interacting with the werkzeug module.

Try downgrading flask and werkzeug

```
pip install Flask==2.3.3 werkzeug==2.3.7
```

aaronwmorris, 2023-10-01:
I am pretty sure this is related:  https://github.com/maxcountryman/flask-login/issues/744

tonycolo, 2023-10-02:
Downgrading both Flash and werkzeug to those versions worked, thanks!

aaronwmorris, 2023-10-31:
Flask-Login 0.6.3 has been released and fixes the compatibility issues with the newer Flask and werkzeug releases.  #1009 

Closing this for now.

closed 2023-10-31 by aaronwmorris