aaronwmorris/indi-allsky #975: [Feature Request/Enhancement]: Support for Google Cloud Platform

paulstey, 2023-10-05:
First, thank you so much for all the work that has gone into indi-allsky! This is an enormously valuable tool!

Would it be possible to add support for Google Cloud Platform (GCP) storage buckets in addition to AWS? Your readme file suggested that this should in principle be possible.

Again, thank you so much for your work on this!!

aaronwmorris, 2023-10-06:
I started on GCP object storage a while back and then just kind of dropped it after trying to figure out the authentication part.  I know it has a compatibility option with AWS object storage, but the normal authentication is not just a simple username/password type.  I just did not spend too much time on it since I was not sure how much interest there would be.

I will spend some time on it.

gaitskell, 2023-10-17:
Appreciate you working on this. We have the same interest as our image handling is on Google cloud (Colab) and so hosting images direct to the Google storage would be much appreciated. If you need any help with debugging let me know.

aaronwmorris, 2023-10-19:
Merged #999 to add GCP storage support

aaronwmorris, 2023-12-08:
Closing this for now.

closed 2023-12-08 by aaronwmorris