aaronwmorris/indi-allsky #980: Feature fisheye to panoramic

bluthen, 2023-10-11:
What are the thoughts of adding fisheye to panoramic option? It would generate a 2nd image to possibly probably after masking.

I wrote some code to do a simple transform that you are free to take for this project. I know the transform is only a small part of what would be needed to integrate into here.

https://github.com/bluthen/fish2pano

I needed I might be able to help integrate it with some guidance.

aaronwmorris, 2023-10-11:
I have read about doing similar perspective transforms with OpenCV.  Sounds interesting, I will check it out.

aaronwmorris, 2023-11-02:
I just merged #1011 with the panoramic processing.  At the moment, I am just modifying the final image, but I will probably create a second image like you originally mentioned.  I will probably add the second image soon.

aaronwmorris, 2023-11-07:
I finally got a chance to test on Raspi hardware and this is a bit of a beast.  Not completely terrible, but a 3000 pixel image circle scaled to 0.5 takes 25 seconds to process.  The process is much faster on older mid-range Intel x86_64.  This is obviously not something that could reasonably be expected to be applied to every image.  I may have to add an option to "perform operation every # images".

HxPI, 2023-12-16:
This sounds like an awesome feature. Looking forward to trying it out!

aaronwmorris, 2023-12-16:
I have not quite settled on how to handle the panorama images.  

Right now, it just creates and overwrites the same file.  This algorithm is too slow to realistically run with every image (at high resolution anyway).  But I do think it would be interesting to have a timelapse generated from a series of panoramas.

aaronwmorris, 2024-01-19:
With #1101 panoramas are now a fully supported asset type in indi-allsky with dedicated views and support in file transfer, S3 Object Storage, and SyncAPI.

StarGeezerPhil, 2024-01-26:
Love this feature @aaronwmorris 👍

Would it be possible to stitch the panoramic images into a timelapse video at the end of the period?

Complete curveball... Do you know whether it's possible to output the images and video as some form of 180-degree VR experiences (being able to look around with a headset or drag mouse/finger-type thing? I'm really not sure what's involved in that.

aaronwmorris, 2024-01-26:
I will look into it, but I will not make any promises.  :-)

My next thing is to create a timelapse for the panoramas.

bluthen, 2024-01-28:
> This algorithm is too slow to realistically run with every image (at high resolution anyway)

I can probably write a version that uses scipy.weave.inline (if including those deps is acceptable), or a standalone C++ executable. I think it would speed it up >10X. I've done both before. Python for loops are slow. Another alternative is try to find some  opencv optimized method or some type of matrix method, but I'm not sure how to do those at the moment.

aaronwmorris, 2024-01-28:
If you have the time, that would be great, but it is entirely up to you.  I also looked at the code and I did not see an obvious way of vectorizing the operation.

aaronwmorris, 2024-01-28:
Merged #1117 with support for panorama timelapses.

I would recommend upgrading to this if you use panoramas because I missed a few things with my previous changes like expiring/cleaning up old panorama images.  This is code I have not had to touch in a very long time.

bluthen, 2024-01-29:
> If you have the time, that would be great, but it is entirely up to you. I also looked at the code and I did not see an obvious way of vectorizing the operation.

I had forgot scipy.weave got removed, so I update my fish2pano repo to have a cython version with parallism.

On my machine some of my tests are over 400 times faster. My changes are in my fish2pano repo, but if you would like me to make a PR to indi-allsky let me know and I can work on that.

https://github.com/bluthen/fish2pano

Just to test the loop, if I use scale 3 to simular a really large pano:

```
$ python fish2pano.py large2.jpg 481 618,538 3 large2_pano.jpg
Loop 16.27480387687683
WARNING: Non-optimized generate_pano
```

```
$ python fish2pano.py large2.jpg 481 618,538 3 large2_pano.jpg
loop 0.04010009765625
```

16.274/0.0401 = 405.8

But even at 0.5 scale it is 288 times faster.

Let me know how you want to move forward, or if you have questions.

aaronwmorris, 2024-01-29:
Very cool.  I have not fully tested myself, but I realized a few things...

indi-allsky currently runs on systems all the way back to Debian 10 meaning Python 3.7 and some older versions of the python modules.

I would not cry if Python 3.7 did not work, but at a minimum Python 3.9 for Debian 11 support would be needed.  I also use opencv-python-headless instead of the full opencv package, just to reduce the module requirements.  I can probably take care of this with my own fork.

Do you know of any show stoppers for older python versions or modules, or were the requirements in pyproject.toml just what you had on your system?

bluthen, 2024-01-29:
> Do you know of any show stoppers for older python versions or modules, or were the requirements in pyproject.toml just what you had on your system?

Yes, it is just what I was working with on my system. I don't think you should have issues with it on older pythons or opencv. I think cython requires only Python 3.7+ from what I can tell from cython's source code.

aaronwmorris, 2024-01-30:
That is certainly impressive.

I forked the code and made the requirements more generic.  Nothing major.

https://github.com/aaronwmorris/indi-allsky

I guess to really make this useful it would require everything to be built and installed as a module.  I will admit I am a bit out of my element.

Would it be better if I opened this as an issue under the repo?

bluthen, 2024-01-30:
Ahh yeah, sorry this wasn't more polished or easier.

I can work on this in my repo. I'll do the following:

- More Generic/Reduce requirements
- Make it a library and remove the non-lib deps (scipy,opencv - only numpy as runtime dep, I think cython can be build dep only)
- Move find_circle and loading a image into another repo: fish2panogui or something.
- Use setuptools
- Get it in pypi
- Try using cibuildwheel to make multi-platform/multi-python binaries https://cibuildwheel.readthedocs.io/en/stable/ Targeting Python3.7+

I'll let you know when it is done.
Then it can be added just as a dep and be installed like other modules.

bluthen, 2024-02-01:
Ok, proper module made with wheels for >Py3.7 [aarch64, i686, x86_64, pi_wheels-rasp pi arm7] (for linux at least).
https://pypi.org/project/fish2pano/
https://www.piwheels.org/project/fish2pano/

`pip3 install fish2pano`

aaronwmorris, 2024-02-23:
I finally had some time to integrate the new module.  Generating panoramas basically has zero performance impact at all resolutions and scales.  This is some really fantastic work @bluthen .

Merged #1141

HxPI, 2024-02-23:
When enabling fisheye to panorama, the normal image capture stops working. I still get raw images saved but jpgs and latest view stops working.

aaronwmorris, 2024-02-23:
Make sure you re-run setup.sh.  Nothing really changed with the database with this release, but I have made some database schema changes recently.

aaronwmorris, 2024-02-23:
You should be able to attach the log here.

HxPI, 2024-02-23:
Here is the log:

[indi-allsky1.log](https://github.com/aaronwmorris/indi-allsky/files/14380931/indi-allsky1.log)

aaronwmorris, 2024-02-23:
I have made a minor change with #1142 that might help.  I do not fully understand the exception you are seeing, unless it is some issue with Python 3.7.

There are some other concerning things, like it should show indi-allsky version 8.0 instead of 7.8.  I am not sure why that is happening too.

HxPI, 2024-02-23:
I’ll update and test it out. Are there any other logs you’d like to see?

HxPI, 2024-02-23:
Still same issues

aaronwmorris, 2024-02-23:
Let me test this out tomorrow.

Can you post the output of the `./misc/support_info.sh` script?

HxPI, 2024-02-23:
Here is the support info script output:

[support_info_output.txt](https://github.com/aaronwmorris/indi-allsky/files/14381383/support_info_output.txt)

aaronwmorris, 2024-02-26:
@bluthen has integrated the change and released the update.  Pull down the latest version of indi-allsky and it the panoramas should be working with Python 3.7 now.  #1148

HxPI, 2024-02-26:
Getting a WARNING when collecting gunicorn "WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response')): /simple/gunicorn/gunicorn-21.2.0-py3-none-any.whl

aaronwmorris, 2024-02-26:
This happens occasionally when trying to download python wheels and the mirror you are connected to is overloaded or down.  It is usually temporary.  You may try adjusting your DNS server in `/etc/resolv.conf` and trying again (to see if you get a different mirror).

HxPI, 2024-02-26:
Does it matter? The system appears to be running as expected, histogram and panoramas are working now.

HxPI, 2024-02-26:
Can panoramas be output to TIFF or one of the other raw formats?

HxPI, 2024-02-26:
DNS server says 127.0.0.1 in resolving.conf. Should it say something like 8.8.8.8 instead?

HxPI, 2024-02-26:
Elapsed average 6, Processing range from 4 to 106!

aaronwmorris, 2024-02-26:
What do your temperatures look like (under System)?  If you are above 80, the Pi will start to throttle and can cause severe slowdowns.

You can also check the logs to see how long the panoramas (and other items) take to process.  `/var/log/indi-allsky/indi-allsky.log`

HxPI, 2024-02-26:
Temp is 129.3°F. See image of lag times

[image]

HxPI, 2024-02-26:
Is there any way to reduce the lag time?

aaronwmorris, 2024-02-26:
I forgot to say Celsius in my message, your temps are fine.  The lag and processing times are just measurements, what you see is what happened.

Your processing times are certainly out of whack.  I would like to see a few hundred lines of your logs to see what is happening in the background.

HxPI, 2024-02-26:
Yeah image processing is really bogged down, some taking over a minute to process. Although I don't understand why the web is showing images over 15min behind what is saved on the thumb drive?

HxPI, 2024-02-26:
See attached log

[indi-allsky2.log](https://github.com/aaronwmorris/indi-allsky/files/14412154/indi-allsky2.log)

HxPI, 2024-02-26:
The drift is so bad the web preview page no longer shows an image after a while, just says no image after 15min.

HxPI, 2024-02-27:
I ran setup.sh again and now the lag seems normal, elapsed around 3 and processing between 7 and 10 with a few outliers. Images are keeping up in the web viewer. I'll leave it at this for now. Thanks for sharing and continue the great development.

aaronwmorris, 2024-02-27:
I did not really see anything in the log that would cause the processing times to be that inconsistent.  Hopefully, the re-run of setup.sh fixed whatever was wrong.

As for how things can get so far behind... This is kind of a key design difference from TJ's allsky and indi-allsky.  The capture process and image processing are performed independently from each other.  The capture process loop only captures images.  Once an image is captured, it is sent to the image processing queue and the next image is captured.  The Image processing handles each image as it comes in, so if the processing slows down, the image queue falls behind, but image capture never stops.

I suppose I probably need to add some settings to pause the capture loop if the queue grows too large.

HxPI, 2024-02-27:
A watchdog and alert for excessive lag delta would be useful. Thanks for the follow-up.

aaronwmorris, 2024-02-27:
As of #1157 there is now an algorithm to introduce additional delays in the capture sequence when the image queue exceeds a certain threshold.

HxPI, 2024-03-02:
Updated to latest, no issues encountered.

aaronwmorris, 2024-04-28:
I think the Fisheye to Panorama functionality is now stable.  Closing this issue for now.