aaronwmorris/indi-allsky #981: Feature request - cardinal directions

bluthen, 2023-10-11:
What are the thoughts of overlaying cardinal directions on the image. Like a letter (N, S, E, W).

Options that would be needed with this feature are:

  * rotation angle
  * vertical flip
  * color?

If integration with platesolving was also added, it could find the angle or flip automatically.

aaronwmorris, 2023-10-11:
I have been considering this for a while, ie adding a compass rose overlay.  I would probably need two versions, one with E/W reversed since the letters will not flip correctly.

caphector, 2023-10-18:
Along similar lines I had been thinking it might be fun to have a wind rose since I could feed in wind direction from my anemoscope.

aaronwmorris, 2023-11-13:
Merged #1018 to add cardinal directions to the image.  The rotation of the cardinal directions is based on the configured azimuth of the camera/lens.

The cardinal directions are also swapped based on the horizontal/vertical flip settings of the image.  Finally, there are also manual swap settings.

gigaklatch, 2023-11-19:
Not sure if this is the place to mention this issue or not, but here goes. For me, there is the issue that once cardinal direction labels have been enabled using the web interface it is virtually impossible to disable them.  Reloading the "Image" tab in the config after trying to disable them shows that the button has switched back to enabled. Meanwhile, the letters remain in the images through all of this. Perhaps I am alone in having this issue?

aaronwmorris, 2023-11-20:
Once you check in, you cannot checkout.  Doh.  Merged #1031 to fix being unable to disable cardinal directions.

FYI, I am working on more fixes for the cardinal directions.  Adding the ability to offset the image circle, broke some of the calculations for the direction placement.

aaronwmorris, 2023-11-20:
I have also merged #1032 to fix the problems when there is an offset of the image circle.  There is also a new setting to outline the image circle for testing/debugging.

aaronwmorris, 2024-01-05:
I think this feature is effectively complete, so I am going to close this for now.

closed 2024-01-05 by aaronwmorris