aaronwmorris/indi-allsky #983: Failure during fresh install / Ubuntu 22.04.3 (PyIndi-Client / SWIG)

SubBass100, 2023-10-12:
Hello, I've been trying to install Indi-Allsky for a few days now and it seems to be getting hung up in the same place every time.

System specifications: 4core 6Gb ram Ubuntu 22.04.3

I install Ubuntu server 22.04.3 Fresh
Apt update / Apt upgrade
I follow the directions listed on the site to clone the repo and ./setup.sh
Select 1 (for Indi)
US > NY
then I let it do it's thing for about 10-15 minutes and it fails with this error:

`Collecting pyindi-client
  Downloading pyindi-client-1.9.1.tar.gz (18 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: pyindi-client
  Building wheel for pyindi-client (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]
      /home/sysadmin/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/setuptools/dist.py:314: InformationOnly: Normalizing 'v1.9.1' to '1.9.1'
        self.metadata.version = self._normalize_version(
      running bdist_wheel
      running build
      running build_ext
      building '_PyIndi' extension
      swigging indiclientpython.i to indiclientpython_wrap.cpp
      swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
      Language subdirectory: python
      Search paths:
         ./
         /usr/include/
         /usr/include/libindi/
         /usr/local/include/libindi/
         ./swig_lib/python/
         /usr/share/swig4.0/python/
         ./swig_lib/
         /usr/share/swig4.0/
      Preprocessing...
      Starting language-specific parse...
      /usr/include/libindi/indibase.h:10: Error: Syntax error in input(1).
      error: command '/usr/bin/swig' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyindi-client
  Running setup.py clean for pyindi-client
Failed to build pyindi-client
ERROR: Could not build wheels for pyindi-client, which is required to install pyproject.toml-based projects
`

So I monkeyed with trying to manually install pyindi-client manually which also breaks. Then reinstall Ubuntu again for a fresh start because I feel I'm doing something wrong. I noticed on the Main Git Repo there is a table that reads:

**Ubuntu 22.04 | x86_64 | Indi installed from ppa:mutlaqja/ppa**

So on the fresh install of Ubuntu I add the repo
sudo add-apt-repository ppa:mutlaqja/ppa
sudo apt update
sudo apt install indi-full gsc

Then I run ./setup.sh
It fails in the same place.

I'm very confused why it is failing. OS is fresh and updated. repo is added. I believe I am following the directions correctly. I've reinstalled Ubuntu quite a few times now and tried everything I can think of and now I'm here. Is anyone else having this issue? Did I overlook one simple instruction?   I really want to use this software but I'm having one heck of a time getting it to install properly. Thank you in advance.

bluthen, 2023-10-12:
Three days ago this was pushed to add indi 2.0.4 support:
https://github.com/aaronwmorris/indi-allsky/pull/978

Have you tried with a new pull from less than 3 days ago?

SubBass100, 2023-10-12:
Yes, I've tried on two fresh installs today, one about 45 minutes ago. I can provide any logging you need. It's still on the screen

aaronwmorris, 2023-10-12:
There is a typo in one of the INDI headers causing this problem.  The INDI devs have fixed the problem in their repo, but I do not think the fix will be distributed until 2.0.5

https://github.com/indilib/indi/pull/1948

You just need to edit /usr/include/libindi/indibase.h (or /usr/local/include) and remove the space between % and "include".

SubBass100, 2023-10-12:
🙏 Thanks so much that got me past with no error. I feel I should have asked sooner but wanted to exhaust me being the problem first. I did notice as the setup continued and it asked me for what camera to select, I would normally select "indi_webcam_ccd" so that I can use my IP camera. however I am not seeing it now, has it's name changed?

aaronwmorris, 2023-10-12:
As far as I can tell, the name has not changed.  It should be part of the `indi-webcam` package in the PPA repo.  Is it possible it is just at the bottom of the list?  You may have to scroll down to see it.

SubBass100, 2023-10-12:
[This is the list that I see](https://imgur.com/a/chn1GtW). It might have something to do with the diagnostics I was doing in attempt to solve the last problem. I will try to blow everything away again and install using the new fix and see if it appears!

aaronwmorris, 2023-10-12:
You may just try installing the package manually.

```
sudo apt-get install indi-webcam
```

SubBass100, 2023-10-12:
Perfect! for completeness, I reinstalled Ubuntu, went through the apt update/upgrade, git cloned indi-allsky and ran ./setup.sh

As it was installing I waited until /usr/include/libindi/indibase.h was created (just used cat /usr/include/libindi/indibase.h repeatedly until it appeared and when it did I used sudo nano /usr/include/libindi/indibase.h and removed the space after the % before the include. The setup installed that part fine and then asked about what camera drivers to use. indi_webcam_ccd was available and I selected it. No errors occurred. I hope this helps someone. Thank you for your help!

closed 2023-10-12 by SubBass100