aaronwmorris/indi-allsky #990: Problems with Svbony SV305 camera

mattyryu, 2023-10-17:
Hi there- I am having some trouble trying to get an Svbony SV305 camera running with a new installation.

- Raspberry Pi 4 4GB
- Raspberry PI OS Bookworm 64-bit (also tried Bullseye 32/64 and Ubuntu 22.04)
- Svbony SV305 Camera (no gps)

After the installation completes successfully and everything is running I see the following in /var/log/syslog:
```
2023-10-17T11:08:34.373945-04:00 allsky kernel: [   57.298062] usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
2023-10-17T11:08:34.815858-04:00 allsky python3[902]: Dispatch command error(-1): INDI: Could not find property CCD_CAPTURE_FORMAT in SVBONY CCD SV305
2023-10-17T11:08:34.816319-04:00 allsky python3[902]: <setSwitchVector device="SVBONY CCD SV305" name="CCD_CAPTURE_FORMAT" state="Idle" timeout="60" timestamp="2023-10-17T15:08:34">
2023-10-17T11:08:34.816620-04:00 allsky python3[902]:     <oneSwitch name="SVB_IMG_RAW8">
2023-10-17T11:08:34.816863-04:00 allsky python3[902]: Off
2023-10-17T11:08:34.817092-04:00 allsky python3[902]:     </oneSwitch>
2023-10-17T11:08:34.817313-04:00 allsky python3[902]:     <oneSwitch name="SVB_IMG_RAW16">
2023-10-17T11:08:34.817526-04:00 allsky python3[902]: On
2023-10-17T11:08:34.817742-04:00 allsky python3[902]:     </oneSwitch>
2023-10-17T11:08:34.817980-04:00 allsky python3[902]:     <oneSwitch name="SVB_IMG_Y8">
2023-10-17T11:08:34.818191-04:00 allsky python3[902]: Off
2023-10-17T11:08:34.818386-04:00 allsky python3[902]:     </oneSwitch>
2023-10-17T11:08:34.818647-04:00 allsky python3[902]:     <oneSwitch name="SVB_IMG_RGB24">
2023-10-17T11:08:34.818890-04:00 allsky python3[902]: Off
2023-10-17T11:08:34.819093-04:00 allsky python3[902]:     </oneSwitch>
2023-10-17T11:08:34.819288-04:00 allsky python3[902]: </setSwitchVector>
2023-10-17T11:08:49.747461-04:00 allsky python3[902]: Process Capture-1:
2023-10-17T11:08:49.752199-04:00 allsky python3[902]: Traceback (most recent call last):
2023-10-17T11:08:49.754847-04:00 allsky python3[902]:   File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
2023-10-17T11:08:49.755441-04:00 allsky python3[902]:     self.run()
2023-10-17T11:08:49.755927-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/capture.py", line 180, in run
2023-10-17T11:08:49.756381-04:00 allsky python3[902]:     raise e
2023-10-17T11:08:49.756937-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/capture.py", line 176, in run
2023-10-17T11:08:49.757416-04:00 allsky python3[902]:     self.saferun()
2023-10-17T11:08:49.757849-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/capture.py", line 188, in saferun
2023-10-17T11:08:49.758380-04:00 allsky python3[902]:     self._initialize()
2023-10-17T11:08:49.758824-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/capture.py", line 589, in _initialize
2023-10-17T11:08:49.759269-04:00 allsky python3[902]:     ccd_info = self.indiclient.getCcdInfo()
2023-10-17T11:08:49.759712-04:00 allsky python3[902]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-10-17T11:08:49.760192-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/camera/indi.py", line 556, in getCcdInfo
2023-10-17T11:08:49.760657-04:00 allsky python3[902]:     gain_info = self.getCcdGain()
2023-10-17T11:08:49.761074-04:00 allsky python3[902]:                 ^^^^^^^^^^^^^^^^^
2023-10-17T11:08:49.761487-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/camera/indi.py", line 1006, in getCcdGain
2023-10-17T11:08:49.761890-04:00 allsky python3[902]:     gain_ctl = self.get_control(self._ccd_device, 'CCD_GAIN', 'number')
2023-10-17T11:08:49.762460-04:00 allsky python3[902]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-10-17T11:08:49.762909-04:00 allsky python3[902]:   File "/home/matt/indi-allsky/indi_allsky/camera/indi.py", line 1223, in get_control
2023-10-17T11:08:49.763362-04:00 allsky python3[902]:     raise TimeOutException('Timeout finding control {0}'.format(name))
2023-10-17T11:08:49.763859-04:00 allsky python3[902]: indi_allsky.exceptions.TimeOutException: Timeout finding control CCD_GAIN
```

Output from indi_getprop:
```
SVBONY CCD SV305.CONNECTION.CONNECT=On
SVBONY CCD SV305.CONNECTION.DISCONNECT=Off
SVBONY CCD SV305.DRIVER_INFO.DRIVER_NAME=SVBONY CCD
SVBONY CCD SV305.DRIVER_INFO.DRIVER_EXEC=indi_svbony_ccd
SVBONY CCD SV305.DRIVER_INFO.DRIVER_VERSION=1.4
SVBONY CCD SV305.DRIVER_INFO.DRIVER_INTERFACE=2
SVBONY CCD SV305.POLLING_PERIOD.PERIOD_MS=1000
SVBONY CCD SV305.DEBUG.ENABLE=Off
SVBONY CCD SV305.DEBUG.DISABLE=On
SVBONY CCD SV305.SIMULATION.ENABLE=Off
SVBONY CCD SV305.SIMULATION.DISABLE=On
SVBONY CCD SV305.CONFIG_PROCESS.CONFIG_LOAD=Off
SVBONY CCD SV305.CONFIG_PROCESS.CONFIG_SAVE=Off
SVBONY CCD SV305.CONFIG_PROCESS.CONFIG_DEFAULT=Off
SVBONY CCD SV305.CONFIG_PROCESS.CONFIG_PURGE=Off
SVBONY CCD SV305.ACTIVE_DEVICES.ACTIVE_TELESCOPE=Telescope Simulator
SVBONY CCD SV305.ACTIVE_DEVICES.ACTIVE_ROTATOR=Rotator Simulator
SVBONY CCD SV305.ACTIVE_DEVICES.ACTIVE_FOCUSER=Focuser Simulator
SVBONY CCD SV305.ACTIVE_DEVICES.ACTIVE_FILTER=CCD Simulator
SVBONY CCD SV305.ACTIVE_DEVICES.ACTIVE_SKYQUALITY=SQM
[... 204 lines cut ...]
SVBONY CCD SV305.CCD_STREAM_FRAME.X=0
SVBONY CCD SV305.CCD_STREAM_FRAME.Y=0
SVBONY CCD SV305.CCD_STREAM_FRAME.WIDTH=1920
SVBONY CCD SV305.CCD_STREAM_FRAME.HEIGHT=1080
SVBONY CCD SV305.CCD_STREAM_ENCODER.RAW=On
SVBONY CCD SV305.CCD_STREAM_ENCODER.MJPEG=Off
SVBONY CCD SV305.CCD_STREAM_RECORDER.SER=On
SVBONY CCD SV305.LIMITS.LIMITS_BUFFER_MAX=512
SVBONY CCD SV305.LIMITS.LIMITS_PREVIEW_FPS=10
Telescope Simulator.DOME_POLICY.DOME_IGNORED=On
Telescope Simulator.DOME_POLICY.DOME_LOCKS=Off
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_APERTURE=10
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_FOCAL_LENGTH=10
Telescope Simulator.TELESCOPE_INFO.GUIDER_APERTURE=120
Telescope Simulator.TELESCOPE_INFO.GUIDER_FOCAL_LENGTH=900
Telescope Simulator.SCOPE_CONFIG_NAME.SCOPE_CONFIG_NAME=indi-allsky
Telescope Simulator.USEJOYSTICK.ENABLE=Off
Telescope Simulator.USEJOYSTICK.DISABLE=On
Telescope Simulator.SNOOP_JOYSTICK.SNOOP_JOYSTICK_DEVICE=Joystick
Telescope Simulator.MOUNT_AXES.PRIMARY=1.6810361103125615045e-07
Telescope Simulator.MOUNT_AXES.SECONDARY=33
```

Cool project!  I'd appreciate any help on this.
-Matt

aaronwmorris, 2023-10-17:
It looks like the INDI developers changed the gain property in the 2.0.4 release if INDI for SVBony cameras.

Some background on this, the gain setting for INDI cameras is one of two possible settings (assuming they support gain).  The latest release of INDI appears to have switched it to the other setting for SVBony cameras.  This is the first time this has happened.

Let me work on this tonight.

mattyryu, 2023-10-17:
Awesome, thanks!  Let me know if I can provide any other data that you may need.

aaronwmorris, 2023-10-18:
Merged #995 with fixes.  I would normally test this with my sv305, but it is currently boxed up due to my current move.  Let me know how it works.

mattyryu, 2023-10-18:
Thanks for the quick turnaround!  It seems much better as the camera loads and I can view all the details about it in the GUI.  I do see that it seems stuck in this loop of exposure state: busy and I have yet to capture an image:
```
2023-10-18T09:59:48.632989-04:00 allsky [INFO] Capture-1-918/MainThread capture.saferun() #209: Camera last ready: 67.6s
2023-10-18T09:59:48.633490-04:00 allsky [INFO] Capture-1-918/MainThread capture.saferun() #210: Exposure state: BUSY
2023-10-18T09:59:48.633634-04:00 allsky [INFO] Capture-1-918/MainThread capture.detectNight() #1272: Sun altitude: 25:53:59.7
2023-10-18T09:59:48.633789-04:00 allsky [INFO] Capture-1-918/MainThread capture.detectMoonMode() #1290: Moon altitude: -18:28:57.6, phase 15.0%
2023-10-18T09:59:48.833919-04:00 allsky [WARNING] Capture-1-918/MainThread indi.getCcdTemperature() #852: Sensor temperature not supported
2023-10-18T09:59:48.934408-04:00 allsky [INFO] Capture-1-918/MainThread indi.getTelescopeRaDec() #842: Telescope Coord: RA 10.18, Dec 33.00
```

I rebooted and started an strace on the indi_svbony_ccd process and saw what appeared to be configuration happening until this:
```
futex(0x55afddcd40, FUTEX_WAKE_PRIVATE, 1) = 0
sendmsg(1, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="<?xml version='1.0'?>\n<setNumber"..., iov_len=260}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 260
pselect6(1, [0], NULL, NULL, NULL, NULL) = 1 (in [0])
read(0, "<setNumberVector device=\"Telesco"..., 2048) = 264
pselect6(1, [0], NULL, NULL, NULL, NULL <unfinished ...>) = ?
+++ killed by SIGSEGV +++
```

I started the strace again on indi_svbony_ccd and it seems to do this every second forever after:
```
pselect6(1, [0], NULL, NULL, NULL, NULL) = 1 (in [0])
read(0, "<setNumberVector device=\"Telesco"..., 2048) = 264
pselect6(1, [0], NULL, NULL, NULL, NULL) = 1 (in [0])
read(0, "<setNumberVector device=\"Telesco"..., 2048) = 262
pselect6(1, [0], NULL, NULL, NULL, NULL) = 1 (in [0])
read(0, "<setNumberVector device=\"Telesco"..., 2048) = 264
pselect6(1, [0], NULL, NULL, NULL, NULL) = 1 (in [0])
read(0, "<setNumberVector device=\"Telesco"..., 2048) = 264
```

This is my first time using this so it could entirely be user error and I need to tune the config.  I did try rebooting and unplugging the camera for a while to make sure it wasn't locked up.  Same result.

aaronwmorris, 2023-10-18:
I am really not sure the INDI 2.0.4 SVBony driver is actually stable in its current form.  Apparently, the driver was completely rewritten last month.

https://github.com/indilib/indi-3rdparty/pull/839

You might want to try INDI 2.0.2 or 2.0.1.

mattyryu, 2023-10-18:
Oh yeah look at that.  I'll work on dropping back the INDI driver.  Thanks again for the help!

SNinfinite, 2023-10-18:
I have INDI 2.0.4 installed. Could you give me some instructions how to downgrade to an earlier version. I'm very new to linux and willing to learn! Thanks

mattyryu, 2023-10-18:
@aaronwmorris I can confirm that running with INDI v2.0.2, the SV305 is working as expected with indi-allsky v7.4 (to go back to the old way of collecting the gain settings before your most recent fix for the v2.0.4 INDI).  I can retest your changes when INDI v2.0.5 is release and also report any driver issues in the INDI project.

@SNinfinite I opted to grab a new SD card and install Bullseye/64-bit to be compatible with the v7.4 release of indi-allsky.  The setup.sh script will detect if the INDI drivers are installed already, so I went to the [indi-core](https://github.com/indilib/indi/tree/v2.0.2) and [indi-3rdparty](https://github.com/indilib/indi-3rdparty/tree/v2.0.2) repos and grabbed v2.0.2 and followed the build instructions.  After that I grabbed v7.4 of indi-allsky and ran thru the setup.  It is possible that you could grab v2.0.2 and follow the instructions to build those on your current system, but I'm not sure it'll work and didn't want to deal with resolving any issues if they came up.

closed 2023-10-18 by mattyryu

aaronwmorris, 2023-10-18:
Let me know about your testing.  The changes I made in the newer indi-allsky releases were intended to be backwards compatible, as well.

mattyryu, 2023-10-19:
I didn't realize the changes were backwards compatible.  I just installed the latest bookworm/64-bit with the latest indi-allsky and the INDI v2.0.2 and everything came up okay.

@SNinfinite so it turns out @aaronwmorris made it really easy for us and you can run:
``` ./misc/build_indi.sh v2.0.2 ``` from the indi-allsky directory to install the v2.0.2 INDI components (even over v2.0.4) without any other work. Amazing!