aaronwmorris/indi-allsky #1000: Python package build on Armbian fails: ERROR: No matching distribution found for numpy==1.21.2

jkoenig72, 2023-10-23:
Collecting astropy (from -r /home/fritz/indi-allsky/requirements/requirements_latest_32.txt (line 3))
  Using cached astropy-5.3.4.tar.gz (7.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy (from -r /home/fritz/indi-allsky/requirements/requirements_latest_32.txt (line 6))
  Using cached numpy-1.26.1-cp311-cp311-linux_armv7l.whl
Collecting opencv-python-headless<=4.6.0.66 (from -r /home/fritz/indi-allsky/requirements/requirements_latest_32.txt (line 8))
  Using cached opencv-python-headless-4.6.0.66.tar.gz (90.3 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      Ignoring numpy: markers 'python_version == "3.6" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
      Ignoring numpy: markers 'python_version == "3.7" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
      Ignoring numpy: markers 'python_version == "3.8" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
      Ignoring numpy: markers 'python_version <= "3.9" and sys_platform == "linux" and platform_machine == "aarch64"' don't match your environment
      Ignoring numpy: markers 'python_version <= "3.9" and sys_platform == "darwin" and platform_machine == "arm64"' don't match your environment
      Ignoring numpy: markers 'python_version == "3.9" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
      Collecting setuptools==59.2.0
        Using cached setuptools-59.2.0-py3-none-any.whl (952 kB)
      Collecting wheel==0.37.0
        Using cached wheel-0.37.0-py2.py3-none-any.whl (35 kB)
      Collecting cmake>=3.1
        Using cached cmake-3.27.7.tar.gz (35 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting pip
        Using cached pip-23.3.1-py3-none-any.whl.metadata (3.5 kB)
      Collecting scikit-build>=0.13.2
        Using cached scikit_build-0.17.6-py3-none-any.whl.metadata (14 kB)
      ERROR: Ignored the following versions that require a different python version: 1.21.2 Requires-Python >=3.7,<3.11; 1.21.3 Requires-Python >=3.7,<3.11; 1.21.4 Requires-Python >=3.7,<3.11; 1.21.5 Requires-Python >=3.7,<3.11; 1.21.6 Requires-Python >=3.7,<3.11
      ERROR: Could not find a version that satisfies the requirement numpy==1.21.2 (from versions: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0, 1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.4, 1.11.0, 1.11.1, 1.11.2, 1.11.3, 1.12.0, 1.12.1, 1.13.0, 1.13.1, 1.13.3, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0, 1.15.1, 1.15.2, 1.15.3, 1.15.4, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6, 1.17.0, 1.17.1, 1.17.2, 1.17.3, 1.17.4, 1.17.5, 1.18.0, 1.18.1, 1.18.2, 1.18.3, 1.18.4, 1.18.5, 1.19.0, 1.19.1, 1.19.2, 1.19.3, 1.19.4, 1.19.5, 1.20.0, 1.20.1, 1.20.2, 1.20.3, 1.21.0, 1.21.1, 1.22.0, 1.22.1, 1.22.2, 1.22.3, 1.22.4, 1.23.0rc1, 1.23.0rc2, 1.23.0rc3, 1.23.0, 1.23.1, 1.23.2, 1.23.3, 1.23.4, 1.23.5, 1.24.0rc1, 1.24.0rc2, 1.24.0, 1.24.1, 1.24.2, 1.24.3, 1.24.4, 1.25.0rc1, 1.25.0, 1.25.1, 1.25.2, 1.26.0b1, 1.26.0rc1, 1.26.0, 1.26.1)
      ERROR: No matching distribution found for numpy==1.21.2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

jkoenig72, 2023-10-23:
> What kind of board are you running?

An old orange pi zero running
Armbian 23.8.3 Jammy with Linux 6.1.53-current-sunxi
And I followed the steps mentioned in the wiki to tune armbian.
Now I run pip it step by step within the virtual env. -  now
installing opencv-python-headless without any version number, looks better
but takes ages.

/jk

aaronwmorris, 2023-10-23:
Yeah, my suggestion was going to be to replace the 32bit python requirements with the non-32bit.

It was bad enough with 32-bit Python 3.9 and 3.10, but the new distributions are being released with 3.11 and now there are even less wheels.  It is going to take a couple of months for the situation to improve.

aaronwmorris, 2023-10-31:
Did you finally get up and running?

jkoenig72, 2023-11-01:
Yes, it is working.

aaronwmorris, 2023-11-06:
Closing for now.  Let me know if there is anything else I can help with.

closed 2023-11-06 by aaronwmorris