aaronwmorris/indi-allsky #1002: Camera not found - the server encountered an internal error ...

wsuy, 2023-10-26:
With a fresh bookworm installation on a RPI 4B and the latest indi-allsky build (indi 2.0.4), the ZWO678mc camera is not found. The software continues to report an internal error on camera info, despite multiple new fresh installltions. Log:
2023-10-26T22:10:52.029519+02:00 raspberrypi [INFO] Capture-20-2062/Dummy-5 indi.newDevice() #204: new device V4L2 CCD
2023-10-26T22:10:51.976207+02:00 raspberrypi [INFO] Capture-20-2062/Dummy-5 indi.newDevice() #204: new device Telescope Simulator
2023-10-26T22:10:51.970562+02:00 raspberrypi [INFO] Capture-20-2062/MainThread indi.serverConnected() #336: Server connected (localhost:7624)
2023-10-26T22:10:51.969892+02:00 raspberrypi [INFO] Capture-20-2062/MainThread capture._initialize() #446: Connecting to indiserver
2023-10-26T22:10:51.968089+02:00 raspberrypi [INFO] Capture-20-2062/MainThread indi.__init__() #136: PyIndi version: 2.0.4
2023-10-26T22:10:51.967186+02:00 raspberrypi [INFO] Capture-20-2062/MainThread indi.__init__() #129: creating an instance of IndiClient
2023-10-26T22:10:51.953358+02:00 raspberrypi [INFO] MainProcess-1958/MainThread allsky._queueManualTasks() #1005: Checking for manually submitted tasks
2023-10-26T22:10:51.939085+02:00 raspberrypi [INFO] MainProcess-1958/MainThread allsky._startCaptureWorker() #287: Starting Capture-20 worker
2023-10-26T22:10:51.938502+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception: 
2023-10-26T22:10:51.937943+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception: indi_allsky.exceptions.TimeOutException: Timeout finding control CCD_EXPOSURE
2023-10-26T22:10:51.937355+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     raise TimeOutException('Timeout finding control {0}'.format(name))
2023-10-26T22:10:51.936628+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/admin/indi-allsky/indi_allsky/camera/indi.py", line 1275, in get_control
2023-10-26T22:10:51.936041+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-10-26T22:10:51.935421+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     ctl_CCD_EXPOSURE = self.get_control(self._ccd_device, 'CCD_EXPOSURE', 'number')
2023-10-26T22:10:51.934746+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/admin/indi-allsky/indi_allsky/camera/indi.py", line 495, in getCcdInfo
2023-10-26T22:10:51.934018+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-10-26T22:10:51.933109+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     ccd_info = self.indiclient.getCcdInfo()
2023-10-26T22:10:51.932321+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/admin/indi-allsky/indi_allsky/capture.py", line 589, in _initialize
2023-10-26T22:10:51.931720+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     self._initialize()
2023-10-26T22:10:51.931104+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/admin/indi-allsky/indi_allsky/capture.py", line 188, in saferun
2023-10-26T22:10:51.930491+02:00 raspberrypi [ERROR] MainProcess-1958/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     self.saferun()

aaronwmorris, 2023-10-26:
You need to re-run setup.sh and select the `indi_asi_ccd` driver and then restart the indiserver service.

```
systemctl --user restart indiserver
```

wsuy, 2023-10-27:
The "indi_asi_ccd" driver is not offered anywhere. Running setup.sh somewhere shows a warning:

Detected INDI version: 2.0.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Cloning https://github.com/indilib/pyindi-client.git (to revision 6f8fa80) to /tmp/pip-install-sjksrwpt/pyindi-client_c28af04159ad49c498c4ee7c3f5b83ab
  Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-sjksrwpt/pyindi-client_c28af04159ad49c498c4ee7c3f5b83ab
  WARNING: Did not find branch or tag '6f8fa80', assuming revision or ref.
  Running command git checkout -q 6f8fa80
  Resolved https://github.com/indilib/pyindi-client.git to commit 6f8fa80
  Preparing metadata (setup.py) ... done

Then, only  2 driver options are consistently offered: "indi_simulator_ccd" and "indi_v4l2_ccd", exactly as offered during first install. I tried running setup both with and without the camera plugged in.

aaronwmorris, 2023-10-27:
Sounds like you installed the base indi system, but not the 3rd party drivers yet.

The 3rdparty repo can be found here:  https://github.com/indilib/indi-3rdparty

aaronwmorris, 2023-10-27:
How did you originally build INDI?

wsuy, 2023-10-27:
Thank you, that will be the solution. Setp.sh indeed mentions skipping third party driver install. As far as I remember, this is different from a previous install that I did earlier this year.

Being a beginner in Linux, could you help me with the required command line commands to install these camera drivers? The instructions on the repo page are a bit confusing to me.

aaronwmorris, 2023-10-27:
The wiki only recommends skipping the 3rdparty if you intend on using a libcamera supported camera.

Building the full INDI only requires you to run the INDI build script with no parameters.

```
./misc/build_indi.sh
```

If you already have core installed, you can skip that and just install 3rd party.

```
BUILD_INDI_CORE=false BUILD_INDI_3RDPARTY=true ./misc/build_indi.sh
```

wsuy, 2023-10-27:
Problem solved.
I am sorry, I completely overlooked the option to set thirdparty driver build=true in the INDI build command!
Everything working fine now.

aaronwmorris, 2023-10-27:
No worries.  I realize the instructions can be confusing.  I will look in to clarifying them.

aaronwmorris, 2023-10-31:
Closing this for now.  Let me know if you have any more issues.

closed 2023-10-31 by aaronwmorris