aaronwmorris/indi-allsky #1003: Setup.sh breaks on bookworm

Sejl, 2023-10-27:
Hi. I have been trying to install indi-allsky on raspberry without success.
I have tried on raspberry pi v3 b+ and raspberry pi 4 compute module.
On both I have installed latest bookworm 32bit lite. Indi was compiled without 3rd party libraries.

Error that I get from setup.sh is:

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libimath-dev : Conflicts: libilmbase-dev
 libopenexr-dev : Breaks: libilmbase-dev (< 2.5.7-3~)
E: Unable to correct problems, you have held broken packages.

The script exited abnormally, please try to run again...

I will now try to install with bullseye to see if the problem persists.
Thank you for any suggestion.

Best regards

Sejl, 2023-10-27:
Just to report that bullseye does not have this problem.
Everything goes smoothly.

Best regards

aaronwmorris, 2023-10-27:
This is a problem with packages in the 32-bit version of bookworm.  Everything should work fine with the 64-bit release.

Sejl, 2023-10-27:
Thank you!
I will try with 64bit release.

Best regards

aaronwmorris, 2023-10-31:
Did the 64-bit release work any better for you?

Sejl, 2023-11-01:
> Did the 64-bit release work any better for you?

Yes! 
Sorry for the late reply 🙈

It works on bookworm 64-bit on rpi compute module 4.

Thank you for your help!

Best regards

aaronwmorris, 2023-11-06:
Closing for now.  Let me know if I can do anything else!

closed 2023-11-06 by aaronwmorris