aaronwmorris/indi-allsky #1004: v7.6 release - keogram/startrail broken at night?

StarGeezerPhil, 2023-10-27:
@aaronwmorris upgraded straight to 7.6 after receiving the notification.

Not had chance to check through the log files etc. yet, but noticed that night time timelapses etc. seem to be broken since the update:

[image]

The timelapse video works absolutely fine, but the keogram and holding image for the star trail seems to be missing. Abysmal weather here so can't comment about actual star trail generation or the star trail timelapse video.

aaronwmorris, 2023-10-27:
My guess is it is related to #1001 assuming you are using a detection mask.  You may need to generate a new detection mask based on the new way of processing the mask.

Let me know if you need some guidance on this, it is very new.

StarGeezerPhil, 2023-10-28:
Yes, using detection mask, but it's the original dimensions and matches the mask_base.png (thanks for adding this as a reference file - very useful! 👍).

aaronwmorris, 2023-10-29:
You may have to try to generate the startrails/keogram again and reference the logs for errors.  I did some basic testing before releasing, but only with a virtual camera.  Currently, my cameras are packed up due to moving, so I have not tested as much as I would have liked.

aaronwmorris, 2023-10-30:
Today I merged #1006 because I forgot to include the scaling step for the mask.  I do not think this will help you, but maybe...

aaronwmorris, 2023-10-30:
#1008 Another fix.  I will get this right eventually.

StarGeezerPhil, 2023-10-30:
Brilliant @aaronwmorris - that's fixed it 👍🔥

closed 2023-10-30 by StarGeezerPhil