aaronwmorris/indi-allsky #1012: error on upgrading on opi pc

pierrebilb, 2023-11-05:
I tried to upgrade indi-allsky from 10 days ago and got error; running armbian 23.8.1 kernel 6.1.53 on orange pi pc:
 FAILED: scipy/interpolate/libfitpack_lib.a.p/fitpack_fppasu.f.o
      gfortran -Iscipy/interpolate/libfitpack_lib.a.p -Iscipy/interpolate -I../scipy/interpolate -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -std=legacy -O3 -Wno-conversion -fPIC -Wno-maybe-uninitialized -Jscipy/interpolate/libfitpack_lib.a.p -o scipy/interpolate/libfitpack_lib.a.p/fitpack_fppasu.f.o -c ../scipy/interpolate/fitpack/fppasu.f
      ../scipy/interpolate/fitpack/fppasu.f:393:72:

        393 |       end
            |                                                                        ^
      Fatal Error: error writing to /tmp/ccvJlma2.s: No space left on device
      compilation terminated.
      [1239/1619] Compiling Fortran object scipy/interpolate/libfitpack_lib.a.p/fitpack_fpperi.f.o
      FAILED: scipy/interpolate/libfitpack_lib.a.p/fitpack_fpperi.f.o
      gfortran -Iscipy/interpolate/libfitpack_lib.a.p -Iscipy/interpolate -I../scipy/interpolate -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -std=legacy -O3 -Wno-conversion -fPIC -Wno-maybe-uninitialized -Jscipy/interpolate/libfitpack_lib.a.p -o scipy/interpolate/libfitpack_lib.a.p/fitpack_fpperi.f.o -c ../scipy/interpolate/fitpack/fpperi.f
      ../scipy/interpolate/fitpack/fpperi.f:617:72:

        617 |       end
            |                                                                        ^
      Fatal Error: error writing to /tmp/cclzaHbN.s: No space left on device
      compilation terminated.
      [1240/1619] Compiling C object scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__peak_finding_utils.c.o
      FAILED: scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__peak_finding_utils.c.o
      cc -Iscipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p -Iscipy/signal -I../scipy/signal -I../../../pip-build-env-y1wtfcul/overlay/lib/python3.10/site-packages/numpy/core/include -I/usr/include/python3.10 -I/usr/include/arm-linux-gnueabihf/python3.10 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c99 -O3 -Wno-unused-but-set-variable -Wno-unused-function -Wno-conversion -Wno-misleading-indentation -fPIC -Wno-cpp -MD -MQ scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__peak_finding_utils.c.o -MF scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__peak_finding_utils.c.o.d -o scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__peak_finding_utils.c.o -c scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/_peak_finding_utils.c
      scipy/signal/_peak_finding_utils.cpython-310-arm-linux-gnueabihf.so.p/_peak_finding_utils.c:26962:1: fatal error: error writing to /tmp/ccnS9HhV.s: No space left on device
      26962 | }
            | ^
      compilation terminated.
      [1241/1619] Compiling C object scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__sosfilt.c.o
      FAILED: scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__sosfilt.c.o
      cc -Iscipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p -Iscipy/signal -I../scipy/signal -I../../../pip-build-env-y1wtfcul/overlay/lib/python3.10/site-packages/numpy/core/include -I/usr/include/python3.10 -I/usr/include/arm-linux-gnueabihf/python3.10 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c99 -O3 -Wno-unused-but-set-variable -Wno-unused-function -Wno-conversion -Wno-misleading-indentation -fPIC -Wno-cpp -MD -MQ scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__sosfilt.c.o -MF scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__sosfilt.c.o.d -o scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__sosfilt.c.o -c scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/_sosfilt.c
      scipy/signal/_sosfilt.cpython-310-arm-linux-gnueabihf.so.p/_sosfilt.c:28927:1: fatal error: error writing to /tmp/cc9xVDUy.s: No space left on device
      28927 | }
            | ^
      compilation terminated.
      [1242/1619] Compiling C++ object scipy/sparse/sparsetools/_sparsetools.cpython-310-arm-linux-gnueabihf.so.p/bsr.cxx.o
      FAILED: scipy/sparse/sparsetools/_sparsetools.cpython-310-arm-linux-gnueabihf.so.p/bsr.cxx.o
      c++ -Iscipy/sparse/sparsetools/_sparsetools.cpython-310-arm-linux-gnueabihf.so.p -Iscipy/sparse/sparsetools -I../scipy/sparse/sparsetools -I../../../pip-build-env-y1wtfcul/overlay/lib/python3.10/site-packages/numpy/core/include -I/usr/include/python3.10 -I/usr/include/arm-linux-gnueabihf/python3.10 -fvisibility=hidden -fvisibility-inlines-hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++14 -O3 -fPIC -DNPY_NO_DEPRECATED_API=NPY_1_9_API_VERSION -MD -MQ scipy/sparse/sparsetools/_sparsetools.cpython-310-arm-linux-gnueabihf.so.p/bsr.cxx.o -MF scipy/sparse/sparsetools/_sparsetools.cpython-310-arm-linux-gnueabihf.so.p/bsr.cxx.o.d -o scipy/sparse/sparsetools/_sparsetools.cpython-310-arm-linux-gnueabihf.so.p/bsr.cxx.o -c ../scipy/sparse/sparsetools/bsr.cxx
      ../scipy/sparse/sparsetools/bsr.cxx:9:1: fatal error: error writing to /tmp/cc8Kui93.s: No space left on device
          9 | }
            | ^
      compilation terminated.
      [1243/1619] Compiling C object scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__upfirdn_apply.c.o
      FAILED: scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__upfirdn_apply.c.o
      cc -Iscipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p -Iscipy/signal -I../scipy/signal -I../../../pip-build-env-y1wtfcul/overlay/lib/python3.10/site-packages/numpy/core/include -I/usr/include/python3.10 -I/usr/include/arm-linux-gnueabihf/python3.10 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c99 -O3 -Wno-unused-but-set-variable -Wno-unused-function -Wno-conversion -Wno-misleading-indentation -fPIC -Wno-cpp -MD -MQ scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__upfirdn_apply.c.o -MF scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__upfirdn_apply.c.o.d -o scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/meson-generated__upfirdn_apply.c.o -c scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/_upfirdn_apply.c
      scipy/signal/_upfirdn_apply.cpython-310-arm-linux-gnueabihf.so.p/_upfirdn_apply.c:41582:1: fatal error: error writing to /tmp/ccC9wfib.s: No space left on device
      41582 | }
            | ^
      compilation terminated.
      ninja: build stopped: subcommand failed.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

The script exited abnormally, please try to run again...

No idea what's going wrong as i installed without any problem indi-allsky 10 days ago.

aaronwmorris, 2023-11-05:
```
Fatal Error: error writing to /tmp/ccvJlma2.s: No space left on device
```

You just ran out of space on the /tmp filesystem.  You can try these options in my wiki to help with this.

https://github.com/aaronwmorris/indi-allsky/wiki/Armbian-Tuning

Once the modules are compiled, you can return these settings back to the defaults.

pierrebilb, 2023-11-05:
I had to modify swapfile config file and put swapsize 4 times RAM size to upgrade successfully..

closed 2023-11-05 by pierrebilb