aaronwmorris/indi-allsky #1020: Frame is dark even though gain and exposure look ok

crankmosh, 2023-11-13:
Camera: SV305Pro
Ubuntu 22.04 (patched up to latest)
Mini PC x86

Here is a sample video: https://www.youtube.com/watch?v=t7XKc2_C-ek

Some frames that are grabbed are dark, even though the gain and exposure time seem regular.

aaronwmorris, 2023-11-13:
Did you compile indi using the build_indi.sh script or did you have it installed from the repo?

crankmosh, 2023-11-13:
repo only has 2.0.4 which doesn't work with svbony so I had to use build_indi.sh, the odd part is, the dark frames still seem to show stars even though they are just really dark

aaronwmorris, 2023-11-13:
During your next run, can you check the "Image Lag" view under tools and see if the dark frames are being received in less time than the exposure?

aaronwmorris, 2023-11-14:
If the frames are darker, it means one of two things:

- The camera has reduced the gain behind the scenes
- or the real exposure was much shorter than requested/reported

The exposure being shorter than reported is more likely.  An issue I experienced with my SV305 was once you go above 5-6 seconds on exposures, approximately 25% of the frames would require double the time to complete.  The exposures were still correct, but the camera spent half the time doing nothing.

This issue was why I started tracking the time it actually takes to get an exposure from a camera in the database.  This has helped me diagnose problematic cameras in many cases.  You just cannot trust the hardware to do what you tell it to do sometimes.

ikeysolomon, 2024-06-12:
I noticed this last night. I have re-installed onto a rpi5 now (an upgrade from the 3b+, with a view to perform onboard timelapse generation). I will see how things go tonight and report back. This wasn't noticed with what I had installed on the 3b+ only say a fortnight ago?

ikeysolomon, 2024-06-13:
Can report back that darks look much better tonight for some reason? Magic of settling in for a day haha.

aaronwmorris, 2024-06-13:
Its possible this was a bug in the version of INDI you were using at the time.  The SVBony driver went through some major changes just before you opened this ticket, it was completely rewritten.

https://github.com/indilib/indi-3rdparty/pull/839

I am not saying this was the cause, but bugs can be expected with such a major rewrite.

closed 2025-02-20 by aaronwmorris