aaronwmorris/indi-allsky #1029: External temperature script

wsuy, 2023-11-19:
Hi,
I am trying to make a py script to read out a DHT22 external temperature sensor. The script. however, keeps resulting in an error when executed by indi-allsky. What am I doing wrong?

#!/usr/bin/env python3

import os
import io
import json
import logging
import Adafruit_DHT
import sys

logging.basicConfig(level=logging.INFO)
logger = logging

sensor = Adafruit_DHT.DHT22
pin = 4

# Set the path for the output JSON file
temp_json_path = "~/indi-allsky/output.json"

try:
    # Set the environment variable TEMP_JSON
    os.environ['TEMP_JSON'] = os.path.expanduser(temp_json_path)
    temp_json = os.environ['TEMP_JSON']
except KeyError:
    logger.error('TEMP_JSON environment variable is not defined')
    sys.exit(1)

humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)

if temperature is not None:
    # dict to be used for json data
    temp_data = {
        'temp': temperature,
    }

    # write json data
    with io.open(temp_json, 'w') as f_temp_json:
        json.dump(temp_data, f_temp_json, indent=4)

    print('Temperature={0:0.1f}*C'.format(temperature))
else:
    logger.error('Failed to read temperature data from DHT22 sensor')
    sys.exit(1)

wsuy, 2023-11-19:
Already solved: I forgot to make the py script executable: chmod +x etc.

aaronwmorris, 2023-11-19:
What error are you receiving?

wsuy, 2023-11-20:
When pointing to the scrip in indi-allsky config, the message ‘script failed to execute’ is returned.

aaronwmorris, 2023-11-20:
It looks like the script returned with a non-zero exit code.  Did you install the `Adafruit_DHT` module in the indi-allsky virtualenv?

```
source ./virtualenv/indi-allsky/bin/activate

pip install Adafruit_DHT
```

wsuy, 2023-11-20:
Yes, I installed again using your commands, but still get: Temperature script failed to execute

wsuy, 2023-11-20:
The webapp-indi-allsky.log file shows:
Nov 20 19:32:41 raspberrypi gunicorn.indi_allsky.wsgi.error: [3546] Starting gunicorn 21.2.0
Nov 20 19:32:41 raspberrypi gunicorn.indi_allsky.wsgi.error: [3546] Listening at: unix:/var/lib/indi-allsky/gunicorn-in>Nov 20 19:32:41 raspberrypi gunicorn.indi_allsky.wsgi.error: [3546] Using worker: gthread
Nov 20 19:32:42 raspberrypi gunicorn.indi_allsky.wsgi.error: [3548] Booting worker with pid: 3548
Nov 20 19:36:06 raspberrypi [INFO] MainProcess-3548 glogging.info() #278: Worker reloading: Adafruit_DHT modified
Nov 20 19:36:06 raspberrypi [INFO] MainProcess-3548 glogging.info() #278: Worker exiting (pid: 3548)
Nov 20 19:36:07 raspberrypi gunicorn.indi_allsky.wsgi.error: [3653] Booting worker with pid: 3653

Running the above command returns: 
# run script
echo $?
0

aaronwmorris, 2023-11-20:
Merged #1033 to show more detail why the temperature script failed to execute.  Pull down the latest code and hopefully this will point you in the right direction.

aaronwmorris, 2023-11-20:
And merged #1034 to correctly log.

aaronwmorris, 2023-11-25:
Were you able to determine why the script was not running?

wsuy, 2023-11-25:
Yes, Indi-Allsky simply couldn't find the script. With the logging this was quickly resolved. Thank you!

closed 2023-11-25 by aaronwmorris