aaronwmorris/indi-allsky #1030: 64 bit Bookworm and HQ camera fails

warwound, 2023-11-19:
I've had indi-allsky with Pi HQ camera running on a 4GB Pi 4 for a few weeks.
I recently bought a 4GB Pi 5 and thoughtI'd get indi-allsky installed.
64 bit Bookworm with desktop is installed on my Pi 5.

I followed the same install procedure as with my Pi 4.
After pulling indi-allsky from github, I execute:

[code]
BUILD_INDI_CORE=true BUILD_INDI_3RDPARTY=false ./misc/build_indi.sh

./setup.sh
[/code]

No image is captured and I see no error logged.
I do see the libcamera-still command in the log viewer but it never seems to succeed.

[quote]
2023-11-19T13:51:52.668046+00:00 pi5 [INFO] Capture-1-3197/MainThread libcamera.setCcdExposure() #241: image command: libcamera-still --immediate --nopreview --raw --denoise off --gain 1 --shutter 1000 --metadata /tmp/tmp2kzamo07.json --metadata-format json --awbgains 1,1 -camera 0 --output /tmp/tmpzsjtlpes.dng
[/quote]

You can see I've added '-camera 0' as an additional libcamera option, I have tried '-camera 1' too as well as no additional option at all.

Eventually the web interface shows a dialog box with the message:

[quote]
Notification: Camera
Date: 2023-11-19 14:01:53

Camera last ready 600s ago. Camera might be hung. Aborting exposure.
[/quote]

If I stop indi-allsky service and run a simple libcamera command on my desktop it works fine:

[code]
libcamera-still -t 0
[/code]

Note I've not had to use the '-camera ?' option for my desktop command.

On both my Pi 4 and Pi 5 I have installed the Arducam Hawkeye 64mp camera driver and Arducam version of libcamera.
This has not prevented me from using my HQ camera on my Pi 4 so I assume it won't prevent me from using my HQ camera on my Pi 5.
(Arducam have not yet released Bookworm compatible drivers for the Hawkeye: https://forum.arducam.com/t/new-raspberry-pi-os-bookworm-and-64mp-arducam/5831)

So what's the problem, where can I look for relevant error messages?

Thanks.

aaronwmorris, 2023-11-19:
I have experienced similar behavior on bookworm, but have not done much testing to see what could be causing the issue.  Give me a few days to dig in and see what I can find.

warwound, 2023-11-20:
OK, if I can help then please let me know.
Meanwhile I plan to uninstall the Arducam Hawkeye 64mp camera driver and Arducam version of libcamera, revert back to official Pi libcamera and then test again.

warwound, 2023-11-20:
Uninstalling the Arducam Hawkeye 64mp camera driver and Arducam version of libcamera appears to be undocumented, I found no way to do so.
So I instead reinstalled 64 bit Bookworm with desktop and fully updated it.
I then installed indi-allsky.

So my setup is now as before except I do not have the Arducam Hawkeye 64mp camera driver and Arducam version of libcamera installed.
The result is the same, no image is captured and eventually an error dialog appears:

[quote]
Notification: Camera
Date: 2023-11-20 16:23:28

Camera last ready 600s ago. Camera might be hung. Aborting exposure.
[/quote]

aaronwmorris, 2023-11-20:
The problem seems to revolve around the `--immediate` flag.  If I remove this flag, an image is captured.  The problem is without the --immediate flag, the "viewfinder" mode is activated and two images are captured, one lower resolution first, then the second full resolution image is returned.

warwound, 2023-11-20:
I just executed this command three times in a terminal on my Pi 5 desktop:

[code]
libcamera-still --immediate -o my_pic.jpg
[/code]

Each time the preview window appears and then the libcamera command seems to hang.
I needed to kill the libcamera process each time using task manager.

So the preview window should not appear when the --immediate flag is set, but on my Pi 5 (and your's?) it does.

aaronwmorris, 2023-11-20:
I am testing on a Pi4, just on bookworm.  This appears to be an issue with with bookworm.

aaronwmorris, 2023-11-21:
A fix has been merged for rpicam-apps.  After rebuilding libcamera and rpicam-apps from source, I can confirm I am receiving images on bookworm

warwound, 2023-11-21:
Great...
How long will it be before the fix is available to me via an 'apt update', any idea?
Can you share your fix or point me to a guide build a fix myself?
Thanks.

aaronwmorris, 2023-11-21:
I have been waiting on new libcamera updates for a couple of months for some other fixes as well.  I do not know when a new version will be released.

The docs for building libcamera and rpicam-apps is pretty decent.

https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera

https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps

warwound, 2023-11-21:
Will the instructions of both of those pages - where 'git clone' is used - get the version that contains today's fix?
Or do I need to specify some version in order to get it?

aaronwmorris, 2023-11-21:
It should, yes.  All of the fixes should be available in HEAD.

I followed those instructions pretty much to the letter.

warwound, 2023-11-21:
The instructions are clear and easy to follow and my HQ camera now works.
Anyone else with the same problem can do as I did or wait patiently for the official fix.
Thanks for your help.

warwound, 2023-11-23:
I just wanted to add to this thread that capture without immediate is now broken!
I run the simple command:

[code]
libcamera-still -t 0
[/code]

No preview window appears, this is the terminal output:

[quote]
martin@pi5:~ $ libcamera-still -t 0
[0:03:23.137200149] [2315]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+117-99b177b3
[0:03:23.152331420] [2319]  INFO RPI pisp.cpp:653 libpisp version v1.0.1 5652744a74d7 21-11-2023 (18:43:28)
[0:03:23.170270853] [2319]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
Stream configuration adjusted
[0:03:23.172669970] [2315]  INFO Camera camera.cpp:1183 configuring streams: (0) 2028x1520-YUV420 (1) 2028x1520-BGGR16_PISP_COMP1
[0:03:23.172894844] [2319]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected CFE format: 2028x1520-PC1B
#0 (0.00 fps) exp 32987.00 ag 8.00 dg 1.00
#1 (30.01 fps) exp 32987.00 ag 8.00 dg 1.00
#2 (30.01 fps) exp 32987.00 ag 8.00 dg 1.00
#3 (30.01 fps) exp 32987.00 ag 8.00 dg 1.00
#4 (30.01 fps) exp 32987.00 ag 8.00 dg 1.00
[/quote]

Those numbered lines #0, #1 etc just continue to appear until I halt the command.
See the line 'Preview window unavailable'?

aaronwmorris, 2023-11-24:
This could have something to do with the options you passed to the meson build commands when you built the libcamera projects from source.

warwound, 2023-11-24:
> This could have something to do with the options you passed to the meson build commands when you built the libcamera projects from source.

Not quite but you pointed me in the right direction...
I had not installed all of the recommended packages on the 'build libcamera and rpicam-apps' page that you linked to.
With all the packages installed, I recompiled both libcamera and rpicam-apps and my preview window now appears.

Thanks again for your help.

closed 2023-11-24 by warwound