aaronwmorris/indi-allsky #1038: Unable to initialise Remote Web Interface likely due to 'No Camera in DB' and CCD device INDI Webcam

keno242, 2023-11-26:
I've been unable to get the remote web interface operational using SyncAPI - I believe because the remote allsky instance is unable to create an initial camera. 

I've followed the discussion here https://github.com/aaronwmorris/indi-allsky/discussions/569 and it appears I may be suffering a similar problem where on the remote setup initally using the INDI Webcam option it's trying (and failing) to query the CCD_EXPOSURE (there is no physical camera on the remote) 

Relevant sections of the logs are below - got any suggestions? TIA.

`/var/log/indi-allsky/indi-allsky.log`  on the remote...

```
Nov 26 11:08:30 allskynas [INFO] Capture-11-369/MainThread indi_passive.__init__() #83: creating an instance of IndiClient
Nov 26 11:08:30 allskynas [INFO] Capture-11-369/MainThread indi_passive.__init__() #90: PyIndi version: 2.0.4
Nov 26 11:08:30 allskynas [INFO] Capture-11-369/MainThread capture._initialize() #448: Connecting to indiserver
Nov 26 11:08:30 allskynas [INFO] Capture-11-369/MainThread indi_passive.serverConnected() #286: Server connected (localhost:7624)
Nov 26 11:08:30 allskynas [INFO] Capture-11-369/Dummy-3 indi_passive.newDevice() #154: new device INDI Webcam
Nov 26 11:08:30 allskynas [INFO] Capture-11-369/Dummy-3 indi_passive.newDevice() #154: new device Telescope Simulator
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi._findCcds() #576: Searching for available cameras
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi._findCcds() #587:  Detected INDI Webcam
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi.findCcd() #596: Found 1 CCDs
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi._findTelescopes() #618: Searching for available telescopes/mounts
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi._findTelescopes() #629:  Detected Telescope Simulator
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi.findTelescope() #638: Found 1 Telescopess
Nov 26 11:08:38 allskynas [INFO] Capture-11-369/MainThread indi._findGpss() #651: Searching for available GPS interfaces
Nov 26 11:08:39 allskynas [INFO] Capture-11-369/MainThread indi.findGps() #671: Found 0 GPSs
Nov 26 11:08:39 allskynas [WARNING] Capture-11-369/MainThread capture._initialize() #485: Connecting to CCD device INDI Webcam
Nov 26 11:08:39 allskynas [WARNING] Capture-11-369/MainThread capture._initialize() #489: Connecting to Telescope device Telescope Simulator
Nov 26 11:08:39 allskynas [INFO] Capture-11-369/Dummy-3 indi_passive.newMessage() #283: new Message 2023-11-26T03:08:38: [INFO] Trying to connect to: /dev/video0, on device: video4linux2,v4l2 with 640x480 at 30 frames per second 
Nov 26 11:08:39 allskynas [INFO] Capture-11-369/Dummy-3 indi_passive.newMessage() #283: new Message 2023-11-26T03:08:38: [INFO] Attempting to connect 
Nov 26 11:08:39 allskynas [INFO] Capture-11-369/Dummy-3 indi_passive.newMessage() #283: new Message 2023-11-26T03:08:38: [INFO] Failed to open source. Check your settings: No such file or directory 
Nov 26 11:08:44 allskynas [INFO] MainProcess-323/MainThread allsky._queueManualTasks() #1005: Checking for manually submitted tasks
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception: Traceback (most recent call last):
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/khall/indi-allsky/indi_allsky/capture.py", line 176, in run
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     self.saferun()
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/khall/indi-allsky/indi_allsky/capture.py", line 188, in saferun
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     self._initialize()
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/khall/indi-allsky/indi_allsky/capture.py", line 591, in _initialize
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     ccd_info = self.indiclient.getCcdInfo()
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/khall/indi-allsky/indi_allsky/camera/indi.py", line 495, in getCcdInfo
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     ctl_CCD_EXPOSURE = self.get_control(self._ccd_device, 'CCD_EXPOSURE', 'number')
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:   File "/home/khall/indi-allsky/indi_allsky/camera/indi_passive.py", line 401, in get_control
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception:     raise TimeOutException('Timeout finding control {0}'.format(name))
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception: indi_allsky.exceptions.TimeOutException: Timeout finding control CCD_EXPOSURE
Nov 26 11:08:57 allskynas [ERROR] MainProcess-323/MainThread allsky._startCaptureWorker() #280: Capture worker exception: 
Nov 26 11:08:57 allskynas [INFO] MainProcess-323/MainThread allsky._startCaptureWorker() #287: Starting Capture-12 worker
Nov 26 11:08:57 allskynas [INFO] MainProcess-323/MainThread allsky._queueManualTasks() #1005: Checking for manually submitted tasks

```

web log on the remote allsky...
So we can see the syncapi requests are coming through

```
Nov 26 11:38:23 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:38:38 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:38:39 allskynas [WARNING] MainProcess-292 base_views.getCameraById() #93: No cameras are defined
Nov 26 11:38:53 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:38:58 allskynas [WARNING] MainProcess-292 base_views.getCameraById() #93: No cameras are defined
Nov 26 11:39:08 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:39:23 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:39:33 allskynas [WARNING] MainProcess-292 base_views.getCameraById() #93: No cameras are defined
Nov 26 11:39:38 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:39:53 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
Nov 26 11:40:08 allskynas [ERROR] MainProcess-292 syncapi_views.post() #105: Camera not found: 6440f61c-0c55-47e9-bb50-d3c7cadb5e8e
```

And the log on the main allsky - 

Showing the attempts to sync the images... 

```
2023-11-26T11:19:35.708956+08:00 allsky [INFO] Image-13-127174/MainThread image.calculate_exposure() #1396: ADU average: 82.00
2023-11-26T11:19:36.124035+08:00 allsky [INFO] Image-13-127174/MainThread image.processImage() #606: Image processed in 0.7712 s
2023-11-26T11:19:36.285030+08:00 allsky [INFO] Image-13-127174/MainThread image.write_img() #1229: Image compressed in 0.1593 s
2023-11-26T11:19:36.296181+08:00 allsky [INFO] Image-13-127174/MainThread miscDb.addImage() #179: Adding image ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231126/day/26_11/ccd1_20231126_111935.jpg to DB
2023-11-26T11:19:36.320577+08:00 allsky [INFO] MainProcess-1159/Upload-20 generic.connect() #40: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/image (443) as foobar123 with requests_s [... cut ...]
2023-11-26T11:19:36.320952+08:00 allsky [INFO] MainProcess-1159/Upload-20 generic.put() #50: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231126/day/26_11/ccd1_20231126_111935.jpg
2023-11-26T11:19:37.727985+08:00 allsky [ERROR] MainProcess-1159/Upload-20 uploader.processUpload() #408: Tranfer failure: Sync error: 400
2023-11-26T11:19:37.732402+08:00 allsky [WARNING] MainProcess-1159/Upload-20 miscDb.addNotification() #719: Not adding existing notification

```

aaronwmorris, 2023-11-26:
Thank you for finding this.  Merged #1040 to fix syncing of the camera object.

The camera object is supposed to be synchronized when indi-allsky starts, however, there was a bug that prevented that.

keno242, 2023-11-26:
Nice! Updated/restarted both instances and now have syncing - thanks...

```2023-11-26T13:24:39.123962+08:00 allsky [INFO] MainProcess-1362/MainThread allsky._fileUploadWorkerStart() #493: Starting Upload-1 worker
2023-11-26T13:24:52.474717+08:00 allsky [INFO] MainProcess-1362/Upload-1 generic.connect() #40: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/camera (443) as foobar123 with requests_syncapi_v1
2023-11-26T13:24:52.475103+08:00 allsky [INFO] MainProcess-1362/Upload-1 generic.put() #50: Uploading camera
2023-11-26T13:24:52.808147+08:00 allsky [INFO] MainProcess-1362/Upload-1 requests_syncapi_v1.put() #172: File transferred in 0.3318 s (3.01 kB/s)
2023-11-26T13:24:52.812282+08:00 allsky [INFO] MainProcess-1362/Upload-1 uploader.processUpload() #439: Upload transaction completed in 0.3369 s
2023-11-26T13:25:07.287304+08:00 allsky [INFO] MainProcess-1362/Upload-1 generic.connect() #40: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/image (443) as foobar123 with requests_syncapi_v1
2023-11-26T13:25:07.287664+08:00 allsky [INFO] MainProcess-1362/Upload-1 generic.put() #50: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231126/day/26_13/ccd1_20231126_132500.jpg
2023-11-26T13:25:08.746296+08:00 allsky [INFO] MainProcess-1362/Upload-1 requests_syncapi_v1.put() #172: File transferred in 1.4567 s (594.62 kB/s)
2023-11-26T13:25:08.750068+08:00 allsky [INFO] MainProcess-1362/Upload-1 uploader.processUpload() #439: Upload transaction completed in 1.4633 s
```

keno242, 2023-11-26:
One other question if I may - any thoughts on why when I try and upload the older time-lapse images etc manually using `./misc/upload_sync.py --syncapi-images sync` I run into TimeOutErrors ... 

May it have something do do with the local account I'm running `upload_sync.py` not having access to the API key?

```2023-11-26 13:47:14,769 [INFO] MainProcess upload_sync.sync() #162: Entries to upload: 47231
2023-11-26 13:47:14,772 [INFO] MainProcess upload_sync.addUploadEntries() #205: Adding 7 upload entries (47224 remaining)
2023-11-26 13:47:16,193 [INFO] MainProcess upload_sync._fileUploadWorkerStart() #720: Starting FileUploader process 1
2023-11-26 13:47:16,218 [INFO] MainProcess generic.connect() #40: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-11-26 13:47:16,218 [INFO] MainProcess generic.put() #50: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231125/allsky-timelapse_ccd1_20231125_night.mp4
2023-11-26 13:47:21,602 [ERROR] MainProcess uploader.processUpload() #369: Connection failure: ('Connection aborted.', TimeoutError('The write operation timed out'))
2023-11-26 13:47:21,616 [WARNING] MainProcess miscDb.addNotification() #719: Not adding existing notification
2023-11-26 13:47:21,624 [INFO] MainProcess generic.connect() #40: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-11-26 13:47:21,624 [INFO] MainProcess generic.put() #50: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231125/allsky-timelapse_ccd1_20231125_day.mp4
2023-11-26 13:47:27,050 [ERROR] MainProcess uploader.processUpload() #369: Connection failure: ('Connection aborted.', TimeoutError('The write operation timed out'))
2023-11-26 13:47:27,070 [WARNING] MainProcess miscDb.addNotification() #719: Not adding existing notification
2023-11-26 13:47:27,078 [INFO] MainProcess generic.connect() #40: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-11-26 13:47:27,078 [INFO] MainProcess generic.put() #50: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231124/allsky-timelapse_ccd1_20231124_night.mp4
2023-11-26 13:47:32,435 [ERROR] MainProcess uploader.processUpload() #369: Connection failure: ('Connection aborted.', TimeoutError('The write operation timed out'))
2023-11-26 13:47:32,442 [WARNING] MainProcess miscDb.addNotification() #719: Not adding existing notification
```

aaronwmorris, 2023-11-26:
I am actually working on this now.  Increase the `timeout` setting for your syncapi requests to something reasonable for uploading your videos.  I am going to be increasing the default to 60 seconds.

aaronwmorris, 2023-11-26:
Merged #1042 which separates the connection timeout from the read timeouts.

The default connection timeout is 10s and the read timeout is 60s.

closed 2023-12-17 by keno242