aaronwmorris/indi-allsky #1045: Remote camera not working: Cannot identify /dev/video0: 2, No such file or directory

bchadwiger, 2023-11-27:
I tried to access a remote camera connected to a Raspberry Pi Zero running buster 32 bit, in the same network . The camera exists (found using `lsusb`), however, I don't get images from the camera receiving the following error: 

`Error: unable to open device /dev/video0: Cannot identify /dev/video0: 2, No such file or directory `

See log for more information:

```
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread miscDb.addNotification() #733: Added State notification: 7
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread allsky._fileUploadWorkerStop() #539: Stopping Upload worker
Nov 28 00:21:26 raspberrypi [WARNING] MainProcess-4901/Upload-2 uploader.saferun() #117: Goodbye
Nov 28 00:21:26 raspberrypi [WARNING] MainProcess-4901/Upload-1 uploader.saferun() #117: Goodbye
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread allsky._fileUploadWorkerStop() #539: Stopping Upload worker
Nov 28 00:21:26 raspberrypi [WARNING] Video-1-4915/MainThread video.saferun() #173: Goodbye
Nov 28 00:21:26 raspberrypi [WARNING] Video-1-4915/MainThread video.sigterm_handler_worker() #127: Caught TERM signal
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread allsky._stopVideoWorker() #464: Stopping Video worker
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread allsky._stopVideoWorker() #461: Terminating Video worker
Nov 28 00:21:26 raspberrypi [WARNING] Image-1-4914/MainThread image.saferun() #255: Goodbye
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread allsky._stopImageWorker() #403: Stopping Image worker
Nov 28 00:21:26 raspberrypi [WARNING] Image-1-4914/MainThread image.sigterm_handler_worker() #208: Caught TERM signal
Nov 28 00:21:26 raspberrypi [INFO] MainProcess-4901/MainThread allsky._stopImageWorker() #400: Terminating Image worker
Nov 28 00:21:21 raspberrypi [WARNING] Capture-29-5057/MainThread capture.sigterm_handler_worker() #149: Caught TERM signal
Nov 28 00:21:21 raspberrypi [INFO] MainProcess-4901/MainThread allsky._stopCaptureWorker() #324: Stopping Capture worker
Nov 28 00:21:21 raspberrypi [INFO] MainProcess-4901/MainThread allsky._stopCaptureWorker() #321: Terminating Capture worker
Nov 28 00:21:21 raspberrypi [WARNING] MainProcess-4901/MainThread allsky.run() #557: Shutting down
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.configureDevice() #704: Setting property (number) TELESCOPE_INFO
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/Dummy-3 indi.newMessage() #333: new Message 2023-11-27T23:21:16: [INFO] Observer location updated: Latitude 33:00:00.0 (33.00) Longitude -84:00:00.0 (-84.00) 
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.set_number() #1291: Setting Lon (dd:mm:ss.s) = -84.0
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.set_number() #1291: Setting Lat (dd:mm:ss.s) = 33.0
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.configureDevice() #704: Setting property (number) GEOGRAPHIC_COORD
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.set_switch() #1319: Enabling Off (TRACK_OFF)
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.configureDevice() #699: Setting switch TELESCOPE_TRACK_STATE
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.set_switch() #1319: Enabling Max (4x)
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.configureDevice() #699: Setting switch TELESCOPE_SLEW_RATE
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/Dummy-3 indi.newMessage() #333: new Message 2023-11-27T23:21:16: [ERROR] Error: unable to open device /dev/video0: Cannot identify /dev/video0: 2, No such file or directory 
Nov 28 00:21:16 raspberrypi [WARNING] Capture-29-5057/MainThread capture._initialize() #489: Connecting to Telescope device Telescope Simulator
Nov 28 00:21:16 raspberrypi [WARNING] Capture-29-5057/MainThread capture._initialize() #485: Connecting to CCD device V4L2 CCD
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.findGps() #671: Found 0 GPSs
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi._findGpss() #651: Searching for available GPS interfaces
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.findTelescope() #638: Found 1 Telescopess
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi._findTelescopes() #629:  Detected Telescope Simulator
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi._findTelescopes() #618: Searching for available telescopes/mounts
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi.findCcd() #596: Found 1 CCDs
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi._findCcds() #587:  Detected V4L2 CCD
Nov 28 00:21:16 raspberrypi [INFO] Capture-29-5057/MainThread indi._findCcds() #576: Searching for available cameras
```

On the server, running `ls /dev/video*` returns around 10 - 15 results, however, all number >= 10, i.e. there is indeed no `/dev/video0`. Is is possible to specify a different `/dev/videoX` ? Or am I missing something else?

I ran `setup.sh` with `indi` and `v4l2`.

aaronwmorris, 2023-11-28:
What camera are you using?

bchadwiger, 2023-11-28:
Canon EOS 300D

aaronwmorris, 2023-11-28:
You should re-setup the indiserver to use the `indi_canon_ccd` server and restart the indiserver.

bchadwiger, 2023-11-28:
Do you know how to tell indiserver to use this driver when starting `indi-allsky`?

Also, it seems like this driver is not installed:
Running `indiserver indi_canon_ccd` yields
Driver indi_canon_ccd: No such file or directory

bchadwiger, 2023-11-28:
Yes, I built it from source. Honestly I tought the ccd drivers were included in indi-core, but I'll try your suggestion, thank you very much!

aaronwmorris, 2023-11-28:
Once you get the 3rd party stuff compiled make sure you set the custom INDI config for DSLRs otherwise it will fail to capture.

https://github.com/aaronwmorris/indi-allsky/wiki/INDI-custom-config#dslr

aaronwmorris, 2024-01-05:
Closing this issue for now.  Let me know if you run into other problems.

closed 2024-01-05 by aaronwmorris