aaronwmorris/indi-allsky #1049: timeouts after upgrade to latest

xsnrg, 2023-12-04:
My camera recently stopped working, so as part of getting it going again, I figured I would upgrade the software to play with some cool new features. 

After running the update, I needed to manually update `libxisf` for some elements missing in indi_asi_ccd, and then everything indi started. 

My remaining error that seems harder to track down is 

```
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'indi_allsky.exceptions.TimeOutException'>
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #36: Value: Timeout finding control TELESCOPE_INFO
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/allsky.py", line 119, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/indi_allsky/allsky.py", line 1026, in run\n    self._initialize()\n'
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/indi_allsky/allsky.py", line 577, in _initialize\n    self.indiclient.configureTelescopeDevice(telescope_config)\n'
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/indi_allsky/camera/indi.py", line 724, in configureTelescopeDevice\n    self.configureDevice(self._telescope_device, *args, **kwargs)\n'
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/indi_allsky/camera/indi.py", line 703, in configureDevice\n    self.set_number(device, k, v)\n'
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/indi_allsky/camera/indi.py", line 1230, in set_number\n    c = self.get_control(device, name, \'number\')\n'
Dec  4 12:00:55 rpi4 [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "indi-allsky/indi_allsky/camera/indi.py", line 1221, in get_control\n    raise TimeOutException(\'Timeout finding control {0}\'.format(name))\n'
```
Any help appreciated

aaronwmorris, 2023-12-04:
Looks like some telescope attributes were removed in indi 2.0.5.

Merged #1051 to fix.

xsnrg, 2023-12-04:
pulling... updating... fixed. Thanks!

closed 2023-12-07 by aaronwmorris