aaronwmorris/indi-allsky #1056: Oracle Cloud Infrastructure (OCI) Support

tjzepf, 2023-12-11:
I'd like to build on the AWS support and make the changes necessary to host object storage, at least, but ideally a remote website in OCI. I've had success with hosting the thomasjacquin/allsky website in OCI and fitting it within the "free tier" compute and storage resources. It costs just a few pennies a month to support the object storage transfer requests above the free tier limit.

aaronwmorris, 2023-12-11:
The OCI object storage API does not look very difficult to use.  I will take a look at this.

aaronwmorris, 2023-12-13:
Merged #1060 to support Oracle OCI for object storage

aaronwmorris, 2024-01-05:
I am going close this issue for now.  Please re-open if you find any problems.

tjzepf, 2024-01-05:
Aaron, this is great and work well from my initial testing. I'm deploying this into the cloud as well as on a mini-pc at home. Is there any chance you can add a flag for Instance Principal authentication for when deployed in the cloud? It's safer as no credentials need storing on the filesystem, etc. You can see some documentation here:

- https://docs.oracle.com/en-us/iaas/Content/API/Concepts/sdk_authentication_methods.htm
- https://github.com/oracle/oci-python-sdk/blob/master/examples/instance_principals_examples.py

Thank you for the support!

aaronwmorris, 2024-01-06:
This authentication would be for the object storage authorization?

tjzepf, 2024-01-06:
Yes for Object Storage it would be nice to have. It's a little work when launching a compute instance in OCI, but the shortish story is you create a dynamic group for the instance(s), write a policy allowing access to object storage (and any other cloud services you need),  and then the compute instance can call OCI services with ephemeral creds passed to it via the hypervisor underneath.

I believe AWS calls these Instance Profile credentials, but same idea - ephemeral creds managed buy the cloud provider so no sensitive information has to be stored on disk.

It's annoying that you can't create an entry in the Credentials File which indicates Instance Principal should be used instead of keys, but OCI doesn't give you a way to do that.

tjzepf, 2024-01-06:
I think I might be misunderstanding how Object Storage and the Sync API interact though. Is it possible to setup a remote server, use the SyncAPI to get data to it, but still have these sync-ed images land in Object Storage. I tried that, and it looks like the images are getting written to /var/www/html/allsky/images/...

aaronwmorris, 2024-01-06:
I am understanding a little better here.  You should not need to use the Instance Principal authentication.  indi-allsky does not interact with Object Storage unless it is uploading assets.  Downloading happens directly from the users browser.  The Object Storage bucket should be set to Public access for the assets.

In the SyncAPI section, there are two options `Sync after S3 Upload` and `Sync empty file`.

`Sync after S3 Upload` will perform the SyncAPI upload after the Object Storage upload has completed.  This ensures the remote DB entry has the reference to the Object Storage assets.

`Sync empty file` will only upload the file once to Object Storage, but the SyncAPI request will only contain metadata and no file data, which is much smaller and faster.

In the Admin tab, you can then enable `Non-Local Images` to force serving images and videos from Object Storage.  Consider this option on both your local and remote instances.

On your remote instance, you will also need to enter the correct `URL Template` in the Object Storage configuration so it sends the correct URL to the browser to access the assets.

closed 2024-04-28 by aaronwmorris