aaronwmorris/indi-allsky #1059: syncapi timelapse video transfers failing but images OK even with 900s timeouts

keno242, 2023-12-13:
Image transfers are working great to replicate allsky data from a remote site (local camera) to a local instance (no local camera) , however video transfers seem to always fail and thus Timelapse's are unavailable on the local allsky instance.

```
2023-12-13T14:48:49.247354+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/image (443) as foobar123 with requests_syncapi_v1
2023-12-13T14:48:49.247823+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/day/13_14/ccd1_20231213_144848.jpg
2023-12-13T14:48:50.091185+08:00 allsky [INFO] MainProcess-2573/Upload-8 requests_syncapi_v1.put() #172: File transferred in 0.8437 s (604.15 kB/s)
2023-12-13T14:48:50.096726+08:00 allsky [INFO] MainProcess-2573/Upload-8 uploader.processUpload() #442: Upload transaction completed in 0.8502 s
```

I've set both timeout settings as 900secs. Usual night timelapse mp4 is around 100MB (for Aurora capture) and scp'ing the file manually between systems takes around 2 mins..

```
allsky-timelapse_ccd1_20231212_night.mp4                                                                                                                                                                                          100%   96MB 928.7KB/s   01:45
```

images mostly appear to transfer fine

```
2023-12-13T15:07:52.219659+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/image (443) as foobar123 with requests_syncapi_v1
2023-12-13T15:07:52.220056+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/day/13_15/ccd1_20231213_150751.jpg
2023-12-13T15:07:53.131416+08:00 allsky [INFO] MainProcess-2573/Upload-8 requests_syncapi_v1.put() #172: File transferred in 0.9116 s (554.39 kB/s)
2023-12-13T15:07:53.137212+08:00 allsky [INFO] MainProcess-2573/Upload-8 uploader.processUpload() #442: Upload transaction completed in 0.9182 s
2023-12-13T15:08:07.199810+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/image (443) as foobar123 with requests_syncapi_v1
2023-12-13T15:08:07.200175+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/day/13_15/ccd1_20231213_150806.jpg
2023-12-13T15:08:08.031462+08:00 allsky [INFO] MainProcess-2573/Upload-8 requests_syncapi_v1.put() #172: File transferred in 0.8314 s (607.96 kB/s)
2023-12-13T15:08:08.037921+08:00 allsky [INFO] MainProcess-2573/Upload-8 uploader.processUpload() #442: Upload transaction completed in 0.8385 s
2023-12-13T15:08:22.347140+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/image (443) as foobar123 with requests_syncapi_v1
2023-12-13T15:08:22.347642+08:00 allsky [INFO] MainProcess-2573/Upload-8 generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/day/13_15/ccd1_20231213_150821.jpg
2023-12-13T15:08:22.955896+08:00 allsky [INFO] MainProcess-2573/Upload-8 requests_syncapi_v1.put() #172: File transferred in 0.6087 s (829.39 kB/s)
2023-12-13T15:08:22.961394+08:00 allsky [INFO] MainProcess-2573/Upload-8 uploader.processUpload() #442: Upload transaction completed in 0.6150 s
```

I get a few 400 errors throughout the day and in the UI on, but logs don't reveal much.

```
cat /var/log/indi-allsky/indi-allsky.log | grep fail
2023-12-13T00:25:00.576993+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T01:15:19.153800+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 502
2023-12-13T01:40:01.434057+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T03:40:02.841513+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T03:50:00.137432+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T04:48:23.252186+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T05:00:19.027633+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T05:50:00.706699+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T06:00:02.056435+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T07:40:01.554521+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T07:45:01.182120+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T09:15:00.960849+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T11:05:00.672337+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13T12:45:01.070195+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
```

when I've ran the manual sync 

I see it attempt to PUT the video on the apache logs

```
sudo cat /var/log/apache2/other_vhosts_access.log  | grep /indi-allsky/sync/v1/video
127.0.1.1:443 X.X.X.X - - [13/Dec/2023:04:44:00 +0800] "PUT /indi-allsky/sync/v1/video HTTP/1.1" 400 2978 "-" "python-requests/2.31.0"
```

I experience the same problems sometimes when using `upload_sync.py`

```
~/indi-allsky/misc$ python upload_sync.py --no-syncapi-images sync
2023-12-13 16:00:11,178 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbVideoTable uploading disabled
2023-12-13 16:00:11,179 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbKeogramTable uploading disabled
2023-12-13 16:00:11,179 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbStarTrailsTable uploading disabled
2023-12-13 16:00:11,179 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbStarTrailsVideoTable uploading disabled
2023-12-13 16:00:11,179 [INFO] MainProcess upload_sync._get_entry_status() #568: S3 uploading disabled (IndiAllSkyDbVideoTable)
[... 25 lines cut ...]
+---------+----------------------------------+----------+---------+
2023-12-13 16:00:18,287 [INFO] MainProcess upload_sync.sync() #162: Entries to upload: 247
2023-12-13 16:00:18,289 [INFO] MainProcess upload_sync.addUploadEntries() #205: Adding 7 upload entries (240 remaining)
2023-12-13 16:00:19,420 [INFO] MainProcess upload_sync._fileUploadWorkerStart() #720: Starting FileUploader process 1
2023-12-13 16:00:19,447 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-13 16:00:19,447 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231212/allsky-timelapse_ccd1_20231212_night.mp4
2023-12-13 16:02:21,007 [INFO] MainProcess requests_syncapi_v1.put() #172: File transferred in 121.5589 s (804.75 kB/s)
2023-12-13 16:02:21,027 [INFO] MainProcess uploader.processUpload() #442: Upload transaction completed in 121.5805 s
2023-12-13 16:02:21,047 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-13 16:02:21,047 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231212/allsky-timelapse_ccd1_20231212_day.mp4
2023-12-13 16:09:57,754 [ERROR] MainProcess uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-13 16:09:57,768 [WARNING] MainProcess miscDb.addNotification() #719: Not adding existing notification
[... cut ...]
2023-12-13 16:18:21,023 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-13 16:18:21,023 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231210/allsky-timelapse_ccd1_20231210_night.mp4
```

I suspect that is a robustness of handling wan type comms with the requests library.

Could we have a built in retry or is there another protocol we can use to keep in sync?

Happy to provide any other logs or try things out - once this is resolved I'll be able to share my allsky with the world!

aaronwmorris, 2023-12-13:
Is the remote site a standard install using Apache and gunicorn?  Are there any interesting errors in the web logs at `/var/log/indi-allsky/webapp-indi-allsky.log`?

Do you have sufficient space available in `/tmp` on the remote server?  When the server downloads the file, it temporarily stores the file in /tmp before copying the file to the final location.

keno242, 2023-12-14:
> Is the remote site a standard install using Apache and gunicorn? Are there any interesting errors in the web logs at `/var/log/indi-allsky/webapp-indi-allsky.log`?
>
> Do you have sufficient space [...]

Yes, the remote site is a dedicated install of indi-allsky running on a nas container with plenty of space here is the support dump

```
sudo ./support_info.sh 
#################################
### indi-allsky support info  ###
#################################

Distribution: Ubuntu
Release: 22.04
Arch: x86_64
Bits: 64

CPUs: 6
Memory: 16287720 kB

Uptime
 09:23:24 up 9 days, 14:50,  2 users,  load average: 3.54, 2.78, 1.79

Filesystems
Filesystem              1K-blocks        Used  Available Use% Mounted on
/dev/mapper/cachedev1 32759190536 27816864380 4941785484  85% /
none                          492           4        488   1% /dev
tmpfs                         100           0        100   0% /dev/lxd
tmpfs                         100           0        100   0% /dev/.lxd-mounts
tmpfs                     8143860         172    8143688   1% /dev/shm
tmpfs                     3257544         180    3257364   1% /run
tmpfs                        5120           0       5120   0% /run/lock
tmpfs                        4096           0       4096   0% /sys/fs/cgroup
tmpfs                     1628772           0    1628772   0% /run/user/1001

system python: Python 3.10.12

indiserver: /usr/bin/indiserver

User info
uid=0(root) gid=0(root) groups=0(root)

Process info
khall      200 15.9  0.1 107420 25352 ?        Ss   Dec13 166:53 /usr/bin/indiserver -p 7624 indi_simulator_telescope indi_webcam_ccd
khall      201  0.0  0.0  40088  1688 ?        S    Dec13   0:13 indi_simulator_telescope
khall      237  0.0  0.0  36080  2820 ?        SNs  Dec13   0:03 /home/khall/indi-allsky/virtualenv/indi-allsky/bin/python /home/khall/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
khall      239  0.1  0.3 831060 58076 ?        SNl  Dec13   1:18 /home/khall/indi-allsky/virtualenv/indi-allsky/bin/python /home/khall/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
khall      255  0.0  0.2 764156 43340 ?        SNsl Dec13   0:07 /home/khall/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
khall      263 10.2  0.5 928256 89636 ?        Sl   Dec13 107:01 /home/khall/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
khall      264  0.3  0.7 914924 118368 ?       SNl  Dec13   3:37 /home/khall/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
khall     7248  4.6 44.4 12799968 7232144 ?    SNl  01:19  22:17 /home/khall/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
khall    12443  0.0  0.1 1049336 24040 ?       SLl  07:29   0:01 indi_webcam_ccd

USB info
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 002: ID 1005:b155 Apacer Technology, Inc. Disk Module
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Module info
Module                  Size  Used by
xt_ipvs                16384  0
ip_vs_rr               16384  0
ip_vs_ftp              16384  0
ip_vs                 139264  6 ip_vs_rr,xt_ipvs,ip_vs_ftp
xt_nat                 16384  0
xt_addrtype            16384  6
kvm_intel             225280  0
kvm                   516096  1 kvm_intel
[... 302 lines cut ...]
Dec 14 05:55:01 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 14 07:35:02 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 14 07:39:59 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
```

```
Dec 14 04:56:22 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045620.jpg
Dec 14 04:56:38 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045635.jpg
Dec 14 04:56:52 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045650.jpg
Dec 14 04:57:06 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045705.jpg
Dec 14 04:57:21 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045720.jpg
Dec 14 04:57:22 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #239: Uploaded file: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/allsky-keogram_ccd1_20231213_night.jpg
[... cut ...]
Dec 14 05:01:47 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045920.jpg
Dec 14 05:01:49 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045935.jpg
Dec 14 05:01:50 allskynas [INFO] MainProcess-239 syncapi_views.processPost() #502: Uploaded image: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/day/14_04/ccd1_20231214_045950.jpg
```

keno242, 2023-12-14:
Ahh - I looked at the logs on the camera connected allsky and the timing of these entries match perfectly... so it's related to Authentication? 

```
~$ cat /var/log/indi-allsky/indi-allsky.log | grep MainProcess-2573 | grep fail
2023-12-14T01:35:01.975774+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T03:55:00.880859+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T04:30:05.230182+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T04:49:21.745002+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T05:01:39.249530+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T05:50:02.293152+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T05:55:03.034341+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T07:35:03.637277+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
2023-12-14T07:40:01.510468+08:00 allsky [ERROR] MainProcess-2573/Upload-8 uploader.processUpload() #411: Tranfer failure: Sync error: 400
```

aaronwmorris, 2023-12-14:
I might know what is going on.  The syncapi transfers are validated based on the HMAC signature of metadata of each file.  The metadata also includes a timestamp, so even if the HMAC signature were somehow recovered, it would only be valid during a 15 minute window:  the current 5 minute block, -5 minutes, and +5 minutes to account for time skew between the local and remote indi-allsky systems.

The HMAC signature is based on the **START TIME** of the transfer and if the transfer takes longer than 5-10 minutes, you could easily start to get API key validation failures (depending on the time skew).  100% of transfers longer than 10 minutes would probably fail.

Give me a day or two to release an update.  I have an existing code change that I am testing and I need to complete that first.

aaronwmorris, 2023-12-14:
I have been reviewing my code and the API authentication is supposed to happen at the beginning of the connection and not the end.  It does not make sense that the API authentication would fail near the end, unless the beginning of the server transaction does not occur until all of the files have been uploaded.  I am still investigating.

aaronwmorris, 2023-12-14:
I think I have confirmed that the processing on the server side does not start until the file upload has completed, since I am not using streamed uploads.  So that makes sense.

aaronwmorris, 2023-12-14:
Merged #1065 to fix.

I am not sure you are on the latest code, but recently I also merged an update to separate the connect timeout from the read timeout.

keno242, 2023-12-15:
Yes I was using the version with both timeouts - but also just upgraded both systems to 7.8 to test the latest code.

Same issue I'm afraid - I did try disabling syncapi for images to check if maybe starting a new image transfer, while a video one was still in progress might be causing issues but it made no difference. 

```
python upload_sync.py --no-syncapi-images sync
2023-12-15 12:59:57,979 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbVideoTable uploading disabled
2023-12-15 12:59:57,980 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbKeogramTable uploading disabled
2023-12-15 12:59:57,980 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbStarTrailsTable uploading disabled
2023-12-15 12:59:57,980 [INFO] MainProcess upload_sync._get_entry_status() #537: IndiAllSkyDbStarTrailsVideoTable uploading disabled
2023-12-15 12:59:57,980 [INFO] MainProcess upload_sync._get_entry_status() #568: S3 uploading disabled (IndiAllSkyDbVideoTable)
2023-12-15 12:59:57,980 [INFO] MainProcess upload_sync._get_entry_status() #568: S3 uploading disabled (IndiAllSkyDbKeogramTable)
2023-12-15 12:59:57,981 [INFO] MainProcess upload_sync._get_entry_status() #568: S3 uploading disabled (IndiAllSkyDbStarTrailsTable)
2023-12-15 12:59:57,981 [INFO] MainProcess upload_sync._get_entry_status() #568: S3 uploading disabled (IndiAllSkyDbStarTrailsVideoTable)
2023-12-15 12:59:57,981 [INFO] MainProcess upload_sync._get_entry_status() #568: S3 uploading disabled (IndiAllSkyDbImageTable)
2023-12-15 12:59:57,981 [INFO] MainProcess upload_sync._get_entry_status() #578: S3 uploading disabled (IndiAllSkyDbFitsImageTable)
2023-12-15 12:59:57,981 [INFO] MainProcess upload_sync._get_entry_status() #588: S3 uploading disabled (IndiAllSkyDbRawImageTable)
2023-12-15 12:59:57,988 [INFO] MainProcess upload_sync._get_entry_status() #616: syncapi disabled (IndiAllSkyDbImageTable)
+---------+----------------------------------+----------+---------+
|   Type  |              Table               | Uploaded | Missing |
+---------+----------------------------------+----------+---------+
|  upload |      IndiAllSkyDbVideoTable      |    -     |    -    |
|  upload |     IndiAllSkyDbKeogramTable     |    -     |    -    |
|  upload |   IndiAllSkyDbStarTrailsTable    |    -     |    -    |
|  upload | IndiAllSkyDbStarTrailsVideoTable |    -     |    -    |
|    s3   |      IndiAllSkyDbVideoTable      |    -     |    -    |
|    s3   |     IndiAllSkyDbKeogramTable     |    -     |    -    |
|    s3   |   IndiAllSkyDbStarTrailsTable    |    -     |    -    |
|    s3   | IndiAllSkyDbStarTrailsVideoTable |    -     |    -    |
|    s3   |      IndiAllSkyDbImageTable      |    -     |    -    |
|    s3   |    IndiAllSkyDbFitsImageTable    |    -     |    -    |
|    s3   |    IndiAllSkyDbRawImageTable     |    -     |    -    |
| syncapi |      IndiAllSkyDbVideoTable      |    48    |    52   |
| syncapi |     IndiAllSkyDbKeogramTable     |    30    |    67   |
| syncapi |   IndiAllSkyDbStarTrailsTable    |    19    |    69   |
| syncapi | IndiAllSkyDbStarTrailsVideoTable |    13    |    64   |
| syncapi |      IndiAllSkyDbImageTable      |    -     |    -    |
+---------+----------------------------------+----------+---------+
2023-12-15 13:00:03,121 [INFO] MainProcess upload_sync.sync() #162: Entries to upload: 252
2023-12-15 13:00:03,122 [INFO] MainProcess upload_sync.addUploadEntries() #205: Adding 7 upload entries (245 remaining)
2023-12-15 13:00:03,383 [INFO] MainProcess upload_sync._fileUploadWorkerStart() #720: Starting FileUploader process 1
2023-12-15 13:00:03,406 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-15 13:00:03,406 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/allsky-timelapse_ccd1_20231214_night.mp4
2023-12-15 13:01:46,549 [INFO] MainProcess requests_syncapi_v1.put() #172: File transferred in 103.1413 s (953.62 kB/s)
2023-12-15 13:01:46,568 [INFO] MainProcess uploader.processUpload() #443: Upload transaction completed in 103.1617 s
2023-12-15 13:01:46,587 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-15 13:01:46,587 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/allsky-timelapse_ccd1_20231214_day.mp4
2023-12-15 13:07:49,641 [ERROR] MainProcess uploader.processUpload() #412: Tranfer failure: Sync error: 400
2023-12-15 13:07:49,654 [WARNING] MainProcess miscDb.addNotification() #719: Not adding existing notification
2023-12-15 13:07:49,707 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-15 13:07:49,707 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/allsky-timelapse_ccd1_20231213_night.mp4
^C2023-12-15 13:08:27,244 [WARNING] MainProcess upload_sync.sigint_handler_main() #119: Caught INT signal, shutting down
```

```
cat /var/log/indi-allsky/indi-allsky.log | grep fail
[... cut ...]
Dec 15 11:39:59 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 15 12:46:13 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 15 12:53:22 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 15 13:07:48 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
```

aaronwmorris, 2023-12-15:
Just for giggles, can you check to see if there is a time delta between both the local and remote servers?

keno242, 2023-12-15:
Yes did check that and rechecked now pretty much within a second of each other... 

And you will have noticed I did get one large file transfer through. I know you said you don't using the stream large files 
feature, but I've noticed other around the web having issues with large files - is it worth a shot?

aaronwmorris, 2023-12-15:
To be perfectly honest, I have been having trouble understanding the full implications of streaming the uploads.  It is difficult to find examples of what I am trying.  Most examples only use one file, but the syncapi transfers are actually two files each:  one file is a json file containing metadata and the second file is the actual image/movie.

I have made a first pass at this, but I do not expect this to really stream the way we need it to.

Merged #1067

keno242, 2023-12-16:
Thanks for having a go at this... 

But still getting ...

```
2023-12-16 09:30:05,166 [INFO] MainProcess upload_sync._fileUploadWorkerStart() #720: Starting FileUploader process 1
2023-12-16 09:30:05,190 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-16 09:30:05,190 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231215/allsky-timelapse_ccd1_20231215_day.mp4
2023-12-16 09:36:30,554 [ERROR] MainProcess uploader.processUpload() #412: Tranfer failure: Sync error: 400
2023-12-16 09:36:30,569 [WARNING] MainProcess miscDb.addNotification() #719: Not adding existing notification
2023-12-16 09:36:30,622 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-16 09:36:30,622 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/allsky-timelapse_ccd1_20231214_day.mp4
2023-12-16 09:43:42,050 [ERROR] MainProcess uploader.processUpload() #412: Tranfer failure: Sync error: 400
```

with corresponding Authentication error on the remote (non camera) site.

```
ec 16 08:59:59 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 16 09:04:59 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 16 09:25:27 allskynas [ERROR] MainProcess-239 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 16 09:36:28 allskynas [ERROR] MainProcess-273 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
Dec 16 09:43:40 allskynas [ERROR] MainProcess-273 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
```

Not sure if this is relevant but it references 'We have a web application which accepts (large) files together with some meta data' https://github.com/psf/requests/issues/1584 and the recommendation is to use this https://github.com/requests/toolbelt 

However - I'm open to trying to achieve what I want in another way - basically a local (to me) / public instance (with a fast internet connection) of the allsky data (images and timelapses)  taken from a remote observatory. I could even manually (via python trigger a scp file copy of the videos as long as I knew how to update the metadata on the other send tell it has a new file.

aaronwmorris, 2023-12-16:
Interesting, I will read about the requests toolbelt.

aaronwmorris, 2023-12-16:
I made another commit to extend the HMAC signature forward in time.  The previous just extended backwards.  I do not think this will fix it, but maybe I had my thinking backwards.  Let me know.

Merged #1068

keno242, 2023-12-16:
Yea same issue... after updating on both systems...

```
+---------+----------------------------------+----------+---------+
2023-12-16 11:22:21,888 [INFO] MainProcess upload_sync.sync() #162: Entries to upload: 20
2023-12-16 11:22:21,889 [INFO] MainProcess upload_sync.addUploadEntries() #205: Adding 7 upload entries (13 remaining)
2023-12-16 11:22:22,156 [INFO] MainProcess upload_sync._fileUploadWorkerStart() #720: Starting FileUploader process 1
2023-12-16 11:22:22,181 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-16 11:22:22,181 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231215/allsky-timelapse_ccd1_20231215_day.mp4
2023-12-16 11:29:02,500 [ERROR] MainProcess uploader.processUpload() #412: Tranfer failure: Sync error: 400
```
```
Dec 16 11:29:00 allskynas [ERROR] MainProcess-287 syncapi_views.dispatch_request() #70: Authentication failure: Unable to authenticate API key
```

aaronwmorris, 2023-12-16:
Well, I feel kind of silly now.  This ended up just being a simple typo.  I setup the testing the variables to test the HMAC signature if transfers took longer than 5 minutes (even before all of these changes), but they were never actually used.

```diff
        time_floor = math.floor(time.time() / self.time_skew)

        # the time on the remote system needs to be plus/minus the time_floor period
        time_floor_list = [
            time_floor,
            time_floor - 1,
            time_floor + 1,
            time_floor - 2,  # large file uploads my take a long time
            time_floor - 3,
            time_floor - 4,
        ]

        for t in time_floor_list:
-            hmac_message = str(time_floor).encode() + data
+            hmac_message = str(t).encode() + data
```

keno242, 2023-12-16:
BOOM! Nice work Aaron - 3 out of 3 ran fine - thanks for persisting with this! 

```
2023-12-16 17:18:20,408 [INFO] MainProcess upload_sync.sync() #162: Entries to upload: 20
2023-12-16 17:18:20,410 [INFO] MainProcess upload_sync.addUploadEntries() #205: Adding 7 upload entries (13 remaining)
2023-12-16 17:18:20,688 [INFO] MainProcess upload_sync._fileUploadWorkerStart() #720: Starting FileUploader process 1
2023-12-16 17:18:20,712 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-16 17:18:20,712 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231215/allsky-timelapse_ccd1_20231215_day.mp4
2023-12-16 17:24:26,835 [INFO] MainProcess requests_syncapi_v1.put() #179: File transferred in 366.1209 s (959.61 kB/s)
2023-12-16 17:24:26,838 [INFO] MainProcess uploader.processUpload() #443: Upload transaction completed in 366.1262 s
2023-12-16 17:24:26,861 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-16 17:24:26,861 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/allsky-timelapse_ccd1_20231214_day.mp4
2023-12-16 17:34:05,369 [INFO] MainProcess requests_syncapi_v1.put() #179: File transferred in 578.5065 s (616.36 kB/s)
2023-12-16 17:34:05,374 [INFO] MainProcess uploader.processUpload() #443: Upload transaction completed in 578.5132 s
2023-12-16 17:34:05,389 [INFO] MainProcess generic.connect() #50: Connecting to https://allskynas.discus-ray.ts.net/indi-allsky/sync/v1/video (443) as foobar123 with requests_syncapi_v1
2023-12-16 17:34:05,389 [INFO] MainProcess generic.put() #60: Uploading /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/allsky-timelapse_ccd1_20231213_day.mp4
2023-12-16 17:44:34,481 [INFO] MainProcess requests_syncapi_v1.put() #179: File transferred in 629.0906 s (570.91 kB/s)
2023-12-16 17:44:34,486 [INFO] MainProcess uploader.processUpload() #443: Upload transaction completed in 629.0963 s
```

```
Dec 16 17:24:23 allskynas [INFO] MainProcess-267 syncapi_views.processPost() #242: Uploaded file: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231215/allsky-timelapse_ccd1_20231215_day.mp4
Dec 16 17:34:02 allskynas [INFO] MainProcess-267 syncapi_views.processPost() #242: Uploaded file: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231214/allsky-timelapse_ccd1_20231214_day.mp4
Dec 16 17:44:31 allskynas [INFO] MainProcess-267 syncapi_views.processPost() #242: Uploaded file: /var/www/html/allsky/images/ccd_6440f61c-0c55-47e9-bb50-d3c7cadb5e8e/20231213/allsky-timelapse_ccd1_20231213_day.mp4
```

closed 2023-12-17 by keno242