aaronwmorris/indi-allsky #1074: Restarting pi services

Adrien5353, 2023-12-19:
Hi, and thank you for this great software .
I installed it on a pi4 8G, it works flawlessly.
I now use a small script to retrieve the states of my I2C sensors so that an ascom client can retrieve this data (safety monitor for observatory).
My script consumes very few resources and very little I/O .
I am facing a problem, when indi allsky is running, I have a reboot of the services: indiallsky is unreachable for a few seconds, and so is my ascom server, and I have to restart the connection from putty.So I lose the connection for the safety monitor (indi allsky restarts by itself). 
After several tests, this does not happen if I disable indi allsky. 
In both cases, the I/O wait rate is low, so I can't put my finger on the service that consumes too much.I clearly saw that you specified that Indi Allsky runs without problems on a pi3b+, so I'm surprised to encounter this problem with a pi4 . 
Are there any options to disable or anything to optimize? I installed all of this on a USB 3.0 key which is normally quite fast. 

I ideally wanted to have only one pi running to centralize everything and not leave the observatory computer running 24/7. 
Where do you start to determine what is causing these repeated crashes? What part of the log will be interesting in this case?

I'm running a fresh install of debian bullseye. Indi and indi allsky installed via the procedure in Doc.

Thanks in advance 
Kind regards, Adrien

aaronwmorris, 2023-12-19:
How are you restarting the services?

Adrien5353, 2023-12-19:
Allsky starts its services on its own, and I put my script as a service on the pi, loaded AT the boot.
When the crash occurs, allsky restarts itself after a few seconds. I don't act on anything. 
In fact, I cannot determine if it is the allsky assembly which requisitions all the resources or if my problem is elsewhere

aaronwmorris, 2023-12-19:
Can you describe the crash?  What behavior indicates a crash?

Adrien5353, 2023-12-19:
The connection to the Raspberry Pi is temporarily cut off: SSH, ASCOM client, and also, I have to refresh the indi-allsky web page to get the new camera images; otherwise, it remains stuck on the last capture before the "crash". My script that collects weather sensor data is also stopped. In fact, it's as if all services were abruptly halted. Only indi-allsky resumes on its own.

However, the crash never occurs if I stop the services (from the web interface, for example). With indi-allsky running, the crash frequency is random (between 10 minutes and 1-2 hours).

sorry if it's not clear  :/

aaronwmorris, 2023-12-19:
Is the Pi rebooting itself?

Adrien5353, 2023-12-19:
no , the pi does not seems to reboot as it's really fast : few seconds later i can log in again, while it takes more time when it's a complete reboot

aaronwmorris, 2023-12-20:
That sounds really weird, I have not experienced this behavior previously.  Are you connected via ethernet or wifi?

Adrien5353, 2023-12-20:
connected with ethernet.
really frustrating :( 

i'm installing debian 11 right now on an old nuc. i would have prefered use my pi4, but i'll try this way

aaronwmorris, 2023-12-20:
What kind of camera are you using?  Could you post the I2C script?

Have you checked the logs in `/var/log/syslog`?

aaronwmorris, 2024-06-08:
Closing this for now.  Let me know if there is anything else I can do.

closed 2024-06-08 by aaronwmorris