aaronwmorris/indi-allsky #1080: Error during install of indi while upgrading indi-allsky

vastmeridian, 2023-12-29:
I'm upgrading to get the advantage of uploading to YouTube but I am getting an error during the upgrade to indi:

```
indi-playerone is already the newest version (1.12+t202310010637~ubuntu20.04.1).
indi-toupbase is already the newest version (2.1+t202309302048~ubuntu20.04.1).
indi-webcam is already the newest version (1.0+t202310010920~ubuntu20.04.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 indi-full : Depends: indi-avalonud but it is not installable
E: Unable to correct problems, you have held broken packages.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```

I can try again without updating indi, but I thought I'd bring it to your attention Aaron.

aaronwmorris, 2023-12-29:
This happens from time to time with the INDI apt repo.  I would just proceed without upgrading these packages, which would have no effect on the YouTube functionality.

vastmeridian, 2023-12-29:
Yes - the rebuild that bypassed the re-install of indi-server fixed that.  Thanks.

closed 2023-12-29 by vastmeridian

aaronwmorris, 2023-12-31:
Did you have any luck with the YouTube function?

vastmeridian, 2024-01-02:
I'll be honest, I've not tried.  Two reasons:
- Christmas visitors (still here!)
- No decent skies here in England.
As soon as I can I'll be trying it out - and I'll let you know.