aaronwmorris/indi-allsky #1085: Daytime exposure used during night

dark-nyx, 2024-01-01:
For some reason the wrong exposure is used during night:

[image]

Here are the settings used:

[image]

Any ideas why?

aaronwmorris, 2024-01-01:
You may need to fully stop the indi-allsky service and start it again to change the min/max exposure settings.

dark-nyx, 2024-01-02:
So, I did as you suggested. I restarted all 3 services, indiserver, indi-allsky and gunicorn-indi-allsky. As a matter of fact I restarted the whole Server this morning. Still same problem. Going from day to night, the exposures are way too short:

[image]

aaronwmorris, 2024-01-02:
Well, that is really weird.  I am trying to track this down.

aaronwmorris, 2024-01-02:
Have you tried updating to the latest code?  I have not found any cause for this behavior and all of the tests in my lab work as they should.

dark-nyx, 2024-01-05:
I've updated to the latest code, restarted everything and let the thing run for 2 nights. No success. The camera gain changes from 100 (day) to moon-mode (800) to night (1600) without a problem. But the exposure remains the same (0.000105s) when switching from day to night.

aaronwmorris, 2024-01-05:
I think I need some log entries to tell what is going on.  A dump of the config would be helpful as well.  You might want to supply a support dump.

```
source virtualenv/indi-allsky/bin/activate
./config.py dump

./misc/support_info.sh
```

If you are able and comfortable to grant remote access to your system via SSH, I could login and poke around, too.

dark-nyx, 2024-01-05:
OK, I believe I found the problem. As a matter of fact there is no problem. The issue was my Target ADU configuration. This is how my current configuration looks like, which looks like it works now:

[image]

I reduced the deviation so that it won't overlap between day and night.

closed 2024-01-07 by dark-nyx