aaronwmorris/indi-allsky #1092: Error: While importing 'app', an ImportError was raised:

herseyc, 2024-01-06:
When trying to run setup.sh to upgrade indi-allsky I am receiving the following:

Error: While importing 'app', an ImportError was raised:

Traceback (most recent call last):
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 219, in locate_app
    __import__(module_name)
  File "/home/pi/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/pi/indi-allsky/indi_allsky/flask/__init__.py", line 9, in <module>
    from flask_wtf.csrf import CSRFProtect
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/__init__.py", line 4, in <module>
    from .recaptcha import Recaptcha
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/__init__.py", line 1, in <module>
    from .fields import RecaptchaField
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/fields.py", line 3, in <module>
    from . import widgets
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/widgets.py", line 2, in <module>
    from flask import Markup
ImportError: cannot import name 'Markup' from 'flask' (/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/__init__.py)

Usage: flask [OPTIONS] COMMAND [ARGS]...
Try 'flask --help' for help.

Error: No such command 'db'.

Tried rebuilding the database but receive the same error when running  flask db revision --autogenerate

Thanks
Hersey

aaronwmorris, 2024-01-06:
Can you post the results of the `./misc/support_info.sh` script?

herseyc, 2024-01-06:
pi@allskypi:~/indi-allsky $ ./misc/support_info.sh
./misc/support_info.sh: line 49: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################

Distribution: Raspbian
Release: 11
Arch: armv7l
Bits: 32

CPUs: 4
Memory: 944092 kB

System: Raspberry Pi 3 Model B Rev 1.2

Uptime
 10:56:34 up 49 min,  4 users,  load average: 0.00, 0.01, 0.17

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122597728 68485472  49075648  59% /
devtmpfs          340460        0    340460   0% /dev
tmpfs             472044        0    472044   0% /dev/shm
tmpfs             188820      800    188020   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    258095    51409    206687  20% /boot
tmpfs              94408       24     94384   1% /run/user/1000

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(systemd-journal),105(input),107(render),109(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)

Process info
pi         615  0.0  0.2   8288  2080 ?        Ss   10:07   0:00 /usr/local/bin/indiserver indi_simulator_telescope indi_sv305_ccd
pi         619  0.0  0.4  32196  4364 ?        S    10:07   0:00 indi_simulator_telescope
pi         620  0.3  0.4  71516  4588 ?        Sl   10:07   0:10 indi_sv305_ccd

USB info
Bus 001 Device 004: ID f266:9a0a CKCamera CKCamera
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      122      0 drwxr-xr-x   3 root     root           60 Dec 31  1969 /dev/bus/usb
      123      0 drwxr-xr-x   2 root     root          120 May 13  2023 /dev/bus/usb/001
      142      0 crw-rw-rw-   1 root     root     189,   3 Jan  6 10:06 /dev/bus/usb/001/004
      135      0 crw-rw-r--   1 root     root     189,   2 Jan  6 10:06 /dev/bus/usb/001/003
      134      0 crw-rw-r--   1 root     root     189,   1 Jan  6 10:06 /dev/bus/usb/001/002
      124      0 crw-rw-r--   1 root     root     189,   0 Jan  6 10:06 /dev/bus/usb/001/001

Module info
Module                  Size  Used by
rfcomm                 49152  4
cmac                   16384  2
algif_hash             16384  1
aes_arm_bs             24576  2
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  2 crypto_simd
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   20480  2
hci_uart               40960  1
btbcm                  20480  1 hci_uart
bluetooth             503808  31 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           16384  1 bluetooth
[... 395 lines cut ...]
requests-oauthlib==1.3.1
requests-toolbelt==1.0.0
rsa==4.9
s3transfer==0.6.0
scikit-image==0.19.3
scipy==1.11.3
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.4.1
shapely==2.0.1
six==1.16.0
SQLAlchemy==1.4.39
tifffile==2022.5.4
typing_extensions==4.3.0
uritemplate==4.1.1
urllib3==1.26.14
wcwidth==0.2.6
Werkzeug==3.0.1
WTForms==3.0.1
zipp==3.8.0

#################################
###     end support info      ###
#################################
pi@allskypi:~/indi-allsky $

aaronwmorris, 2024-01-06:
```
ii python3-flask 1.1.2-2+deb11u1 all micro web framework based on Werkzeug and Jinja2 - Python 3.x
```

I think your local system version of flask is somehow getting precedence from the version that is installed in the indi-allsky virtualenv.

Try running `sudo apt-get remove python3-flask` and then run setup.sh.  There is still a chance there is another flask floating around on your system, so let me know

herseyc, 2024-01-06:
Same issue after removing python3-flask.  Here is the support_info.sh output:

pi@allskypi:~/indi-allsky $ ./misc/support_info.sh
./misc/support_info.sh: line 49: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################

Distribution: Raspbian
Release: 11
Arch: armv7l
Bits: 32

CPUs: 4
Memory: 944092 kB

System: Raspberry Pi 3 Model B Rev 1.2

Uptime
 12:53:28 up  2:46,  4 users,  load average: 0.79, 0.67, 0.28

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122597728 68484692  49076428  59% /
devtmpfs          340460        0    340460   0% /dev
tmpfs             472044        0    472044   0% /dev/shm
tmpfs             188820      800    188020   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    258095    51409    206687  20% /boot
tmpfs              94408       24     94384   1% /run/user/1000

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(systemd-journal),105(input),107(render),109(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)

Process info
pi         615  0.0  0.2   8288  2280 ?        Ss   10:07   0:00 /usr/local/bin/indiserver indi_simulator_telescope indi_sv305_ccd
pi         619  0.0  0.5  32196  5324 ?        S    10:07   0:00 indi_simulator_telescope
pi         620  0.3  0.5  71516  5172 ?        Sl   10:07   0:31 indi_sv305_ccd

USB info
Bus 001 Device 004: ID f266:9a0a CKCamera CKCamera
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      122      0 drwxr-xr-x   3 root     root           60 Dec 31  1969 /dev/bus/usb
      123      0 drwxr-xr-x   2 root     root          120 May 13  2023 /dev/bus/usb/001
      142      0 crw-rw-rw-   1 root     root     189,   3 Jan  6 10:06 /dev/bus/usb/001/004
      135      0 crw-rw-r--   1 root     root     189,   2 Jan  6 10:06 /dev/bus/usb/001/003
      134      0 crw-rw-r--   1 root     root     189,   1 Jan  6 10:06 /dev/bus/usb/001/002
      124      0 crw-rw-r--   1 root     root     189,   0 Jan  6 10:06 /dev/bus/usb/001/001

Module info
Module                  Size  Used by
rfcomm                 49152  4
cmac                   16384  2
algif_hash             16384  1
aes_arm_bs             24576  2
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  2 crypto_simd
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   20480  2
hci_uart               40960  1
btbcm                  20480  1 hci_uart
[... 395 lines cut ...]
requests==2.28.2
requests-oauthlib==1.3.1
requests-toolbelt==1.0.0
rsa==4.9
s3transfer==0.6.0
scikit-image==0.19.3
scipy==1.11.3
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.4.1
shapely==2.0.1
six==1.16.0
SQLAlchemy==1.4.39
tifffile==2022.5.4
typing_extensions==4.3.0
uritemplate==4.1.1
urllib3==1.26.14
wcwidth==0.2.6
Werkzeug==3.0.1
WTForms==3.0.1
zipp==3.8.0

#################################
###     end support info      ###
#################################

herseyc, 2024-01-06:
All was well till I tried to upgrade to your latest build this morning.

aaronwmorris, 2024-01-06:
You can try to apply the database migrations manually.  This is basically what the setup script is doing.

https://github.com/aaronwmorris/indi-allsky/wiki/Database-migrations

```
source virtualenv/indi-allsky/bin/activate

flask db revision --autogenerate
flask db upgrade head
```

herseyc, 2024-01-06:
I receive the same error when I do that as well.

Hersey

herseyc, 2024-01-06:
(indi-allsky) pi@allskypi:~/indi-allsky $ flask db revision --autogenerate
Error: While importing 'app', an ImportError was raised:

Traceback (most recent call last):
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 219, in locate_app
    __import__(module_name)
  File "/home/pi/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/pi/indi-allsky/indi_allsky/flask/__init__.py", line 9, in <module>
    from flask_wtf.csrf import CSRFProtect
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/__init__.py", line 4, in <module>
    from .recaptcha import Recaptcha
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/__init__.py", line 1, in <module>
    from .fields import RecaptchaField
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/fields.py", line 3, in <module>
    from . import widgets
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_wtf/recaptcha/widgets.py", line 2, in <module>
    from flask import Markup
ImportError: cannot import name 'Markup' from 'flask' (/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/__init__.py)

Usage: flask [OPTIONS] COMMAND [ARGS]...
Try 'flask --help' for help.

Error: No such command 'db'.
(indi-allsky) pi@allskypi:~/indi-allsky $

aaronwmorris, 2024-01-06:
Try upgrading your python modules:

```
source virtualenv/indi-allsky/bin/activate

pip install -r requirements/requirements_latest.txt --upgrade
```

herseyc, 2024-01-06:
OK that took a good bit, but looks like we are back up and running now.  Thanks!

herseyc, 2024-01-06:
Appreciate all the help!

closed 2024-01-06 by herseyc