aaronwmorris/indi-allsky #1094: Upgrade fails

maphilli14, 2024-01-08:
Following directions here - https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky
gives

```
allsky2:~/indi-allsky $  git pull origin main
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 26, done.
remote: Counting objects: 100% (26/26), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 26 (delta 12), reused 26 (delta 12), pack-reused 0
Unpacking objects: 100% (26/26), 13.48 KiB | 255.00 KiB/s, done.
From https://github.com/aaronwmorris/indi-allsky
 * branch              main       -> FETCH_HEAD
   07cdf75e..a5fff428  main       -> origin/main
Updating d2a13f23..a5fff428
error: Your local changes to the following files would be overwritten by merge:
        README.md
        ansible/requirements.txt
        ansible/roles/build_indi/defaults/main.yml
        ansible/roles/build_indi/tasks/common/install/main.yml
        ansible/roles/build_indi/tasks/indi_3rdparty/install/RedHat.yml
        ansible/roles/build_indi/tasks/indi_3rdparty/install/main.yml
        ansible/roles/build_indi/tasks/indi_3rdparty/main.yml
        ansible/roles/build_indi/tasks/indi_core/install/main.yml
        ansible/roles/build_indi/tasks/indi_core/main.yml
        ansible/roles/build_indi/tasks/main.yml
        ansible/roles/build_indi/vars/Debian-10.yml
        ansible/roles/build_indi/vars/Debian-11.yml
        ansible/roles/build_indi/vars/Debian-9.yml
        ansible/roles/build_indi/vars/Fedora-34.yml
        ansible/roles/build_indi/vars/Ubuntu-18.yml
        ansible/roles/build_indi/vars/Ubuntu-20.yml
        ansible/roles/build_indi/vars/Ubuntu-22.yml
        ansible/roles/build_indi/vars/main.yml
        config.json_template
        darks.py
        flask.json_template
        indi_allsky/allsky.py
        indi_allsky/darks.py
        indi_allsky/exceptions.py
        indi_allsky/filetransfer/__init__.py
        indi_allsky/flask/__init__.py
        indi_allsky/flask/forms.py
        indi_allsky/flask/miscDb.py
        indi_allsky/flask/models.py
        indi_allsky/flask/templates/base.html
        indi_allsky/flask/templates/chart.html
        indi_allsky/flask/templates/config.html
        indi_allsky/flask/templates/imageviewer.html
        indi_allsky/flask/templates/index.html
        indi_allsky/flask/templates/loop.html
        indi_allsky/flask/templates/sqm.html
        indi_allsky/flask/templates/system.html
        indi_allsky/flask/templates/videoviewer.html
        indi_allsky/flask/views.py
        indi_allsky/image.py
        indi_allsky/keogram.py
        indi_allsky/sqm.py
        indi_allsky/starTrails.py
        indi_allsky/stars.py
        indi_allsky/uploader.py
        indi_allsky/video.py
        misc/build_indi.sh
        misc/rebuild_pyindi-client.sh
        misc/validate_db_entries.py
        requirements_debian10.txt
        requirements_debian11.txt
        requirements_ubuntu18.txt
        service/90-org.aaronwmorris.indi-allsky.pkla
        service/apache_indi-allsky.conf
        service/gunicorn.conf.py
        service/indi-allsky.service
        service/indiserver.service
        service/nginx_astroberry_ssl
        setup.sh
        testing/exposureTest.py
        testing/starTrails_max.py
Please commit your changes or stash them before you merge.
Aborting
```

Doing fresh pull into existing and working legacy version errors with flask migration:

```

Removed /home/maphilli14/.config/systemd/user/default.target.wants/gunicorn-indi-allsky.service.
**** Setup policy kit permissions ****
**** Ensure user is a member of the systemd-journal group ****
**** Setup rsyslog logging ****
**** Indi-allsky config ****
**** Flask config ****
**** Setup DB ****
**** Backup DB prior to migration ****
  Creating directory '/var/lib/indi-allsky/migrations' ...  done
  Creating directory '/var/lib/indi-allsky/migrations/versions' ...  done
  Generating /var/lib/indi-allsky/migrations/env.py ...  done
  Generating /var/lib/indi-allsky/migrations/alembic.ini ...  done
  Generating /var/lib/indi-allsky/migrations/script.py.mako ...  done
  Generating /var/lib/indi-allsky/migrations/README ...  done
  Please edit configuration/connection/logging settings in '/var/lib/indi-allsky/migrations/alembic.ini' before proceeding.
0 blocks
ERROR [flask_migrate] Error: Can't locate revision identified by 'fc09cb85323d'

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

```

maphilli14, 2024-01-09:
Not that I'm aware, I did a mv indi-allsky indi-allsky.20240108bak first then a fresh, full pull and setup.sh to give that error.  I'd like to be able to do the suggested upgrade

`
git pull origin main`

But it gave the odd error
```

Please commit your changes or stash them before you merge.
Aborting
```

maphilli14, 2024-01-09:
no, didn't mean to close

aaronwmorris, 2024-01-09:
I'm not sure how old your original installation was, but that may be part of the issue.

Do you have any `.py` files in `indi-allsky.20240108bak/migrations/versions/` ?

maphilli14, 2024-01-14:
Yes, I reverted back for now...

```

 cd indi-allsky/migrations/versions/
maphilli14@allsky2:~/indi-allsky/migrations/versions $ ls
fbd01843477f_.py  fc09cb85323d_.py  __pycache__
maphilli14@allsky2:~/indi-allsky/migrations/versions $ cat fbd01843477f_.py
"""empty message

Revision ID: fbd01843477f
Revises:
Create Date: 2022-08-20 19:59:45.106592

"""
from alembic import op
import sqlalchemy as sa

```

aaronwmorris, 2024-01-15:
Okay, I understand what is going on.  I moved the migrations out of the git checkout folder a while back.  The setup script would fix this, but since there is a new checkout, it fails.

You will just need to manually move the migration files.

```
# create a new checkout of indi-allsky

rm /var/lib/indi-allsky/migrations/versions/*.py

cp migrations/versions/*.py /var/lib/indi-allsky/migrations/versions
```

Hopefully, the setup.sh script can complete at this point.

maphilli14, 2024-01-15:
Upgrade is complete, now have to fix the GUI!  :O 

```
tart apache2
Site 000-default already disabled
Site default-ssl already disabled
Site indi-allsky already enabled
Synchronizing state of apache2.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable apache2
**** Setup HTDOCS folder ****
**** Setup image folder ****
**** Ensure user is a member of the dialout group ****
**** Disabling Thomas Jacquin's allsky (ignore errors) ****
Failed to stop allsky.service: Unit allsky.service not loaded.
Failed to disable unit: Unit file allsky.service does not exist.
**** Starting gunicorn-indi-allsky.socket
**** Update config camera interface ****
2024-01-15 16:30:38,633 [INFO] MainProcess config._load() #676: Loading configuration from file

*** Configurations are now stored in the database and *NOT* /etc/indi-allsky/config.json ***

Services can be started at the command line or can be started from the web interface

    systemctl --user start indiserver
    systemctl --user start indi-allsky

The web interface may be accessed with the following URL
 (You may have to manually access by IP)

    https://allsky2.local/indi-allsky/

Completed in 484s

Enjoy!
maphilli14@allsky2:~/indi-allsky $ sudo reboot
Connection to allsky closed by remote host.
Connection to allsky closed.
```

maphilli14, 2024-01-15:
```
Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
```

aaronwmorris, 2024-01-15:
Check the exception logged in `/var/log/indi-allsky/webapp-indi-allsky.log`.

You may just need to start indi-allsky and let it run for a minute to stabilize.  You may also try clearing you cookies in your browser.

maphilli14, 2024-01-15:
:(```

maphilli14@allsky2:~ $ tail /var/log/indi-allsky/webapp-indi-allsky.log
Jan 15 16:37:08 allsky2 [ERROR] MainProcess-849 app.log_exception() #825: Exception on /indi-allsky/ [GET]#012Traceback (most recent call last):#012  File "/home/maphilli14/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1455, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/home/maphilli14/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 869, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/home/maphilli14/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 867, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/home/maphilli14/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 852, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)#012  File "/home/maphilli14/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012  File "/home/maphilli14/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 109, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)#012  File "/home/maphilli14/indi-allsky/indi_allsky/flask/base_views.py", line 212, in dispatch_request#012    context = self.get_context()#012  File "/home/maphilli14/indi-allsky/indi_allsky/flask/views.py", line 115, in get_context#012    context = super(IndexView, self).get_context()#012  File "/home/maphilli14/indi-allsky/indi_allsky/flask/base_views.py", line 219, in get_context#012    'astrometric_data'   : self.get_astrometric_info(),#012  File "/home/maphilli14/indi-allsky/indi_allsky/flask/base_views.py", line 356, in get_astrometric_info#012    obs.lon = math.radians(longitude)#012TypeError: must be real number, not NoneType
```

aaronwmorris, 2024-01-16:
Starting indi-allsky should fix this issue.

maphilli14, 2024-01-16:
```
  471  history | grep start
maphilli14@allsky2:~ $ systemctl --user status indiserver
* indiserver.service - Indi Server
     Loaded: loaded (/home/maphilli14/.config/systemd/user/indiserver.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-01-15 16:31:28 EST; 3h 48min ago
   Main PID: 614 (indiserver)
      Tasks: 9 (limit: 3839)
        CPU: 1min 1.478s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/indiserver.service
             |-614 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_qhy_ccd indi_simulator_gps
             |-617 indi_simulator_telescope
             |-618 indi_qhy_ccd
             `-619 indi_simulator_gps

Jan 15 19:28:05 allsky2 indiserver[614]: 2024-01-16T00:28:05: Client 0: read: Connection reset by peer
Jan 15 19:31:21 allsky2 indiserver[614]: 2024-01-16T00:31:21: Client 0: read: Connection reset by peer
Jan 15 19:37:55 allsky2 indiserver[614]: 2024-01-16T00:37:55: Client 0: read: Connection reset by peer
Jan 15 19:42:05 allsky2 indiserver[614]: 2024-01-16T00:42:05: Client 0: read: Connection reset by peer
Jan 15 19:50:37 allsky2 indiserver[614]: 2024-01-16T00:50:37: Client 0: read: Connection reset by peer
Jan 15 19:58:16 allsky2 indiserver[614]: 2024-01-16T00:58:16: Client 0: read: Connection reset by peer
Jan 15 20:04:10 allsky2 indiserver[614]: 2024-01-16T01:04:10: Client 0: read: Connection reset by peer
Jan 15 20:13:08 allsky2 indiserver[614]: 2024-01-16T01:13:08: Client 0: read: Connection reset by peer
Jan 15 20:15:59 allsky2 indiserver[614]: 2024-01-16T01:15:59: Client 0: read: Connection reset by peer
Jan 15 20:18:23 allsky2 indiserver[614]: 2024-01-16T01:18:23: Client 0: read: Connection reset by peer
maphilli14@allsky2:~ $ systemctl --user restart indiserver
maphilli14@allsky2:~ $ systemctl --user status indiserver
* indiserver.service - Indi Server
     Loaded: loaded (/home/maphilli14/.config/systemd/user/indiserver.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-01-15 20:20:08 EST; 1s ago
   Main PID: 5650 (indiserver)
      Tasks: 6 (limit: 3839)
        CPU: 225ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/indiserver.service
             |-5650 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_qhy_ccd indi_simulator_gps
             |-5651 indi_simulator_telescope
             |-5652 indi_qhy_ccd
             `-5653 indi_simulator_gps

Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: -- qhyccd.cpp param
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResource()|START
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResource|auto_detect_camera:false,call InitQHYCCDResourceInside
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|START
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|libusb_version 1.0.24.11584
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|libusb_init(libqhyccd_context) called...
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|numdev set to 0
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|END
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: ************************** config file path  22.2.28.14 svn: 11998  ************************************
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResource|Load ini filePath = /home/maphilli14  fileName = qhyccd.ini
maphilli14@allsky2:~ $ systemctl --user status indiserver
* indiserver.service - Indi Server
     Loaded: loaded (/home/maphilli14/.config/systemd/user/indiserver.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-01-15 20:20:08 EST; 9s ago
   Main PID: 5650 (indiserver)
      Tasks: 6 (limit: 3839)
        CPU: 236ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/indiserver.service
             |-5650 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_qhy_ccd indi_simulator_gps
             |-5651 indi_simulator_telescope
             |-5652 indi_qhy_ccd
             `-5653 indi_simulator_gps

Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: -- qhyccd.cpp param
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResource()|START
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResource|auto_detect_camera:false,call InitQHYCCDResourceInside
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|START
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|libusb_version 1.0.24.11584
Jan 15 20:20:08 allsky2 indiserver[5650]: 2024-01-16T01:20:08: Driver indi_qhy_ccd: QHYCCD|QHYCCD.CPP|libusb_init(libqhyccd_context) called...
[... cut ...]
Jan 15 20:20:38 allsky2 python3[5674]: [INFO] Capture-9-5674/MainThread indi.serverConnected() #347: Server connected (localhost:7624)
Jan 15 20:20:38 allsky2 python3[5674]: [INFO] Capture-9-5674/Dummy-3 indi.newDevice() #215: new device GPS Simulator
Jan 15 20:20:38 allsky2 python3[5674]: [INFO] Capture-9-5674/Dummy-3 indi.newDevice() #215: new device QHY CCD QHY5LII-M-60a7d
Jan 15 20:20:38 allsky2 python3[5674]: [INFO] Capture-9-5674/Dummy-3 indi.newDevice() #215: new device Telescope Simulator
maphilli14@allsky2:~ $
```

maphilli14, 2024-01-16:
no change, still 500 with full purge of all chrome params, cookies, history, etc...

maphilli14, 2024-01-16:
output from later after restart....

```

maphilli14@allsky2:~ $ systemctl --user status indi-allsky
* indi-allsky.service - Indi AllSky Service
     Loaded: loaded (/home/maphilli14/.config/systemd/user/indi-allsky.service; disabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-01-15 20:18:50 EST; 5min ago
TriggeredBy: * indi-allsky.timer
   Main PID: 5604 (python3)
      Tasks: 7 (limit: 3839)
        CPU: 11.907s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/indi-allsky.service
             |-5604 /home/maphilli14/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
             `-5610 /home/maphilli14/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run

Jan 15 20:24:43 allsky2 python3[5744]:   File "/home/maphilli14/indi-allsky/indi_allsky/camera/indi.py", line 711, in configureDevice
Jan 15 20:24:43 allsky2 python3[5744]:     self.set_switch(device, k, on_switches=v.get('on', []), off_switches=v.get('off', []))
Jan 15 20:24:43 allsky2 python3[5744]:   File "/home/maphilli14/indi-allsky/indi_allsky/camera/indi.py", line 1337, in set_switch
Jan 15 20:24:43 allsky2 python3[5744]:     self.sendNewSwitch(c)
Jan 15 20:24:43 allsky2 python3[5744]:   File "/home/maphilli14/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/PyIndi.py", line 755, in sendNewSwitch
Jan 15 20:24:43 allsky2 python3[5744]:     return _PyIndi.BaseClient_sendNewSwitch(self, *args)
Jan 15 20:24:43 allsky2 python3[5744]: TypeError: Wrong number or type of arguments for overloaded function 'BaseClient_sendNewSwitch'.
Jan 15 20:24:43 allsky2 python3[5744]:   Possible C/C++ prototypes are:
Jan 15 20:24:43 allsky2 python3[5744]:     INDI::BaseClient::sendNewSwitch(ISwitchVectorProperty *)
Jan 15 20:24:43 allsky2 python3[5744]:     INDI::BaseClient::sendNewSwitch(char const *,char const *,char const *)
maphilli14@allsky2:~ $
```

aaronwmorris, 2024-01-16:
Can you send the output of `./misc/support_info.sh`?

maphilli14, 2024-01-16:
```
Welcome to Ubuntu 22.04.3 LTS (GNU/Linux 5.15.133.1-microsoft-standard-WSL2 x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

This message is shown once a day. To disable it please create the
/home/maphilli14/.hushlogin file.
maphilli14@Libra:~$ ssh allsky
Linux allsky2 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jan 15 16:32:19 2024 from 192.168.43.55
maphilli14@allsky2:~ $ cat indi-allsky/misc/support_info.sh
#!/bin/bash

#set -x  # command tracing
set -o errexit
set -o nounset

PATH=/usr/local/bin:/usr/bin:/bin
export PATH

function catch_error() {
    echo
    echo
    echo "###############"
    echo "###  ERROR  ###"
    echo "###############"
    echo
    echo "The script exited abnormally, please try to run again..."
    echo
    echo
    exit 1
}
trap catch_error ERR

function catch_sigint() {
    echo
    echo
    echo "###############"
    echo "###  ERROR  ###"
    echo "###############"
    echo
    echo "The script was interrupted, please run the script again to finish..."
    echo
    echo
    exit 1
}
trap catch_sigint SIGINT

DISTRO_NAME=$(lsb_release -s -i)
DISTRO_RELEASE=$(lsb_release -s -r)
CPU_ARCH=$(uname -m)
CPU_BITS=$(getconf LONG_BIT)
CPU_TOTAL=$(grep -c "^proc" /proc/cpuinfo)
MEM_TOTAL=$(grep MemTotal /proc/meminfo | awk "{print \$2}")

if [ -f "/proc/device-tree/model" ]; then
    SYSTEM_MODEL=$(cat /proc/device-tree/model)
else
    SYSTEM_MODEL="Generic PC"
fi

[... 815 lines cut ...]
scipy==1.11.4
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.8.1
shapely==2.0.2
six==1.16.0
SQLAlchemy==2.0.25
support-developer==1.0.5
tifffile==2023.12.9
tomli==2.0.1
toolz==0.12.0
typing_extensions==4.9.0
uritemplate==4.1.1
urllib3==1.26.18
wcwidth==0.2.13
Werkzeug==3.0.1
WTForms==3.1.2
zarr==2.16.1
zipp==3.17.0

#################################
###     end support info      ###
#################################
maphilli14@allsky2:~ $
```

aaronwmorris, 2024-01-16:
Your version of INDI is very old (relatively speaking).  My suggestion would be to upgrade to at least v2.0.3 and rebuild the pyindi-client module.

Building indi will probably take about 40-50 minutes on your system.

```
./misc/build_indi.sh v2.0.3

./misc/rebuild_pyindi-client.sh
```

maphilli14, 2024-01-16:
This did the trick!  Thanks.  I was anticipating that the inid-allsky setup script would pull all the requirements, could you either add this step or adjust the allsky setup to pull new indiserver?  Also, if you want help udating docs, I'm a n00b but willing!  It's not detecting the camera now but I'll upgrade to server 2.0.5 and try again...

```
	
	-- Installing: /usr/local/share/indi/indi_nut.xml
	-- Installing: /usr/local/bin/indi_oasis_focuser
	-- Set runtime path of "/usr/local/bin/indi_oasis_focuser" to ""
	-- Installing: /usr/local/share/indi/indi_astroasis.xml
	
	
	Completed in 2803s
	
	maphilli14@allsky2:~/indi-allsky $ ./misc/rebuild_pyindi-client.sh
	
	
	Detected a custom installation of INDI in /usr/local/bin
	
	
	
	
	The purpose of this script is to rebuild pyindi-client against
	 newer indilib packages.
	
	Rebuild proceeding in 10 seconds... (control-c to cancel)
	
	
	
	Detected INDI version: 2.0.3
	Found existing installation: pyindi-client 1.9.1
	Uninstalling pyindi-client-1.9.1:
	  Successfully uninstalled pyindi-client-1.9.1
	Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
	Collecting pyindi-client
	  Cloning https://github.com/indilib/pyindi-client.git (to revision 674706f) to /tmp/pip-install-ifxo7a2e/pyindi-client_50289e916d514bb7901f178e446ce297
	  Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-ifxo7a2e/pyindi-client_50289e916d514bb7901f178e446ce297
	  WARNING: Did not find branch or tag '674706f', assuming revision or ref.
	  Running command git checkout -q 674706f
	  Resolved https://github.com/indilib/pyindi-client.git to commit 674706f
	  Installing build dependencies ... done
	  Getting requirements to build wheel ... done
	  Preparing metadata (pyproject.toml) ... done
	Building wheels for collected packages: pyindi-client
	  Building wheel for pyindi-client (pyproject.toml) ... done
	  Created wheel for pyindi-client: filename=pyindi_client-1.9.1-cp39-cp39-linux_armv7l.whl size=4355605 sha256=1872dafebd814b54cb10c18726003d8253e14d79ec6daea4a9d05d9d387a84b0
	  Stored in directory: /tmp/pip-ephem-wheel-cache-3xw11bm0/wheels/f5/43/08/a6ab23044641d7a9d7627e6d53b21fbab28ed0731b8ea3dd86
	Successfully built pyindi-client
	Installing collected packages: pyindi-client
	Successfully installed pyindi-client-1.9.1
	
	
	Completed in 130s
	
	maphilli14@allsky2:~/indi-allsky $ systemctl --user restart indiserver
	maphilli14@allsky2:~/indi-allsky $ systemctl --user restart indi-allsky
	maphilli14@allsky2:~/indi-allsky $ sudo reboot
	Connection to allsky closed by remote host.
	Connection to allsky closed.
	
	

```

maphilli14, 2024-01-16:
Ran the indiserver upgrade and client and a couple reboots, browser flushes etc and it's now working!  I need to figure out all the new knobs and features, some of the char's look funny.....

[image]

aaronwmorris, 2024-01-16:
Fantastic.  It looks like there may be some extra characters in there.  Are you using OpenCV or Pillow for drawing text?

maphilli14, 2024-01-16:
> Fantastic. It looks like there may be some extra characters in there. Are you using OpenCV or Pillow for drawing text?

Just the defaults from upgrade!  Love to get started with overlay compass etc

closed 2024-04-28 by aaronwmorris