aaronwmorris/indi-allsky #1095: Installation fails on fresh SD card Pi4

pterodyne, 2024-01-11:
,Not sure what is going on. Im helping a friend install.  We even re-imaged the pi and after installing git, pulling it down and running setup it goes on for a while and then fails:

Collecting pip
  Using cached https://www.piwheels.org/simple/pip/pip-23.3.2-py3-none-any.whl (2.1 MB)
Requirement already satisfied: setuptools in ./virtualenv/indi-allsky/lib/python3.11/site-packages (66.1.1)
Collecting setuptools
  Downloading https://www.piwheels.org/simple/setuptools/setuptools-69.0.3-py3-none-any.whl (819 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺ 802.8/819.5 kB 123.3 kB/s eta 0:00:01
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    setuptools from https://www.piwheels.org/simple/setuptools/setuptools-69.0.3-py3-none-any.whl#sha256=55a27b1207ffd06fc72552920c9ea1ce81541b8b651ae406bf3d3bb153bf316f:
        Expected sha256 55a27b1207ffd06fc72552920c9ea1ce81541b8b651ae406bf3d3bb153bf316f
             Got        b945db0b7f7606a493cbda212b6373648eb07b8efedd591f9d7a946994f5638e

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

Ive also tried removing the venv. 

Any ideas what to do here?

Trying to use libcamera on an imx477

thanks

Bryan

uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm

aaronwmorris, 2024-01-11:
I think you might have caught some of the python modules in the middle of an upgrade.  You may want to wait a while (maybe even a few hours) and try again.  Clear the pip cache before you try again.

```
source virtualenv/indi-allsky/bin/activate

python -m pip cache purge
```

aaronwmorris, 2024-01-11:
Have you had a chance to try again?

pterodyne, 2024-01-12:
Just did a fresh SD card again, and ran from scratch:

Collecting astropy (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 3))
  Downloading astropy-6.0.0-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (9.8 kB)
Collecting numpy>=1.22.2 (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 5))
  Downloading numpy-1.26.3-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.7/62.7 kB 211.2 kB/s eta 0:00:00
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/opencv-python-headless/
Collecting opencv-python-headless>=4.8.1.78 (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 7))
  Downloading opencv_python_headless-4.9.0.80-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (20 kB)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/scipy/
Collecting scipy>=1.11.3 (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 9))
  Downloading scipy-1.11.4-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (60 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.4/60.4 kB 380.4 kB/s eta 0:00:00
Collecting ccdproc (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 11))
  Downloading https://www.piwheels.org/simple/ccdproc/ccdproc-2.4.1-py3-none-any.whl (355 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 355.4/355.4 kB 327.6 kB/s eta 0:00:00
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/scikit-image/
Collecting scikit-image>=0.20.0 (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 13))
  Downloading scikit_image-0.22.0-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (13 kB)
Collecting astroalign (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 14))
  Downloading https://www.piwheels.org/simple/astroalign/astroalign-2.5.1-py3-none-any.whl (11 kB)
Collecting bottleneck (from -r /home/ecjoyner/indi-allsky/requirements/requirements_latest.txt (line 15))
  Downloading Bottleneck-1.3.7.tar.gz (103 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 103.1/103.1 kB 331.2 kB/s eta 0:00:00
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
      Collecting setuptools
        Downloading https://www.piwheels.org/simple/setuptools/setuptools-69.0.3-py3-none-any.whl (819 kB)
           ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  802.8/819.5 kB 290.3 kB/s eta 0:00:01
      ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
          setuptools from https://www.piwheels.org/simple/setuptools/setuptools-69.0.3-py3-none-any.whl#sha256=55a27b1207ffd06fc72552920c9ea1ce81541b8b651ae406bf3d3bb153bf316f:
              Expected sha256 55a27b1207ffd06fc72552920c9ea1ce81541b8b651ae406bf3d3bb153bf316f
                   Got        b945db0b7f7606a493cbda212b6373648eb07b8efedd591f9d7a946994f5638e

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

aaronwmorris, 2024-01-12:
I tested a new build on an x86_64 system today and I did not have any issues.  I will test on a Rpi4 tomorrow.

aaronwmorris, 2024-01-12:
No issues on the build on a Raspi.

While I have not had this particular issue, I have experienced some weird issues that affected my ability to download packages and python modules due to what I suspect being old DNS entries floating around my ISPs DNS servers.

I would like you to try to manually change your DNS servers in /etc/resolv.conf to be `8.8.8.8` (Google) or `1.1.1.1` (Cloudflare) and see if you have a better experience.

pterodyne, 2024-01-16:
I tried that also.  Strange thing is it continued to happen even after wiping the SD again (several times now).  So we started fresh one more time and did Thomas' allsky instead just to see.  It also failed with hash errors.  No idea here.  I did ultimately get Thomas' installed by deleting /tmp/package and re-runining where it picked up where it left off.  After several deletes of /tmp/package and restarting of the installer it finished.  So at least he has that version now. Not sure if that would have worked for indi-allsky.

Not being a python person, I really don't know what to do. I can tell you on my own allsky camera (indi-allsky) I had a similar issue with hashes, but eventually I got it to work (1 month ago or so).No Idea what I did.  Some permutation of deleting things in venv /bin/_pycache_ etc.

Thanks.

Bryan

pterodyne, 2024-01-16:
> Is it possible that your raspberry pi disk image was corrupted?

I suppose that is possible, but otherwise everything is working fine. apt-get update upgrade, etc.. He was using the default pi imager to write the default 64bit image.
It's just interesting that had the same problem with hashes, a little while ago even if it ultimately worked with some encouragement.

aaronwmorris, 2024-01-17:
The only other thing I can think of is perhaps the sdcard is failing (assuming it was the same card in both cases).

pterodyne, 2024-01-17:
We'll try another one for fun.

Thanks.

closed 2024-04-28 by aaronwmorris