aaronwmorris/indi-allsky #1100: Unable to start services

aras-vbo, 2024-01-15:
Hi,
On Raspberry Pi 4, with RaspberryPi OS 11 (Bullseye) 64 bits + Pi cam HQ
I was able to install with the shell script (setup.sh) then to compile Indi with the script (BUILD_INDI_CORE=true BUILD_INDI_3RDPARTY=false ./misc/build_indi.sh).

But when I tried to launch services (systemctl --user start indiserver), I received a message :
Failed to start indiserver.service : Unit indiserver.service not found.
The same message for service indi-allsky

Vincent

aaronwmorris, 2024-01-15:
You have to run setup.sh after the INDI compile has completed.  That should setup the indiserver and indi-allsky services.

aaronwmorris, 2024-01-26:
Closing this for now.  Let me know if there is anything else.

closed 2024-01-26 by aaronwmorris