aaronwmorris/indi-allsky #1126: Error during setup

WirthmU, 2024-02-09:
I got the following error during setup. Repeating the setup gave the same result:

```
ERROR: Could not open requirements file: [Errno 21] Is a directory: '/home/pi/indi-allsky/'

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```
Any help tho this'

aaronwmorris, 2024-02-11:
Sorry, I had to travel for work this past week.

Have you made any progress on this?  I have not seen this error reported previously, do you have a few more lines of the log available to see where this failure occurred?

aaronwmorris, 2024-04-28:
I think you got through this issue (based on your other posts) so I am going close for now.  Feel free to reopen.

closed 2024-04-28 by aaronwmorris