aaronwmorris/indi-allsky #1130: Trouble with getting Dark frames with darks.py

jproehl, 2024-02-11:
### Discussed in https://github.com/aaronwmorris/indi-allsky/discussions/1128

<div type='discussions-op-text'>

<sup>Originally posted by **jproehl** February  9, 2024</sup>
I've run into a problem using darks.py. I set my Gain at 150 and my max exposure to 20s using the web interface to indi-allsky.local.
 I then warmed my ASI120MC-S up to ~35C (to try to get some warm darks for summertime use), placed it in a plastic bag and wrapped it a towel and then into my freezer. I stopped the indi-allsky service and activated the python interface. I then ran:

./darks.py --Count 5 tempsigmaclip 

and everything seems to roll along fine. The code goes through taking 5 exposures of 20s, 15s, 10s, 5s and 1s at Gain 150, then repeats that sequence at Gain 75.  By the end of that sequence the temperature reported from the sensor was +31.5C.  It echos that the next temperature threshold is +29.5C. That's where my problem begins .... for the next 2+ hours my terminal running darks.py endlessly repeats the line that the sensor is at +31.5C and then the line that the next threshold is +29.5C (each of these lines list the correct updated time, so the RPi4 system has not frozen).  The python program darks.py never gets to the second temperature to take the next set of darks.

So, after the 2+ hours I let it run I stopped it with Ctrl-C, restarted the indi-allsky service and immediately looked at the temperature listed on the next image and the sensor is at -23.5C!!

Any idea what I've done wrong?

Thanks!
Jeff</div>

aaronwmorris, 2024-02-12:
That is really odd.  I confirmed the behavior with my own test.  Let me research this some more.

aaronwmorris, 2024-02-12:
I have not found a code problem yet.  I am starting to suspect the temperature reported by the camera does not get updated until an exposure is taken... stay tuned.

jproehl, 2024-02-14:
Any update? 

Could you just add a 1sec exposure  (that you don’t save) right before darks.py requests the updated sensor temperature?

aaronwmorris, 2024-02-14:
That is what I am working on, but I am having more trouble with this than I should be.

jproehl, 2024-02-14:
Great, thanks! How do I fix it on my RPi? Is there an update process to get the changes from GitHub?

aaronwmorris, 2024-02-14:
indi-allsky releases are just rolling updates.  You can always just pull down the latest code and re-run setup.sh.

A more detailed procedure can be found here:
https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

closed 2024-04-28 by aaronwmorris