aaronwmorris/indi-allsky #1164: CLAHE 16 bit Processing Distorted Colors

HxPI, 2024-03-02:
When CLAHE is enabled with 16 bit processing, the images from my INDI camera ASI678MC set to daytime exposure settings have green patches in various places in the image. It doesn't appear to be associated with over/under exposure. Is there anything to check in settings or are there any known issues/limitations with this feature?

aaronwmorris, 2024-03-02:
I have noticed the same thing.  I have not figured out why that is happening.  It is possible it is an overflow in one of the channels in LAB colorspace.

aaronwmorris, 2024-06-07:
Merged #1309 to fix.

This was much easier than I realized.  I spent quite a bit of time trying to fix this when I first developed the algorithm, but I never saw my mistake.  I calculated the max value for 16-bit numbers as `2 ** 16`, but for the purpose of calculations, it needs to be `(2 ** 16) - 1`

closed 2024-07-17 by aaronwmorris