aaronwmorris/indi-allsky #1169: Transition from Day to Night - Image capture doesn't resume

gaitskell, 2024-03-04:
For the last 7 days I have has an issue

aaronwmorris, 2024-03-04:
Is it a correct assumption you are using an ZWO ASI120?

gaitskell, 2024-03-05:
CAMERA:

Name
SX CCD UltraStar

Driver
indi_sx_ccd

Resolution
1392 x 1040 (1737 diag)

Dimensions
9.0mm x 6.7mm (11.2mm diag)
Pixels
6.45µm
CFA
None
Bits
16
Gain
-1--1
Exposure
0.000100-3600s

gaitskell, 2024-03-07:
I have 60 seconds as the maximum exposure - longer than that and the star trails become too noticeable. This works well on moonless dark nights in suburbia. 
I have set 4 secs as the minimum, 5 secs as default. I did this because if I allow much shorter times the autoexposure adjust at the start of the night will not work. It remains stuck at very short exposures, and all images are dark.

gaitskell, 2024-03-07:
Are there different default times I should try? Can I expand the log file level to get more diagnostic data for the exposure logic and error case?

aaronwmorris, 2024-03-07:
This is probably a problem with my bit depth scaling auto-detection magic.

indi-allsky accepts RAW image data by default, which is usually uint16 data.  There is some variability in the INDI drivers as to the data range used for the FITS data.  For instance, if you have a 12-bit camera, some drivers auto-scale the data to the full 16-bit space and the maximum data value will be 65535.  Unfortunately, some drivers will leave the data in the 12-bit space, so the maximum value is 4095.  indi-allsky needs to know this to properly scale the data to 8-bits for the resulting JPEG data.

indi-allsky has a simple algorithm that assumes the data is 8-bits by default, and based on the maximum value in the data, detects the likely data range.  Sometimes, there is not enough noise (hot pixels actually help here) and the algorithm will assume 12-bit data is 10-bits, scale the data incorrectly, and the image will be "gray" at the target brightness value.  The exposure is never scaled up because the algorithm thinks it achieved the target exposure.

One of the fixes is what you already discovered, setting the exposure to a sufficiently high value so there is enough data to properly detect the data range.

The other option is to override the auto-detection algorithm and explicitly define the bit depth of the data.  This setting is on the Camera Tab at `Camera Bit Depth`

gaitskell, 2024-03-09:
I changed the Bit Depth to manual value - 16 bit (correct for CCD)
I also put Min Exposure back on Auto values. This seems to be working when manually started.

However, the issue with failed re-start at the beginning of the night after being suspended all day is still happening.

The Notifcation window says "Camera last ready 45288s ago. Camera might be hung." (TIme is half a day)

Is it possible the code is misinterpreting the daytime suspension as a non-responsive camera?  

Thanks, Rick

[image]

aaronwmorris, 2024-03-09:
No, it should not misinterpret that.  That is a problem I fixed long ago.

Can you post the output of `./misc/support_info.sh` ?

gaitskell, 2024-03-14:
admin@PiAllSky:~/indi-allsky $ cat misc/support_info.sh 
#!/bin/bash

#set -x  # command tracing
set -o errexit
set -o nounset

PATH=/usr/local/bin:/usr/bin:/bin
export PATH

function catch_error() {
    echo
    echo
    echo "###############"
    echo "###  ERROR  ###"
    echo "###############"
    echo
    echo "The script exited abnormally, please try to run again..."
    echo
    echo
    exit 1
}
trap catch_error ERR

function catch_sigint() {
    echo
    echo
    echo "###############"
    echo "###  ERROR  ###"
    echo "###############"
    echo
    echo "The script was interrupted, please run the script again to finish..."
    echo
    echo
    exit 1
}
trap catch_sigint SIGINT

DISTRO_NAME=$(lsb_release -s -i)
DISTRO_RELEASE=$(lsb_release -s -r)
CPU_ARCH=$(uname -m)
CPU_BITS=$(getconf LONG_BIT)
CPU_TOTAL=$(grep -c "^proc" /proc/cpuinfo)
MEM_TOTAL=$(grep MemTotal /proc/meminfo | awk "{print \$2}")

if [ -f "/proc/device-tree/model" ]; then
    SYSTEM_MODEL=$(cat /proc/device-tree/model)
else
    SYSTEM_MODEL="Generic PC"
fi

if which indiserver >/dev/null 2>&1; then
    INDISERVER=$(which indiserver)
else
    INDISERVER="not found"
fi

SCRIPT_DIR=$(dirname "$0")
cd "$SCRIPT_DIR/.."
ALLSKY_DIRECTORY=$PWD
cd "$OLDPWD"

echo "#################################"
echo "### indi-allsky support info  ###"
echo "#################################"

sleep 3

echo
echo "Distribution: $DISTRO_NAME"
echo "Release: $DISTRO_RELEASE"
echo "Arch: $CPU_ARCH"
echo "Bits: $CPU_BITS"
echo
echo "CPUs: $CPU_TOTAL"
echo "Memory: $MEM_TOTAL kB"
echo
echo "System: $SYSTEM_MODEL"
echo
echo "Uptime"
uptime
echo
echo "Filesystems"
df -k
echo
echo "system python: $(python3 -V)"
echo
echo "indiserver: $INDISERVER"
echo

if [ -f "/etc/astroberry.version" ]; then
    echo "Detected Astroberry server"
    echo
fi

echo
echo "User info"
id
echo

echo "Process info"
# shellcheck disable=SC2009
ps auxwww | grep indi | grep -v grep || true
echo

echo "USB info"
lsusb
echo

echo "USB Permissions"
find /dev/bus/usb -ls
echo

echo "Module info"
lsmod
echo

echo "git status"
git status | head -n 100
echo

echo "git log"
git log -n 1 | head -n 100
echo

if pkg-config --exists libindi; then
    DETECTED_INDIVERSION=$(pkg-config --modversion libindi)
    echo "indi version: $DETECTED_INDIVERSION"
    echo
else
    echo "indi version: not detected"
    echo
fi

echo "indi packages"
dpkg -l | grep libindi || true
echo

if which indi_getprop >/dev/null 2>&1; then
    echo "Detected indi properties"
    indi_getprop -v 2>&1 || true
    echo
fi

if pkg-config --exists libcamera; then
    DETECTED_LIBCAMERA=$(pkg-config --modversion libcamera)
    echo "libcamera version: $DETECTED_LIBCAMERA"
    echo
else
    echo "libcamera: not detected"
    echo
fi

echo "libcamera packages"
dpkg -l | grep libcamera || true
echo

echo "libcamera cameras"
if which rpicam-hello >/dev/null 2>&1; then
    echo "rpicam-hello: $(which rpicam-hello)"
    rpicam-hello --list-cameras || true
    echo
elif which libcamera-hello >/dev/null 2>&1; then
    echo "libcamera-hello: $(which libcamera-hello)"
    libcamera-hello --list-cameras || true
    echo
else
    echo "libcamera-hello not available"
    echo
fi

echo "python packages"
dpkg -l | grep python || true
echo

if [ -d "${ALLSKY_DIRECTORY}/virtualenv/indi-allsky" ]; then
    echo "Detected indi-allsky virtualenv"

    # shellcheck source=/dev/null
    source "${ALLSKY_DIRECTORY}/virtualenv/indi-allsky/bin/activate"
    echo "virtualenv python: $(python3 -V)"
    echo "virtualenv PATH: $PATH"

    if which flask >/dev/null 2>&1; then
        echo "flask command: $(which flask)"
    else
        echo "flask: not found"
    fi

    echo
    echo "virtualenv python modules"
    pip freeze
    deactivate
    echo
else
    echo "indi-allsky virtualenv is not created"
    echo
fi

echo "#################################"
echo "###     end support info      ###"
echo "#################################"

gaitskell, 2024-03-14:
Please note that the Starlight Xpress (SX) Oculus Pro uses the ICX267AL Sony SuperHAD interline CCD 
https://www.sxccd.com/product/oculus-pro-all-sky-camera-150o/

* CCD type: ICX267AL Sony SuperHAD interline CCD with low dark current and vertical anti-blooming. 
* CCD Full resolution pixel data: Pixel size: 4.65uM x 4.65uM, Image format: 1392 x 1040 pixels 
* CCD Image area: 6.4mm (Horizontal) x 4.75mm (Vertical). 
* CCD quality: Grade 1 
* Spectral Response: QE max at 520nM (~50%), 30% at 420nM and 670nM. 
* Readout Noise: Less than 10 electrons RMS – typically only 7 electrons. 
* Full-well capacity: Greater than 15,000 e- (unbinned) 
* Anti-blooming: Overload margin greater than 1000x. 
* Dark current: Less than 0.1 electrons/second @ + 10C ambient. 
* Data format: 16 bits. 
* System gain: 0.3 electrons per ADU 
* Lens details: 2.5mm f/1.2 150 degree ‘Fish Eye’ 
* Computer Interface: Built-in USB 2.0 compatible interface. 
* Image download time: Typically 0.6 seconds at full resolution using USB 2.0. 
* Power requirements: Camera is USB powered, dew heater requires 12v at 1 amp max. 
* Input connection: ‘Mini B’ USB socket for USB2.0 and a 2.1mm jack for +12v heater supply. 
* Cooling system: Ambient air cooling. 
* Viewing dome: High quality polycarbonate hemisphere (replaceable) 
* Overall size: 150mm tall x 95mm diameter

gaitskell, 2024-03-15:
I tried a new setup with the INDI_SONY_CCD selected and the hardware camera was not recognized. No connection possible.

So I went back to the original setup with INDI_SX_CCD.

Let me know if you see anything critical in the diagnostic printout associated with the previous INDI_SX_CCD  setup shown in the above support_info.sh.

aaronwmorris, 2024-03-15:
I have reviewed the indilib docs a little more closely.  I have always considered when the exposure control is in IPS_ALERT state, that the camera was effectively in a failed status.  However, that is not really true.  It just means the previous command/action failed.

Let me think about this to see if I can integrate this concept.

aaronwmorris, 2024-03-16:
Merged #1193 as an attempt to continue taking images after the camera exposure reports ALERT.

gaitskell, 2024-03-16:
thanks for feedback on the way ALERTS are handled. Should I monitor issue here or on the other 1193 issue chain for updates on that?

On the hardware side; Tonight I have determined that the SX Oculus Camera USB connection management may behave differently depending on the condition of a separate 12V DC line that is also fed into the Oculus unit. I am going to start running with the 12V DC connection improved (I made a better connector)  to see if that influences the initial ALERT state at nightfall. The DC supply is for the dew heater, but it appears it could possibly affect the USB behavior based on my tests plugging/unplugging the 12V line.

(Before working on the 12V line  I  checked the USB connector on the Oculus. The USB connector showed no issues with mechanical flexing/slipping causing electrical intermittency or poor connections.)

aaronwmorris, 2024-03-16:
We'll keep the discussion here.

gaitskell, 2024-03-19:
The potential issue that I saw with 12V heater supply possibly not being connected properly and a cross interaction with USB was a red herring. Everything is connected properly. I still see the Camera logic hang at the start of the night’s observation. 

I tried saving images once an hour during the day. This didn’t change an ALERT being issues at the start of the night (smaller number of seconds in message box), and the acquisition hanging at that point. 

I will try a regular daytime schedule, images once a minute and see how this works.

Thanks, Rick

aaronwmorris, 2024-03-19:
Did you upgrade the code yet for the fix I attempted?  Its fine if it did not work, I have another idea if that fix did not work.

gaitskell, 2024-03-22:
Aaron, apologies was away from the lab. Will test new code now and report after a couple of cycles. Assume I should go back to having it suspended during day and only run at night. 

(My tests with older code this week where it was running at 1 minute intervals both day and night were stable. As discussed the issue was occurring when I have larger steps in day or if camera is fully suspended during day)

gaitskell, 2024-03-24:
@aaronwmorris the new code logic is working. Image capture slept during day and restarted successfully two days running. Thank you so much for working on this issue.
I really like the new changes to status reporting, menus and state change count down.

(A separate question I should ask in a new thread, if it has an interesting answer… did you design any of the databases you use inside your code to also be potentially read-only/accessed externally? At present I create a new external database with summary characteristics of the FITS files that have been copied to the GCP based on my own processing. I realize you have already calculated/recorded this information in your own internal database files (e.g. Exposure time, number of stars identified, moon position, etc.…). Is there a logical way to serve this data externally to people analysing the FITS files? No big deal if this is not part of your design, but I would hate to think I have simply misunderstood your intended data export interface.

Another question which I should post in discussion?? Has anyone had success creating a pipeline which automatically determines the full sky coordinates for each all sky image so any desired object can then be located within the images? Where is the right place to have that discussion?)

aaronwmorris, 2024-03-27:
The database is a standard Sqlite database. The database is configured with the Write Ahead Logging [WAL] journal so it may be accessed concurrently, for both reads and writes, with almost no impact.  Since it is Sqlite, it is not really capable of being accessed remotely.  I have a page that has some basic query info:  https://github.com/aaronwmorris/indi-allsky/wiki/SQL-queries

It is also possible to configure the database as MariaDB/MySQL which could be accessed remotely.  I have a procedure for converting an existing Sqlite DB to MariaDB.  https://github.com/aaronwmorris/indi-allsky/wiki/Database-conversion

As for the FITS files, they also have most of the standard FITS headers.  Exposure, latitude, longitude, etc so it should be easy enough to calculate some standard astrometric info.  I could look into adding things like stars if it would be useful.

closed 2024-04-28 by aaronwmorris