aaronwmorris/indi-allsky #1180: MQTT fails to publish messages

giangitesta, 2024-03-07:
Hi,
even if you connect to the broker the following error appears into log:
2024-03-07T16:48:24.367970+01:00 indiallsky [INFO] MainProcess-4553/MainThread allsky._fileUploadWorkerStart() [488]: Starting Upload-61 worker
2024-03-07T16:48:24.367709+01:00 indiallsky [ERROR] MainProcess-4553/MainThread allsky._fileUploadWorkerStart() [481]: Upload worker exception: 
2024-03-07T16:48:24.367416+01:00 indiallsky [ERROR] MainProcess-4553/MainThread allsky._fileUploadWorkerStart() [481]: Upload worker exception: TypeError: payload must be bytes if set
2024-03-07T16:48:24.367124+01:00 indiallsky [ERROR] MainProcess-4553/MainThread allsky._fileUploadWorkerStart() [481]: Upload worker exception:     raise TypeError('payload must be bytes if set')

Maybe it could be due to a paho library update?
In practice the functionality is completely out of order.

Thank you

aaronwmorris, 2024-03-07:
Indeed, I have seen this exception, as well.  Thank you for reminding me, I will track this down.

aaronwmorris, 2024-03-07:
I guess this was a breaking change for paho-mqtt v2.0.0.  Merged #1182 to fix.

Previously, binary image data had to be submitted to the queue as a bytearray.  I think it what has happened is it will now it only accept bytes from the interface and converts the data to a bytearrray internally.  I could not find any reference to this change in the change logs.

Please let me know if everything still works.

giangitesta, 2024-03-07:
Hi, it works now. There are no more errors in the log and it publishes correctly. A thousand thanks.

closed 2024-03-11 by aaronwmorris