aaronwmorris/indi-allsky #1183: Unable to access website

WirthmU, 2024-03-08:
I am new to this so its rather a beginner-issue.

Indi-allsky is up and running on a RPI/bookworm and I can access the indi-allsky website from my LAN. The router/mywebserver has been configured for port forwarding TCP/UDP via port 1500 to the RPI with the local indi-allsky website.

Unfortunately remote access via https://mywebserver:1500/allsky-indi does not work (says connection is disconnected).

Under "Remote Web Portal" its written that "The indi-allsky web interface is designed to be directly exposed to the Internet". Do I have change any of the default settings?

aaronwmorris, 2024-03-08:
Port forwarding should work fine.  Just make sure you are forwarding port 1500/tcp to port 443/tcp.

Have you tested from a mobile device NOT using wifi?  You may not be able to access the external site from your own internal network.  Some home routers do not support hairpinning very well.

WirthmU, 2024-03-08:
443 die the trick - it's working now.

closed 2024-03-11 by aaronwmorris