aaronwmorris/indi-allsky #1187: Large X shaped pattern tiled across images

mburge, 2024-03-12:
Any idea where this large X shaped pattern tiled across the images is originating?  I am stacking 5 images with registration and my guess is it is originating in that process.  The raw images are being captured in 16 bit from an ASI294 with 60 second exposure. I'll try to chase it down but thought you might have an idea off hand. I'm saving images as jpg's using the default settings. I attached a sample in PNG format.

[image]

aaronwmorris, 2024-03-12:
Are you using the CLAHE for contrast enhancement?

mburge, 2024-03-12:
No, it is not enabled, in addition to the stack with registration, the only things on are AWB, saturation at 1.5, average neutral SCNR, and stretching. All at defaults. I defocused the camera to check if it was in the light path, but no luck, here's an externally contrast enhanced detail:

[image]

mburge, 2024-03-12:
Looks like it is an artifact of the registration process. I disabled all stacking. No artifacts. Enabled stacking 5 with maximum but w/out registration. Got images with star trails consisting of 5 stars, as you'd expect with maximum and no registration. Enabled registration and artifacts returned. My guess is that when solving for the transformation matrix it is iteratively alternating between close solutions, maybe just because of precision. Kind of neat actually, that is why the structure is so regular. I'll look tomorrow. By the way, love your system.

aaronwmorris, 2024-03-12:
Thank you!

I could barely see the pattern on my laptop last night (but I could see it), but I can see it much better on my workstation.

I have used registration quite a few times, although, I rarely leave it on.  It has to be a clear night.  However, I have never seen that.  My cameras are not as high resolution as the ASI294.

https://github.com/quatrope/astroalign is the module used for registration.  I will be the first to admit that the registration can be less than perfect.  All sky images have enough distortion and enough variations due to planes and satellites that misalignments are very common.  After I implemented the filter to check for rotations exceeding a standard deviation, misalignments are less common now.

I wonder if this pattern is some interaction due to the quad bayer matrix on the IMX294.

mburge, 2024-03-14:
I had a chance to run some tests and it is definitely a result of registration.

Split screen with registration
[image]

Split screen with just stacking
[image]

I haven't found where it is yet, I think you're right, it's an artifact of the quad mask / 16 bit format. Solution for now is to just turn off stacking and registration.

[image]

jschmied, 2024-11-26:
A example with imx477. 
[image]

aaronwmorris, 2024-11-26:
I still have not been able to reproduce this.  Are you using CLAHE?

jschmied, 2024-11-26:
I don't use any enhancements. Only SCNR. Pattern changes with number of registered pictures. Maybe I should ask in the astroalign project ..

aaronwmorris, 2024-11-28:
@jschmied Did you happen to have your camera sitting in front of a screen door or something like that?

jschmied, 2024-11-28:
No, pictures without stacking are without pattern of course. I suspect its a product of imperfect sensor, debayer and stacking. How can I keep the dng files saved by libcamera? Export RAW is exporting after debayer. I would like to check if processing with siril or deepskystacker produces pattern also. Maybe I find a option causing this. Pattern looks like Moire of superimposed gitter:

[image]

aaronwmorris, 2024-11-29:
The FITS processor is probably the best way to test image changes faster than others.  Changes to the code are immediately available without having to restart anything.

The FITS processor uses the same exact code methods as the image processing during capture.  All of the processing magic happens in `indi_allsky/processing.py` (and included modules).

I have had this thought previously, however, I have not investigated yet.  You can see in the order of operations https://github.com/aaronwmorris/indi-allsky/wiki/Image-order-of-operations that stacking occurs before debayering and that could be the cause of the moire

jschmied, 2024-11-29:
Simple stacking needs to be moved after debayer. Otherweise offset and rotation aligns different colors into same pixel.

Just swaping the function calls is not working because stack() is populating self.image and debayer() gets AttributeError in flask/views.py", line 5777 -> processing.py", line 1061

Also other participants in stacking (self.image_list) must be debayered.

Have to get deeper into the code ...

aaronwmorris, 2024-12-05:
Merged #1699 which performs the debayering before stacking.  The astroalign module does indeed support registering color images.

@jschmied @mburge Please let me know if this corrects the issue with the moire pattern.

jschmied, 2024-12-05:
Thanks for your effort! I installed it but its very cloudy the next days.

jschmied, 2024-12-06:
It's fixed! Thanks for your effort.

Stacked (I need to do darks, you can see trails of hot pixels)
[image]

Original (more noise)
[image]

Stacked: (Detla, Gamma, Eta Cassiopeiae)
[image]

Unstacked:
[image]

aaronwmorris, 2024-12-06:
You may already be using this, but I recommend using Maximum instead of Average for the stacking of color sensors.  

This is kind of unique to all sky cameras, but when you have achieved critical focus, most stars will only illuminate a single pixel on the camera.  Due to the bayer filter, the star will sometimes just be red, green, or blue.  When a star moves between frames, it might be red one frame and green in another.  The averaging occurs in each color channel, so if you had a star run across 3 different frames, the overall luminance can be reduced by up to 66% (for the star).

For example, the average of

- [255, 0, 0]
- [0, 255, 0]
- [0, 0, 255]

is

- [85, 85, 85]

Of course, this is all a matter of taste.  :-)

jschmied, 2024-12-06:
Sounds reasonable. I already added some code because it tries to stack pictures without stars and fails after 7 seconds to find a match. I'll send a PR next week.

closed 2025-02-20 by aaronwmorris