aaronwmorris/indi-allsky #1199: Remote connection issues

henrikeri, 2024-03-18:
Hi,

I have moved the server to run remotely with regards to the indiserver. It's a 100 mbit connection with low latency (10-20 ms) and they are bridged in the form that the remote RaspberryPi with the camera is connected to the server at home with wireguard.

It seems to be very random if the remote connection works or not. When not working I get:
2024-03-18T20:29:53.164437+01:00 server [ERROR] Capture-6-3391/MainThread capture.saferun() [348]: 4.7832s EXPOSURE RECEIVED IN 0.1503s.  POSSIBLE CAMERA PROBLEM.

Are there settings I can change to improve? I see that I am able to connect with Ekos, and the Remote raspberry pi indi-allsky instance works ok when I start it.

aaronwmorris, 2024-03-18:
What kind of camera are you using?

henrikeri, 2024-03-18:
> What kind of camera are you using?

ASI224MC

henrikeri, 2024-03-18:
It seems more likely to work, if I am running the remote indi-allsky at a "reasonable" frame rate such as 1 exp/30 s

aaronwmorris, 2024-03-18:
Is it possible you have the indi-allsky capture process also running on the Raspi?  Meaning there are two capture processes sending exposure commands to the camera?

```
systemctl --user status indi-allsky
```

henrikeri, 2024-03-18:
> Is it possible you have the indi-allsky capture process also running on the Raspi? Meaning there are two capture processes sending exposure commands to the camera?
> 
> ```
> systemctl --user status indi-allsky
> ```

Yeah, funny enough. It only seems to work if the camera has recently been "used"  by the indi-allsky capture process on the raspi. If i stop it alltogether and only run the indiserver.service, it almost never works.

aaronwmorris, 2024-03-18:
Can you provide the output of `./misc/support_info.sh` on the Raspi?

henrikeri, 2024-03-18:
See below, let me know if you would like the same from the home server as well.

```
#################################
### indi-allsky support info  ###
#################################

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3883064 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allsky 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Uptime
 20:55:02 up 30 min,  1 user,  load average: 0.92, 0.58, 0.39

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1671540        0   1671540   0% /dev
tmpfs             388308     9572    378736   3% /run
/dev/mmcblk0p2  29914000 15728112  12647172  56% /
tmpfs            1941532      188   1941344   1% /dev/shm
tmpfs               5120       12      5108   1% /run/lock
tmpfs             524288     2492    521796   1% /tmp
/dev/mmcblk0p1    522232    63596    458636  13% /boot/firmware
/dev/sda1      117703576       24 111678388   1% /data
tmpfs             388304        0    388304   0% /run/user/1000

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(username) gid=1000(username) groups=1000(username),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),993(gpio),994(i2c),995(spi),999(systemd-journal)

Process info
henrik       966  0.2  0.1  84072  4072 ?        Ss   20:25   0:04 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_asi_single_ccd
henrik       967  0.4  3.2 724932 127456 ?       SNsl 20:25   0:07 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik       968  0.0  0.5  33376 22384 ?        SNs  20:25   0:01 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/username/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
henrik       969  0.0  0.3  41232 12032 ?        S    20:25   0:01 indi_simulator_telescope
henrik       970  0.1  0.7 308972 28748 ?        Sl   20:25   0:02 indi_asi_single_ccd
henrik       980  0.6  5.5 858652 213880 ?       SNl  20:25   0:12 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/username/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
henrik      1393  0.8  3.0 735684 117864 ?       SNl  20:39   0:08 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      1395  2.9  3.6 764092 140152 ?       SNl  20:39   0:27 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      1396  0.0  2.2 724900 89084 ?        SN   20:39   0:00 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      1712  0.3  0.0   6800  3072 pts/0    S+   20:54   0:00 /bin/bash ./indi-allsky/misc/support_info.sh

USB info
Bus 002 Device 002: ID 03c3:224a ZWO ASI224MC
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 003: ID 0781:5581 SanDisk Corp. Ultra
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      123      0 drwxr-xr-x   4 root     root           80 Jan  1  1970 /dev/bus/usb
      124      0 drwxr-xr-x   2 root     root          120 Jan  1  1970 /dev/bus/usb/001
      125      0 crw-rw-r--   1 root     root     189,   0 Mar 18 20:24 /dev/bus/usb/001/001
      143      0 crw-rw-r--   1 root     root     189,   1 Mar 18 20:24 /dev/bus/usb/001/002
      254      0 crw-rw-r--   1 root     root     189,   2 Mar 18 20:24 /dev/bus/usb/001/003
      310      0 crw-rw-rw-   1 root     plugdev  189,   3 Mar 18 20:24 /dev/bus/usb/001/004
      126      0 drwxr-xr-x   2 root     root           80 Jan  1  1970 /dev/bus/usb/002
      127      0 crw-rw-r--   1 root     root     189, 128 Mar 18 20:24 /dev/bus/usb/002/001
[... 567 lines cut ...]
scikit-image==0.22.0
scipy==1.12.0
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.9.0
shapely==2.0.3
six==1.16.0
SQLAlchemy==2.0.28
support-developer==1.0.5
tifffile==2024.2.12
toolz==0.12.1
typing_extensions==4.10.0
uritemplate==4.1.1
urllib3==2.2.1
wcwidth==0.2.13
Werkzeug==3.0.1
WTForms==3.1.2
zarr==2.17.1
zipp==3.18.1

#################################
###     end support info      ###
#################################

```

aaronwmorris, 2024-03-18:
Nothing is unusual there.  That specific error just means that an exposure was received before it was expected.  If you request a 4s exposure, it should be received after 4s.  Some cameras misbehave, which is why the check is in place, but most ZWO cameras are well behaved.

Can you provide the output of `./misc/support_info.sh` of the other system?  Just curious.

henrikeri, 2024-03-18:
Here you go :)

I noticed that when it is not working, the recieved time is aways the same 0.1504 seconds. It is as if the indiserver on the recieveing end does not really expose. As nothing is actually sent back. I was looking at the data moving with tcpdump + bmon. 

```
#################################
### indi-allsky support info  ###
#################################

Distribution: debian
Release: 12
Arch: x86_64
Bits: 64

CPUs: 12
Memory: 65763244 kB

System: Generic PC

Linux server 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux

Uptime
 21:15:17 up 55 min,  1 user,  load average: 0.12, 0.06, 0.01

Filesystems
Filesystem       1K-blocks        Used  Available Use% Mounted on
udev              32852940           0   32852940   0% /dev
tmpfs              6576328        3188    6573140   1% /run
/dev/nvme0n1p1   478612200   199707512  254519052  44% /
tmpfs             32881620         180   32881440   1% /dev/shm
tmpfs                 5120           0       5120   0% /run/lock
tank/data      27939739648 24423538688 3516200960  88% /srv/data
tmpfs              6576324          48    6576276   1% /run/user/1000

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(henrik) gid=1000(henrik) groups=1000(henrik),20(dialout),24(cdrom),25(floppy),29(audio),30(dip),44(video),46(plugdev),101(systemd-journal),108(netdev),1001(mediaStorage),1004(bilder)

Process info
root        1278  0.0  0.0      0     0 ?        SN   20:20   0:00 [z_indirect_cond]
henrik      2380  0.0  0.0   9632  3656 ?        Ss   20:20   0:00 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd
henrik      2382  0.0  0.0  41584 12904 ?        S    20:20   0:00 indi_simulator_telescope
henrik      2383  0.0  0.0 123704 14892 ?        Sl   20:20   0:00 indi_simulator_ccd
henrik      3021  0.0  0.0  32296 23956 ?        SNs  20:20   0:00 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/username/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
henrik      3023  0.1  0.3 1720516 235400 ?      SNl  20:20   0:05 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/username/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
henrik      3194  0.0  0.2 1068216 133900 ?      SNsl 20:22   0:01 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      4735  0.3  0.1 1079020 118060 ?      SNl  20:45   0:06 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      4737  0.8  0.3 1617784 222004 ?      SNl  20:45   0:14 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      4738  0.0  0.1 1068184 90236 ?       SN   20:45   0:00 /home/username/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
henrik      6365  0.0  0.0   6932  3436 pts/0    S+   21:15   0:00 /bin/bash ./indi-allsky/misc/support_info.sh

USB info
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 046d:c328 Logitech, Inc. Corded Keyboard K280e
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      282      0 drwxr-xr-x   6 root     root          120 Mar 18 20:19 /dev/bus/usb
      289      0 drwxr-xr-x   2 root     root           60 Mar 18 20:19 /dev/bus/usb/004
      290      0 crw-rw-r--   1 root     root     189, 384 Mar 18 20:20 /dev/bus/usb/004/001
      287      0 drwxr-xr-x   2 root     root           60 Mar 18 20:19 /dev/bus/usb/003
      288      0 crw-rw-r--   1 root     root     189, 256 Mar 18 20:20 /dev/bus/usb/003/001
      285      0 drwxr-xr-x   2 root     root           60 Mar 18 20:19 /dev/bus/usb/002
      286      0 crw-rw-r--   1 root     root     189, 128 Mar 18 20:20 /dev/bus/usb/002/001
      283      0 drwxr-xr-x   2 root     root           80 Mar 18 20:19 /dev/bus/usb/001
[... 572 lines cut ...]
scikit-image==0.22.0
scipy==1.12.0
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.9.0
shapely==2.0.3
six==1.16.0
SQLAlchemy==2.0.28
support_developer==1.0.5
tifffile==2024.2.12
toolz==0.12.1
typing_extensions==4.10.0
uritemplate==4.1.1
urllib3==2.2.1
wcwidth==0.2.13
Werkzeug==3.0.1
WTForms==3.1.2
zarr==2.17.1
zipp==3.18.1

#################################
###     end support info      ###
#################################

```

aaronwmorris, 2024-03-18:
How many times do you receive that error message during a run?  Is it only one time at the beginning, or constantly?

henrikeri, 2024-03-18:
If the RPi indi-allsky server is _not_ running and looping (meaning I only have indiserver.service running, or a 3600s time span between exposure to avoid crash), it instantly starts failing on the local x64 machine (every single exposure).

aaronwmorris, 2024-03-18:
That is very strange and I would not expect that.  There is practically no difference between connecting to an indiserver locally and remotely.

It is possible that by connecting EKOS, it saved defaults that have persisted and may cause issues.  I would not expect that kind of issue though.

```
systemctl --user stop indiserver

rm $HOME/.indi/*.xml

systemctl --user start indiserver
```

henrikeri, 2024-03-18:
I doubt it. But I will try. The issues started before I ran Ekos. That was just to verify before contacting here. 
I'll try the following on the RPi:

Your steps above
systemctl --user stop indi-allsky.service

This should leave only the indiserver running? I'll then restart the server at home to see what the log shows

aaronwmorris, 2024-03-18:
> Your steps above
> systemctl --user stop indi-allsky.service
> 
> This should leave only the indiserver running? I'll then restart the server at home to see what the log shows

That is correct.

henrikeri, 2024-03-18:
Fails instantly. But it is obviously connecting to the indiserver on the RPi as it is getting the temperature. However it does not expose. If I now go ahead and start the rpi indi-allsky and let it expose. The x64 PC at home also starts exposing 

So weird!

```
2024-03-18T21:43:43.780668+01:00 server [INFO] Capture-1-9010/MainThread indi.getTelescopeRaDec() [853]: Telescope Coord: RA 2.92, Dec 33.00
2024-03-18T21:43:43.680541+01:00 server [INFO] Capture-1-9010/MainThread indi.getCcdTemperature() [868]: Sensor temperature: 16.2
2024-03-18T21:43:43.580344+01:00 server [INFO] Capture-1-9010/MainThread capture.detectMoonMode() [1220]: Moon Mode conditions detected
2024-03-18T21:43:43.580299+01:00 server [INFO] Capture-1-9010/MainThread capture.detectMoonMode() [1216]: Moon altitude: 54:02:33.0, phase 66.9%
2024-03-18T21:43:43.580264+01:00 server [INFO] Capture-1-9010/MainThread capture.detectNight() [1198]: Sun altitude: -21:48:48.9
2024-03-18T21:43:43.580228+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [215]: Exposure state: OK
2024-03-18T21:43:43.580132+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [214]: Camera last ready: 22.0s
2024-03-18T21:43:38.551131+01:00 server [INFO] MainProcess-8994/MainThread allsky._queueManualTasks() [1113]: Checking for manually submitted tasks
2024-03-18T21:43:32.561250+01:00 server [INFO] Capture-1-9010/MainThread indi.getTelescopeRaDec() [853]: Telescope Coord: RA 2.92, Dec 33.00
2024-03-18T21:43:32.461066+01:00 server [INFO] Capture-1-9010/MainThread indi.getCcdTemperature() [868]: Sensor temperature: 16.2
2024-03-18T21:43:32.360869+01:00 server [INFO] Capture-1-9010/MainThread capture.detectMoonMode() [1220]: Moon Mode conditions detected
2024-03-18T21:43:32.360829+01:00 server [INFO] Capture-1-9010/MainThread capture.detectMoonMode() [1216]: Moon altitude: 54:03:24.8, phase 66.9%
2024-03-18T21:43:32.360794+01:00 server [INFO] Capture-1-9010/MainThread capture.detectNight() [1198]: Sun altitude: -21:47:43.0
2024-03-18T21:43:32.360748+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [215]: Exposure state: OK
2024-03-18T21:43:32.360610+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [214]: Camera last ready: 10.8s
2024-03-18T21:43:25.550111+01:00 server [INFO] MainProcess-8994/MainThread allsky._queueManualTasks() [1113]: Checking for manually submitted tasks
2024-03-18T21:43:21.541693+01:00 server [WARNING] Capture-1-9010/MainThread miscDb.addNotification() [857]: Not adding existing notification
2024-03-18T21:43:21.539389+01:00 server [ERROR] Capture-1-9010/MainThread capture.saferun() [348]: 15.0000s EXPOSURE RECEIVED IN 0.1504s.  POSSIBLE CAMERA PROBLEM.
2024-03-18T21:43:21.539324+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [344]: Exposure received in 0.1504 s (-14.8496)
2024-03-18T21:43:21.489029+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [451]: Total time since last exposure 0.2511 s
2024-03-18T21:43:21.488967+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [420]: Image queue depth: 0
2024-03-18T21:43:21.388800+01:00 server [INFO] Capture-1-9010/MainThread capture.shoot() [1404]: Taking 15.00000000 s exposure (gain 100)
2024-03-18T21:43:21.388739+01:00 server [WARNING] Capture-1-9010/MainThread capture.reconfigureCcd() [1164]: Change to night (moon mode)
2024-03-18T21:43:21.338549+01:00 server [INFO] Capture-1-9010/MainThread indi.getTelescopeRaDec() [853]: Telescope Coord: RA 2.92, Dec 33.00
2024-03-18T21:43:21.238367+01:00 server [INFO] Capture-1-9010/MainThread indi.getCcdTemperature() [868]: Sensor temperature: 16.2
2024-03-18T21:43:21.138206+01:00 server [INFO] Capture-1-9010/MainThread capture.detectMoonMode() [1220]: Moon Mode conditions detected
2024-03-18T21:43:21.138162+01:00 server [INFO] Capture-1-9010/MainThread capture.detectMoonMode() [1216]: Moon altitude: 54:04:16.4, phase 66.9%
2024-03-18T21:43:21.138017+01:00 server [INFO] Capture-1-9010/MainThread capture.detectNight() [1198]: Sun altitude: -21:46:37.1
2024-03-18T21:43:21.137630+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [215]: Exposure state: unset
2024-03-18T21:43:21.137580+01:00 server [INFO] Capture-1-9010/MainThread capture.saferun() [214]: Camera last ready: 0.0s
2024-03-18T21:43:21.136527+01:00 server [INFO] Capture-1-9010/MainThread capture._initialize() [834]: Default CCD exposure: 15.00000000
2024-03-18T21:43:21.136480+01:00 server [WARNING] Capture-1-9010/MainThread capture._initialize() [815]: Reusing last stable exposure: 15.000000
2024-03-18T21:43:21.132216+01:00 server [INFO] Capture-1-9010/MainThread capture._initialize() [795]: Maximum CCD exposure: 15.00000000
2024-03-18T21:43:21.132188+01:00 server [INFO] Capture-1-9010/MainThread capture._initialize() [774]: Minimum CCD exposure: 0.00003201 (night)
2024-03-18T21:43:21.132142+01:00 server [INFO] Capture-1-9010/MainThread capture._initialize() [756]: Minimum CCD exposure: 0.00003201 (day)
2024-03-18T21:43:21.132008+01:00 server [INFO] Capture-1-9010/MainThread indi.updateCcdBlobMode() [409]: Set BLOB mode
2024-03-18T21:43:21.131191+01:00 server [INFO] Capture-1-9010/MainThread miscDb.addCamera() [101]: Camera DB ID: 2
2024-03-18T21:43:20.723879+01:00 server [INFO] Capture-1-9010/MainThread indi.getCcdInfo() [532]: Detecting bayer pattern
[... cut ...]
2024-03-18T21:43:12.443853+01:00 server [INFO] MainProcess-8994/MainThread allsky._startup() [228]: indi-allsky release: indi_v2024.02.1
2024-03-18T21:43:11.872953+01:00 server [INFO] MainProcess-3194/MainThread miscDb.addNotification() [871]: Added State notification: 17
2024-03-18T21:43:11.870881+01:00 server [INFO] MainProcess-3194/MainThread allsky._fileUploadWorkerStop() [535]: Stopping Upload worker
2024-03-18T21:43:11.870804+01:00 server [WARNING] MainProcess-3194/Upload-19 uploader.saferun() [120]: Goodbye
```

henrikeri, 2024-03-18:
I re-started indi-allsky on the RPi and it instantly started working on the x64. 
Just to check, me using nginx instead of apache should not impact this, right? 
I deactivated apache and I am using it with this nginx config (which seems to work perfect outside this issue)

```

server {
    listen 80;
    server_name allsky.mydomain.com *.allsky.mydomain.com;

    return 301 https://$host$request_uri;
}

server {
    listen 443 ssl http2;
    listen [::]:443 ssl http2;
    server_name allsky.mydomain.com *.allsky.mydomain.com;

    ssl_certificate /etc/letsencrypt/live/allsky.mydomain.com-0001/fullchain.pem;
    ssl_certificate_key /etc/letsencrypt/live/allsky.mydomain.com-0001/privkey.pem;

    ssl_protocols TLSv1.2 TLSv1.3;
    ssl_prefer_server_ciphers off;
    ssl_session_timeout 1d;
    ssl_session_cache shared:SSL:50m;
    ssl_session_tickets off;

    add_header Strict-Transport-Security "max-age=604800; includeSubDomains" always;

    location / {
        rewrite ^/$ /indi-allsky/ permanent;
    }

    location /indi-allsky {
        proxy_pass http://unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock;
        proxy_http_version 1.1;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_set_header X-Forwarded-SSL on;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_read_timeout 180s;
        proxy_redirect off;
    }

    # Serve static files
    location /static {
        root /home/username/allsky/indi-allsky/indi_allsky/flask;
        expires 1d;
    }

    # Serve images
    location /images {
        root /opt/allsky;
        expires 1d;
    }
}

```

aaronwmorris, 2024-03-18:
Apache vs nginx would not affect anything.

You mentioned using wireguard.  You may need to lower the MTU on your Raspi in case you having fragmentation issues.  Just a guess.

```
sudo ip link set eth0 mtu 1300
```

henrikeri, 2024-03-18:
Already running it at 1280. I can try lowering to 1200

henrikeri, 2024-03-18:
> Apache vs nginx would not affect anything.
> 
> You mentioned using wireguard. You may need to lower the MTU on your Raspi in case you having fragmentation issues. Just a guess.
> 
> ```
> sudo ip link set eth0 mtu 1300
> ```
Changing made no difference. And it seems so strange that it starts working as long as indi-allsky is exposing on the RPi

aaronwmorris, 2024-03-18:
Are you running a lower MTU on the x86_64 system?

henrikeri, 2024-03-18:
> Are you running a lower MTU on the x86_64 system?

I have only tried with 1280 for the wg connection at both ends. Try lower?

henrikeri, 2024-03-23:
@aaronwmorris I noticed that the x64 machine at home does not properly show it's exposure time even when it is working. Could this be related to the exposure problem? How it polls the remote indi driver or similar? I seleceted option 3 indi_passive while installing. Is it possible that changing to option 1 would make it work?

x64:
[image]

RPi
[image]

aaronwmorris, 2024-03-24:
indi passive is for when you have one instance of indi-allsky managing a camera (via indiserver) and you want to connect a second instance of indi-allsky to the same indiserver.  The "passive" instance cannot change the camera gain or exposure settings, it just receives the images initiated by the other camera.  You should be using "indi".

henrikeri, 2024-03-30:
This explains and solves the issue. Thanks for helping out

closed 2024-03-30 by henrikeri