aaronwmorris/indi-allsky #1203: ZWO 678MC Pro not taking any pictures

ericcatta, 2024-03-22:
A couple of days ago I installed the indi-allsky with another ASI camera and it worked just fine. I decided to change camera and the 678 wouldn't show up. I reinstalled everything from scratch (OS included) and still no camera showing up. After trying some reboots, connecting and disconnecting the camera the system would finally recognize it but it won't show any pictures. Could you help me fixing it?

aaronwmorris, 2024-03-22:
Can you post the output of `./misc/support_info.sh`?

ericcatta, 2024-03-22:
I have no idea about what happened but I left home after posting the question and now I just got home and realized that it started taking pictures 10 minutes ago. anyway that's the output it gives me

./misc/support_info.sh: line 59: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245728 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux IndiAllsky 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Uptime
 23:06:06 up  4:52,  3 users,  load average: 0.22, 0.20, 0.19

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3948144        0   3948144   0% /dev
tmpfs             824576     5904    818672   1% /run
/dev/mmcblk0p2 122334472 12363908 103737172  11% /
tmpfs            4122864     1008   4121856   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522230    75372    446858  15% /boot/firmware
tmpfs             824560      144    824416   1% /run/user/1000

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

Process info
pi          1127  0.2  0.0  83328  4336 ?        Ss   18:14   0:36 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_asi_ccd
pi          1143  0.0  0.1  41968 11264 ?        S    18:14   0:03 indi_simulator_telescope
pi          1144  0.2  1.1 405792 97296 ?        Sl   18:14   0:40 indi_asi_ccd
pi          1656  0.0  0.2  32688 23040 ?        SNs  18:14   0:01 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
pi          1658  0.0  2.4 930416 202016 ?       SNl  18:14   0:12 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
pi          2048  0.0  1.5 579152 130528 ?       SNsl 18:16   0:04 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi          2052  1.3  2.0 588112 168896 ?       Sl   18:16   3:53 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi          2054  5.2  3.5 767568 292704 ?       Sl   18:16  15:03 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi          2055  0.4  1.6 607712 139904 ?       SNl  18:16   1:23 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run

USB info
Bus 004 Device 002: ID 03c3:678b ZWO ASI678MC
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      133      0 drwxr-xr-x   6 root     root          120 Mar 22 18:14 /dev/bus/usb
      134      0 drwxr-xr-x   2 root     root           60 Mar 22 18:14 /dev/bus/usb/001
      135      0 crw-rw-r--   1 root     root     189,   0 Mar 22 18:14 /dev/bus/usb/001/001
      136      0 drwxr-xr-x   2 root     root           60 Mar 22 18:14 /dev/bus/usb/002
      137      0 crw-rw-r--   1 root     root     189, 128 Mar 22 18:14 /dev/bus/usb/002/001
      138      0 drwxr-xr-x   2 root     root           60 Mar 22 18:14 /dev/bus/usb/003
      139      0 crw-rw-r--   1 root     root     189, 256 Mar 22 18:14 /dev/bus/usb/003/001
      140      0 drwxr-xr-x   2 root     root           80 Mar 22 18:14 /dev/bus/usb/004
      141      0 crw-rw-r--   1 root     root     189, 384 Mar 22 18:14 /dev/bus/usb/004/001
      355      0 crw-rw-rw-   1 root     root     189, 385 Mar 22 23:06 /dev/bus/usb/004/002

video device Permissions
[... 598 lines cut ...]
reproject==0.13.0
requests==2.31.0
requests-toolbelt==1.0.0
scikit-image==0.22.0
scipy==1.12.0
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.9.0
shapely==2.0.3
six==1.16.0
SQLAlchemy==2.0.28
support-developer==1.0.5
tifffile==2024.2.12
toolz==0.12.1
typing_extensions==4.10.0
urllib3==2.2.1
wcwidth==0.2.13
Werkzeug==3.0.1
WTForms==3.1.2
zarr==2.17.1
zipp==3.18.1

#################################
###     end support info      ###
#################################

aaronwmorris, 2024-03-24:
I wonder if maybe it started under the wrong timezone and then perhaps you changed it?  That could cause problems with querying images until the times were aligned.

Anyway, I am glad it is working!

closed 2024-04-28 by aaronwmorris

Errabundo66, 2024-05-17:
Hello.
I had the same problem for weeks and reinstalled all with no success.
But tonight I sent the order "timedatectl set-ntp true" at the CLI, and all works again!
Thank you!!