aaronwmorris/indi-allsky #1206: setup.sh reports raspbian 11 as unknown distribution

caphector, 2024-03-26:
I ran ./setup.sh and got this:

```
./setup.sh
###############################################
### Welcome to the indi-allsky setup script ###
###############################################

Detected a custom installation of INDI in /usr/local/bin

Distribution: raspbian
Release: 11
Arch: aarch64
Bits: 32

CPUs: 4
Memory: 3834296 kB

INDI_DRIVER_PATH: /usr/local/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
DB_FILE: /var/lib/indi-allsky/indi-allsky.sqlite
INSTALL_INDI: false
HTTP_PORT: 80
HTTPS_PORT: 443

Setup proceeding in 10 seconds... (control-c to cancel)

indi-allsky supports the following camera interfaces.

          indi: For astro/planetary cameras normally connected via USB
     libcamera: Supports cameras connected via CSI interface on Raspberry Pi SoCs
  indi_passive: Connect a second instance of indi-allsky to an existing indi-allsky indiserver
 pycurl_camera: Download images from a remote web camera

1) indi
2) libcamera
3) indi_passive
4) pycurl_camera
Select a camera interface: 2

1) libcamera_imx477	     3) libcamera_ov5647	 5) libcamera_imx519	     7) libcamera_imx296_gs	 9) libcamera_imx462
2) libcamera_imx378	     4) libcamera_imx219	 6) libcamera_imx708	     8) libcamera_imx290	10) libcamera_64mp_hawkeye
Select a libcamera interface: 1

Fixing git checkout permissions
**** Installing packages... ****
Unknown distribution raspbian 11 (aarch64)
```

I think Raspbian is supported; at least on my machine it's `raspbian`. I patched my scripts with a simple `s/Raspbian/raspbian/` but I didn't test further after getting setup to run.

aaronwmorris, 2024-03-27:
Sorry about that.  I recently updated the OS detection and I missed the update to the Raspbian name.  Merged #1207

caphector, 2024-03-28:
Looks good now! 

As a note, is there any way that setup.sh could be aware of my config? It'd be nice to not have to answer all the questions when I'm running it in an update context.

aaronwmorris, 2024-03-29:
This is a tough one.  The questions are important to reconfigure systems or fix issues that might have occurred.

I have updated the wiki page on updating to include the manual steps that should work most of the time to upgrade the system.

https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

closed 2024-04-08 by caphector