aaronwmorris/indi-allsky #1225: SVBONY SV205 *easy issue* camera is unavailable

Binkosss, 2024-04-18:
Hi

very simple installation and I failed on one of the first steps, perfect.

the camera is not available(0 in indi allsky panel), but visible in lsusb, I did the entire installation several times, including rasbian, it is visible and working correctly in astrodmx capture.

Current configuration:
-Raspberry 4B 4 GB
-SV205
-original power supply
-system is 64-bit Bookworm (latest)

any ideas ? it must be some simple glitch

[image]

[image]

Binkosss, 2024-04-18:
Previously I selected Svbony_ccd and now I tried to reinstall Setup.sh and select indi_v4l2_ccd... it still doesn't work, maybe reinstall all ?

aaronwmorris, 2024-04-18:
When you select a new server, you will need to restart the indiserver or reboot.

```
systemctl --user restart indiserver
```

aaronwmorris, 2024-04-19:
Sorry, the link was not intended to provide instructions, but just a reference for the sv205 being a UVC camera.

Can you provide the output by running the `./misc/support_info.sh` script?

Binkosss, 2024-04-19:
yeah, no problem:

[New.txt](https://github.com/aaronwmorris/indi-allsky/files/15043545/New.txt)

aaronwmorris, 2024-04-19:
I suspect you my have multiple v4l2 devices and indi might be using the wrong one.  Can you provide the output of `v4l2-ctl --list-devices` ?

Binkosss, 2024-04-19:
[New1.txt](https://github.com/aaronwmorris/indi-allsky/files/15044353/New1.txt)

aaronwmorris, 2024-04-19:
Everything appears in order.  The SV205 is /dev/video0 which is what indi is supposed to be using.

Can you manually start indi-allsky and show me the first 50-100 or so lines of output?

```
source virtualenv/indi-allsky/bin/activate

systemctl --user stop indi-allsky

./allsky.py run
```

Binkosss, 2024-04-19:
earching for available cameras
Detected V4L2 CCD
Found 1 CCDs
Camera error: !!! CAMERA NOT FOUND: SVBONY !!!

when try to turn on camera then disconnects the camera? the power supply is 100% ok, the camera is seen normally by the astrodmx capture program. video and info like this:

[image]

all Terminal: 
[New3.txt](https://github.com/aaronwmorris/indi-allsky/files/15045187/New3.txt)

aaronwmorris, 2024-04-19:
Ah, I think I know the problem.  I think you defined a specific camera name to connect to in the Camera tab on the config page (near the top).  Set that field to blank, save, and then reboot.

Binkosss, 2024-04-19:
still the same and i make save and restart RPI.:
[image]

aaronwmorris, 2024-04-19:
FYI, it takes 2-3 minutes for the indi-allsky service to start from boot.  If it still is not showing data, try running the service manually again and post 50-100 lines.

Binkosss, 2024-04-20:
After reinstalling everything and selecting indi_v4l2_ccd it works great, as I said, easy problem :D Thank you very much, can I count on help if I still have a problem? write here or start new threads?

[image]

Have a nice day ! and clear sky !

aaronwmorris, 2024-04-20:
I will leave it up to you if you want to open a different issue.  Do whatever makes sense.

If you can send me the output of `indi_getprop` I can send you the config you need to add to the camera to get it to run in full resolution.

Binkosss, 2024-04-21:
I thing max is "3264 x 2448"...that's enough :D you see something else ? 

{
    "PROPERTIES": {},
    "SWITCHES": {
        "V4L2_SIZE_DISCRETE": {
            "on": [
                "3264 x 2448"
            ]
        }
    }
}

and there is all terminal:
[New.txt](https://github.com/aaronwmorris/indi-allsky/files/15052477/New.txt)

aaronwmorris, 2024-04-21:
Looks like you figured it out.  I appreciate the info anyway.  This confirms something I recently "discovered" about the SV205, but I did not have a device to confirm.

I had to setup 2 different "cameras" in the Camera Simulator.  The SV205 max resolution is 3264x2448 (here) but the IMX415 datasheet says the sensor is 3864x2160.  It looks like SVBony cut off 300 pixels from each side of the sensor.

Now that I see your information, I do not understand how the height of the sensor is 2448, when both SVBony and the datasheet say it is 2160.  Very strange.

This page even shows the discrepancy in the resolutions as it lists two different max resolutions.  https://xinchenguang.en.made-in-china.com/product/YnzRLfQDRFkP/China-Imx415-CMOS-Sensor-Face-Recognition-Wide-Angle-4K-8MP-3264-2448-30fps-Hdr-Starvis-USB-Camera-Module.html

Binkosss, 2024-04-22:
yes, but I have a problem, I thought that it would be configured and that would be enough, unfortunately, after a whole day of work, the camera does not want to switch to *4K* mode, only the default 640 x 480, do you have any ideas what to enter to make it work in higher resolution than 1080p (the entry on your website works All right( "V4L2_SIZE_DISCRETE": { "on": ["1920x1080"]))

Binkosss, 2024-04-22:
OK, I test it: "2592x1944" works fine, mayby its max fot this setup.

Binkosss, 2024-04-22:
"Looks like you figured it out." but I couldn't have done it without you! Thank you !

aaronwmorris, 2024-04-22:
Interesting.  According to the indi properties, any of the following values should work:

```
640x480
3264x2448
2592x1944
1920x1080
1600x1200
1280x720
960x540
848x480
640x360
424x240
320x240
320x180
640x480
```

I do not know why 640x480 is listed twice, it could be that the different settings are achieved via binning vs cropping.

Anyway, if 3264x2448 does not work, you may try 2592x1944 as the next size down.

Binkosss, 2024-04-25:
2592x1944 works fine, enough for me, maybe I'll play with it later when it's hanging on the roof ;)

thank you for your help, I'm closing the topic.

closed 2024-04-25 by Binkosss

blablazzz, 2024-08-27:
Where to select proper camera type for this camera? i dont see indi_v4l2_ccd