aaronwmorris/indi-allsky #1227: Graphs stopped working ?!

afjkeese, 2024-04-21:
wonderful product and great documentation, Aaron - really a superb piece of work you have created here!

BUT, After running it for a few days and night successfully, the creation of graphs stopped working on the respective tab?!

I am on a a raspi 4 with 4GB and 128GB ssd.
Image taking, loops and timelapses all work fine - until yesterday so did the graphs generation, but then stopped. Restart or reboot did not help ?!

Also I wonder is there a way to get more SQM and star cound graphs from the database than just 4 hours back?

Arny

Here my setup:

[image]

aaronwmorris, 2024-04-21:
Do you mean the graphs stopped working on the Charts page or on the System page?

I can always extend the max timeframe of the Charts to larger windows.  It is just a matter of adjusting the SQL queries.

afjkeese, 2024-04-21:
They are missing on The Charts Page: See Screenshot below.
I Wonder WhatsApp triggered them to stop working …

[image]

aaronwmorris, 2024-04-22:
Its possible the AJAX call is failing in the background.  Can you report any exceptions being generated in `/var/log/indi-allsky/webapp-indi-allsky.log` when you view the chart view?

afjkeese, 2024-04-22:
> Its possible the AJAX call is failing in the background. Can you report any exceptions being generated in `/var/log/indi-allsky/webapp-indi-allsky.log` when you view the chart view?

yes, here it is:

2024-04-22T06:32:56.641468+02:00 allsky [INFO] MainProcess-2625 views.getChartData() [1021]: Image read in 0.0299 s
2024-04-22T06:32:56.641757+02:00 allsky [INFO] MainProcess-2625 views.getChartData() [1025]: Calculating histogram from RoI
2024-04-22T06:32:56.641867+02:00 allsky [WARNING] MainProcess-2625 base_views._load_detection_mask() [221]: No detection mask defined
2024-04-22T06:32:56.644244+02:00 allsky [ERROR] MainProcess-2625 app.log_exception() [838]: Exception on /indi-allsky/js/charts [GET]#012Traceback (most recent call last):#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012         ^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/base_views.py", line 753, in dispatch_request#012    json_data = self.get_objects()#012                ^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/views.py", line 880, in get_objects#012    'chart_data' : self.getChartData(ts_dt, history_seconds),#012                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/views.py", line 1037, in getChartData#012    x1 = int(sqm_roi[0] / self.bin_v.value)#012                          ^^^^^^^^^^#012AttributeError: 'JsonChartView' object has no attribute 'bin_v'
2024-04-22T06:33:11.916705+02:00 allsky [INFO] MainProcess-2625 views.getChartData() [1021]: Image read in 0.0248 s
2024-04-22T06:33:11.916959+02:00 allsky [INFO] MainProcess-2625 views.getChartData() [1025]: Calculating histogram from RoI
2024-04-22T06:33:11.917055+02:00 allsky [WARNING] MainProcess-2625 base_views._load_detection_mask() [221]: No detection mask defined
2024-04-22T06:33:11.918739+02:00 allsky [ERROR] MainProcess-2625 app.log_exception() [838]: Exception on /indi-allsky/js/charts [GET]#012Traceback (most recent call last):#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012
^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/base_views.py", line 753, in dispatch_request#012    json_data = self.get_objects()#012                ^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/views.py", line 880, in get_objects#012    'chart_data' : self.getChartData(ts_dt, history_seconds),#012                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/allsky/indi-allsky/indi_allsky/flask/views.py", line 1037, in getChartData#012    x1 = int(sqm_roi[0] / self.bin_v.value)#012
    ^^^^^^^^^^#012AttributeError: 'JsonChartView' object has no attribute 'bin_v'

(sorry for the garble, but it comes out with few linebreaks in the log)

aaronwmorris, 2024-04-22:
Merged #1231 

Okay, I suspect you are using a RoI for the detection mask area and I had a condition that caused that to fail.

Pull down the latest code and this should fix the issue.  It also includes an update to show an error when the data fails to load.

afjkeese, 2024-04-22:
> Merged #1231
> 
> Okay, I suspect you are using a RoI for the detection mask area and I had a condition that caused that to fail.
> 
> Pull down the latest code and this should fix the issue. It also includes an update to show an error when the data fails to load.

Wow, eliminating the ROI worked immediately - great & thanks!
I will upgrade asap nevertheless.

Arny

closed 2024-04-22 by afjkeese

hedrot, 2025-02-13:
Hello Arron. I'm not sure if I am having a similar problem. The heading at the top of the "Charts" page says "No chart data in history range" and then the sensor readings (temp ++) dissappear from the "Device" page. I have experienced this for some weeks and am able to reset it sometimes by reloading the configuration and sometime restarting the rPI5. Every now and then I have to power off and restart. I have the latest software from Github since last week. Any thoughts?

[image]