aaronwmorris/indi-allsky #1228: Creation of darks fails (ValueError)

WirthmU, 2024-04-21:
I tried to create darks using the procedure described (https://github.com/aaronwmorris/indi-allsky/wiki/Dark-Calibration-Frames).

It sees pictures are taken but at the end I get a "ValueError":

```
(indi-allsky) pi@meteotux:~/indi-allsky $ systemctl --user stop indi-allsky
Warning: Stopping indi-allsky.service, but it can still be activated by:
  indi-allsky.timer
(indi-allsky) pi@meteotux:~/indi-allsky $ ./darks.py sigmaclip
2024-04-21 15:33:53,991 [INFO] MainProcess indi.__init__() [131]: creating an instance of IndiClient
2024-04-21 15:33:53,992 [INFO] MainProcess indi.__init__() [138]: PyIndi version: 2.0.6
2024-04-21 15:33:53,993 [INFO] MainProcess darks._initialize() [201]: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2024-04-21 15:33:53,994 [INFO] MainProcess indi.serverConnected() [358]: Server connected (localhost:7624)
2024-04-21 15:33:53,997 [INFO] MainProcess indi.newDevice() [226]: new device Telescope Simulator
2024-04-21 15:33:54,002 [INFO] MainProcess indi.newDevice() [226]: new device CCD Simulator
2024-04-21 15:34:01,994 [WARNING] MainProcess darks._initialize() [230]: Connecting to device libcamera_imx477
BaseClientQt: Error. Unable to find driver libcamera_imx477
2024-04-21 15:34:02,038 [INFO] MainProcess miscDb.addCamera() [101]: Camera DB ID: 1
2024-04-21 15:34:02,038 [INFO] MainProcess indi.updateCcdBlobMode() [424]: Set BLOB mode
2024-04-21 15:34:02,039 [ERROR] MainProcess darks._initialize() [319]: CCD night gain above maximum, changing to 22
2024-04-21 15:34:05,039 [ERROR] MainProcess darks._initialize() [328]: CCD moon mode gain above maximum, changing to 22
2024-04-21 15:34:08,039 [ERROR] MainProcess darks._initialize() [333]: CCD day gain below minimum, changing to 1
2024-04-21 15:34:11,040 [WARNING] MainProcess darks._run() [746]: ****** IF THE CCD COOLER WAS ENABLED, YOU MAY CONSIDER STOPPING THIS UNTIL THE SENSOR HAS WARMED ******
2024-04-21 15:34:19,041 [INFO] MainProcess darks._run() [755]: Processing 8 darks, 10 exposures each. Estimated time left: 00:14:20
2024-04-21 15:34:19,043 [INFO] MainProcess darks._take_exposures() [835]: Temp folder: /tmp/tmpvhv1hrhw
2024-04-21 15:34:19,044 [INFO] MainProcess darks._take_exposures() [842]: Starting image 1/10.
2024-04-21 15:34:19,044 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:34:19,046 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmpqq5fnmym.json --metadata-format json --awbgains 1,1 --output /tmp/tmp6coshfup.jpg
2024-04-21 15:34:35,844 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.7998s (1.7998)
2024-04-21 15:34:39,826 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.34
2024-04-21 15:34:39,827 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 39.00
2024-04-21 15:34:39,827 [INFO] MainProcess darks._take_exposures() [842]: Starting image 2/10.
2024-04-21 15:34:39,827 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:34:39,829 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmpw_7kkmxz.json --metadata-format json --awbgains 1,1 --output /tmp/tmpnzaxy4cn.jpg
2024-04-21 15:34:56,577 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.7498s (1.7498)
2024-04-21 15:35:00,175 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.34
2024-04-21 15:35:00,175 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 40.00
2024-04-21 15:35:00,176 [INFO] MainProcess darks._take_exposures() [842]: Starting image 3/10.
2024-04-21 15:35:00,176 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:35:00,177 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmp9re_tgn3.json --metadata-format json --awbgains 1,1 --output /tmp/tmpi_xjcx78.jpg
2024-04-21 15:35:16,823 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.6468s (1.6468)
2024-04-21 15:35:20,479 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.34
2024-04-21 15:35:20,480 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 40.00
2024-04-21 15:35:20,480 [INFO] MainProcess darks._take_exposures() [842]: Starting image 4/10.
2024-04-21 15:35:20,480 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:35:20,481 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmpzd6yr86v.json --metadata-format json --awbgains 1,1 --output /tmp/tmppvc2opk8.jpg
2024-04-21 15:35:37,079 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.5987s (1.5987)
2024-04-21 15:35:40,677 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.34
2024-04-21 15:35:40,677 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 40.00
2024-04-21 15:35:40,677 [INFO] MainProcess darks._take_exposures() [842]: Starting image 5/10.
2024-04-21 15:35:40,677 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:35:40,678 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmped4tb405.json --metadata-format json --awbgains 1,1 --output /tmp/tmpeijt549d.jpg
2024-04-21 15:35:57,376 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.6987s (1.6987)
2024-04-21 15:36:00,977 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.34
2024-04-21 15:36:00,977 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 40.00
2024-04-21 15:36:00,977 [INFO] MainProcess darks._take_exposures() [842]: Starting image 6/10.
2024-04-21 15:36:00,977 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:36:00,978 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmplrzaux8q.json --metadata-format json --awbgains 1,1 --output /tmp/tmpbkqvgsmw.jpg
2024-04-21 15:36:17,674 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.6965s (1.6965)
2024-04-21 15:36:21,276 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.33
2024-04-21 15:36:21,276 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 41.00
2024-04-21 15:36:21,276 [INFO] MainProcess darks._take_exposures() [842]: Starting image 7/10.
2024-04-21 15:36:21,276 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
2024-04-21 15:36:21,277 [INFO] MainProcess libcamera.setCcdExposure() [251]: image command: rpicam-still --immediate --nopreview --encoding jpg --quality 95 --gain 1 --shutter 15000000 --metadata /tmp/tmpw2ble_gb.json --metadata-format json --awbgains 1,1 --output /tmp/tmpzzxmhvcz.jpg
2024-04-21 15:36:37,874 [INFO] MainProcess darks._take_exposures() [852]: Exposure received in 16.5978s (1.5978)
2024-04-21 15:36:41,506 [INFO] MainProcess darks._take_exposures() [886]: Image average adu: 0.34
2024-04-21 15:36:41,506 [INFO] MainProcess darks._take_exposures() [889]: Sensor temperature: 41.00
2024-04-21 15:36:41,507 [INFO] MainProcess darks._take_exposures() [842]: Starting image 8/10.
2024-04-21 15:36:41,507 [INFO] MainProcess darks.shoot() [343]: Taking 15.00000000 s exposure (gain 1)
[... 18 lines cut ...]
2024-04-21 15:37:43,346 [INFO] MainProcess darks.buildBadPixelMap() [1191]: Master BPM average adu: 0.00
2024-04-21 15:37:47,207 [INFO] MainProcess darks.stack() [1252]: Stacking dark frames for exposure 15.0s, gain 1, bin 1
INFO: splitting each image into 9 chunks to limit memory usage to 350000000 bytes. [ccdproc.combiner]
[2024-04-21 15:37:47,493] [INFO] MainProcess-1548 combiner.combine() [883]: splitting each image into 9 chunks to limit memory usage to 350000000 bytes.
Traceback (most recent call last):
[... cut ...]
    dark_adu_avg = s.stack(tmp_fit_dir_p, full_dark_filename_p, exposure_f, image_bitpix)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 1266, in stack
    combined_dark = ccdproc.combine(
                    ^^^^^^^^^^^^^^^^
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/ccdproc/combiner.py", line 885, in combine
    xs, ys = ccd.data.shape
    ^^^^^^
ValueError: too many values to unpack (expected 2)
```
Any help to this?

aaronwmorris, 2024-04-21:
If you use JPEG mode instead of RAW mode, you may only use `average` for the creation of darks as the sigma clipping module cannot process RGB data.  Sigma clipping only works with RAW data.

WirthmU, 2024-04-22:
Thank you, changing to RAW did the trick.

closed 2024-05-26 by aaronwmorris