aaronwmorris/indi-allsky #1229: Image loop out of order

smokedsalmonbagel, 2024-04-21:
Maybe I am missing something but when I click "Loop" on the left hand menu the images being played back seems to skip around quite a bit.  It will play in order ben then go the other direction and speed up / slow down etc.  The frame counter in the lower right shows the same, counting up, down etc.

aaronwmorris, 2024-04-22:
When you first load the loop, if the images load slowly, they will appear to load out of order and skip around until all of the images are completely downloaded.

This is an unfortunate side effect of the browser.  The display of the images is asynchronous from the download of the images.  The javascript is telling the browser to show an image, the browser starts downloading the image, but the next image is shown before the first finished downloading, repeat for each image.

smokedsalmonbagel, 2024-04-22:
Ok, that's what I suspected.  Also what is "Rock"?  Sorry if I missed this in the wiki.

mburge, 2024-04-26:
And here I've been working on decoding ROCK as an acronym for months. Reverse Order of Capture K?!

closed 2024-05-26 by aaronwmorris