aaronwmorris/indi-allsky #1233: Internal Server Error

Dentonknifeworks, 2024-04-23:
I have 2 that have been running for months. Today I went to the System menu and get The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

I have re-boot both systems with the same issue. I did a git pull on one of them and ran the setup. Same error and now I cant stop the service as it says the service is not running.

Not sure what to do next.
Thanks
Brad

aaronwmorris, 2024-04-23:
Can you show any exceptions from `/var/log/indi-allsky/webapp-indi-allsky.log` ?

Dentonknifeworks, 2024-04-23:
Apr 23 10:37:07 allskyDevHQ [ERROR] MainProcess-1692 app.log_exception() [828]: Exception on /indi-allsky/system [GET]#012Traceback (most recent call last):#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1463, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 872, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 870, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 855, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_login/utils.py", line 290, in decorated_view#012    return current_app.ensure_sync(func)(*args, **kwargs)#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 110, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 274, in dispatch_request#012    context = self.get_context()#012  File "/home/pi/indi-allsky/indi_allsky/flask/views.py", line 2565, in get_context#012    import boto3#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/boto3/__init__.py", line 17, in <module>#012    from boto3.session import Session#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/boto3/session.py", line 17, in <module>#012    import botocore.session#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/session.py", line 26, in <module>#012    import botocore.client#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/client.py", line 15, in <module>#012    from botocore import waiter, xform_name#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/waiter.py", line 18, in <module>#012    from botocore.docs.docstring import WaiterDocstring#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/docs/__init__.py", line 15, in <module>#012    from botocore.docs.service import ServiceDocumenter#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/docs/service.py", line 14, in <module>#012    from botocore.docs.client import ClientDocumenter, ClientExceptionsDocumenter#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/docs/client.py", line 17, in <module>#012    from botocore.docs.example import ResponseExampleDocumenter#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/docs/example.py", line 13, in <module>#012    from botocore.docs.shape import ShapeDocumenter#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/docs/shape.py", line 19, in <module>#012    from botocore.utils import is_json_value_header#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/utils.py", line 37, in <module>#012    import botocore.httpsession#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/botocore/httpsession.py", line 45, in <module>#012    from urllib3.contrib.pyopenssl import (#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/urllib3/contrib/pyopenssl.py", line 50, in <module>#012    import OpenSSL.crypto#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/OpenSSL/__init__.py", line 8, in <module>#012    from OpenSSL import SSL, crypto#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/OpenSSL/SSL.py", line 276, in <module>#012    class X509VerificationCodes:#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/OpenSSL/SSL.py", line 288, in X509VerificationCodes#012    ERR_UNABLE_TO_GET_ISSUER_CERT = _lib.X509_V_ERR_UNABLE_TO_GET_ISSUER_CERT#012AttributeError: module 'lib' has no attribute 'X509_V_ERR_UNABLE_TO_GET_ISSUER_CERT'

Dentonknifeworks, 2024-04-23:
I checked the other one and found the same error.

aaronwmorris, 2024-04-23:
Assuming you do not use the Oracle OCI object storage, let's try to remove that python module and re-run setup.

Make sure you install the optional modules as well.

```
source virtualenv/indi-allsky/bin/active

pip uninstall oci

deactivate

./setup.sh
```

Dentonknifeworks, 2024-04-23:
Done and Same issue

Dentonknifeworks, 2024-04-23:
On the second one I am getting this error.
ERROR: indi-allsky is running.  Please stop the service before running this script.

    systemctl --user stop indi-allsky

And when I run systemctl stop indi-allsky I get Failed to stop indi-allsky.service: Unit indi-allsky.service not loaded.

I am getting this on both did the service name change?
This looks like two issues I think.

aaronwmorris, 2024-04-23:
The indi-allsky.service systemd unit is defined under the user bus and not the system bus which requires the `--user` flag to reference.

Let's just rebuild the virtualenv.

```
source virtualenv/indi-allsky/bin/activate

pip cache purge

deactivate

rm -fR virtualenv/indi-allsky

./setup.sh
```

Dentonknifeworks, 2024-04-23:
I am in a loop I cant sto indi-allsky and setup says its running.
[image]

Dentonknifeworks, 2024-04-23:
Got around it with a reboot.

aaronwmorris, 2024-04-23:
Just make sure to include the `--user` flag with the systemctl commands.

Dentonknifeworks, 2024-04-23:
Got one of them working trying the other one now.

Dentonknifeworks, 2024-04-23:
Second one after the commands now has the error Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application. on the main page

Dentonknifeworks, 2024-04-23:
Nope error on the Second one is now on all pages.

aaronwmorris, 2024-04-23:
Can I get the output from `./misc/support_info.sh` ?

Dentonknifeworks, 2024-04-23:
./support_info.sh: line 59: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: raspbian
Release: 11
Arch: armv7l
Bits: 32

CPUs: 4
Memory: 1861900 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allskyDevHQ 6.1.34-v7l+ #1 SMP Wed Jun 21 08:37:38 UTC 2023 armv7l GNU/Linux

Uptime
 18:26:26 up  1:27,  3 users,  load average: 0.31, 0.35, 0.45

Memory
               total        used        free      shared  buff/cache   available
Mem:         1861900      546352      377388       94860      938160     1131076
Swap:        2097148           0     2097148

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       60778332 51984016   5674496  91% /
devtmpfs          667780        0    667780   0% /dev
tmpfs             930948    10372    920576   2% /dev/shm
tmpfs             372380     1256    371124   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs              76800        4     76796   1% /home/pi/allsky/tmp
/dev/mmcblk0p1    258095    53046    205049  21% /boot
tmpfs             186188       28    186160   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),33(www-data),44(video),46(plugdev),60(games),100(users),101(systemd-journal),104(input),106(render),108(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)

Process info
pi         794  0.1  0.1   8420  2552 ?        Ss   16:59   0:06 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd
pi         800  0.0  0.4  32192  9040 ?        S    16:59   0:04 indi_simulator_telescope
pi         801  0.0  0.4  49644  9028 ?        Sl   16:59   0:03 indi_simulator_ccd
pi        1277  0.0  0.7  23184 14396 ?        SNs  16:59   0:01 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
pi        1279  0.1  4.5 391400 84296 ?        SNLl 16:59   0:05 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
pi        1801  0.1  5.1 390864 96404 ?        SNLsl 17:01   0:07 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        1805  0.4  3.8 383228 70920 ?        Sl   17:01   0:24 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        1807  8.3  5.7 454016 107764 ?       Rl   17:01   7:07 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        1808  0.1  4.4 393184 82000 ?        SN   17:01   0:08 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run

USB info
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      123      0 drwxr-xr-x   4 root     root           80 Dec 31  1969 /dev/bus/usb
      126      0 drwxr-xr-x   2 root     root           60 Dec 31  1969 /dev/bus/usb/002
      127      0 crw-rw-r--   1 root     root     189, 128 Apr 23 16:58 /dev/bus/usb/002/001
      124      0 drwxr-xr-x   2 root     root           80 Dec 31  1969 /dev/bus/usb/001
      136      0 crw-rw-r--   1 root     root     189,   1 Apr 23 16:58 /dev/bus/usb/001/002
      125      0 crw-rw-r--   1 root     root     189,   0 Apr 23 16:58 /dev/bus/usb/001/001
[... 603 lines cut ...]
scikit-image==0.19.3
scipy==1.13.0
semantic-version==2.10.0
sep==1.2.1
setuptools-rust==1.9.0
shapely==2.0.4
six==1.16.0
SQLAlchemy==2.0.29
support-developer==1.0.5
tifffile==2024.4.18
tomli==2.0.1
toolz==0.12.1
typing_extensions==4.11.0
uritemplate==4.1.1
urllib3==1.26.18
wcwidth==0.2.13
Werkzeug==3.0.2
WTForms==3.1.2
zarr==2.17.2
zipp==3.18.1

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-04-24:
I am wondering if the exception has changed in `/var/log/indi-allsky/webapp-indi-allsky.log` ?

Dentonknifeworks, 2024-04-24:
Apr 23 18:28:08 allskyDevHQ [ERROR] MainProcess-1279 app.log_exception() [838]: Exception on /indi-allsky/ [GET]#012Traceback (most recent call last):#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012  File "/home/pi/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 107, in view#012    self = view.view_class(  # type: ignore[attr-defined]#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 258, in __init__#012    super(TemplateView, self).__init__(**kwargs)#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 63, in __init__#012    self.getSunSetDate()#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 195, in getSunSetDate#012    obs.elevation = self.camera.elevation#012TypeError: Elevation must be numeric

Dentonknifeworks, 2024-04-24:
I did change

closed 2024-04-24 by Dentonknifeworks