aaronwmorris/indi-allsky #1239: bloom ccd

fcbarclo, 2024-04-28:
Using Oculus Pro 180 all sky camera , on a raspberry pi 4 8gb ram, ssd disk,  and bullseye os. Latest version of indi allsky.
During day it shows multi-columns bloom on the area overexposed (it seems that there is not any setting to manage overexpose for this camera). 
This multi-columns 
[image]
increases during day , up to completely fill the image (mainly if there are clouds), like this
[image]
or this :
[image]

During night, it is fine for the most cases, but sometimes i will get image like:
[image]

All settings relative to, min, max, default exposure are set to 0 (auto) 

Franco

aaronwmorris, 2024-04-28:
Thank you for being patient.  I had to complete a merge for this.

If you could update the the latest code, I added a new script to explore all of the properties available for a camera.

https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

Once updated and restarted, please report the output of the following commands:

```
./misc/support_info.sh
```

```
source virtualenv/indi-allsky/bin/active

./misc/camera_properties.py
```

fcbarclo, 2024-04-28:
Following  is the support_info.sh output:
[support_info.txt](https://github.com/aaronwmorris/indi-allsky/files/15144231/support_info.txt)

and the output of the camera_properties.py:
[camera_properties.txt](https://github.com/aaronwmorris/indi-allsky/files/15144240/camera_properties.txt)

aaronwmorris, 2024-04-29:
I was hoping there would be something... anything... that could be changed, but there is nothing available.  This mirrors my experience with other Starlight Xpress cameras.  They have a fixed gain and almost no options to make any other adjustments.  It is very unfortunate because the nighttime performance is quite good (with stretching).

closed 2024-07-05 by aaronwmorris