aaronwmorris/indi-allsky #1244: Best way to refresh the database if images removed

smokedsalmonbagel, 2024-05-01:
I am syncing detection images to an archival server and want to remove the old media after a certain number of days.  I couldn't find a way to remove old images in indie-allsky.  If I remove the files do they ever get removed from the database?  Is there a procedure for checking for deleted files and removing them?

aaronwmorris, 2024-05-01:
If you manually delete files, you can use the `Validate DB` option under System -> Utilities to clean up orphaned database database entries.

smokedsalmonbagel, 2024-05-01:
Thanks. Exactly what I was looking for.  I invoked this with the following:
```
cd ~/indi-allsky/virtualenv/indi-allsky
source bin/activate
python3 ~/indi-allsky/misc/validate_db_entries.py -y
```
To run this from a script I added a `-y` option:

https://gist.github.com/smokedsalmonbagel/93d11adf1d3e94c5745e3ffd9733866f

aaronwmorris, 2024-05-01:
Yep, that script performs the same exact function.  I just did not mention it for the sake of simplicity.

closed 2024-05-22 by aaronwmorris