aaronwmorris/indi-allsky #1249: Focusing with a 28BYJ

Mooflotic, 2024-05-06:
Add a focusing routine to drive a simple 28BYJ.
I currently manually drive through a python script via SSL:

`import curses
import RPi.GPIO as GPIO
from time import sleep

# Definizione dei pin del motore
IN1 = 17
IN2 = 18
IN3 = 27
IN4 = 22

# Sequenze per il motore 28BYJ
SEQ = [
    [1, 0, 0, 1],
    [1, 0, 0, 0],
    [1, 1, 0, 0],
    [0, 1, 0, 0],
    [0, 1, 1, 0],
    [0, 0, 1, 0],
    [0, 0, 1, 1],
    [0, 0, 0, 1]
]

# Inizializzazione dei GPIO
def setup():
    GPIO.setmode(GPIO.BCM)
    GPIO.setup(IN1, GPIO.OUT)
    GPIO.setup(IN2, GPIO.OUT)
    GPIO.setup(IN3, GPIO.OUT)
    GPIO.setup(IN4, GPIO.OUT)
    GPIO.setwarnings(False)  # Disabilita i warning sui pin GPIO già in uso

# Funzione per impostare i pin del motore
def set_step(w1, w2, w3, w4):
    GPIO.output(IN1, w1)
    GPIO.output(IN2, w2)
    GPIO.output(IN3, w3)
    GPIO.output(IN4, w4)

# Funzione per far ruotare il motore di un certo numero di passi
def step(steps, direction):
    if direction == 'clockwise':
        seq = SEQ
    elif direction == 'counterclockwise':
        seq = SEQ[::-1]  # Inverti la sequenza per il senso antiorario
    
    for i in range(steps):
        for j in range(8):
            set_step(*seq[j])
            sleep(0.005)  # Ritardo di 5 millisecondi tra ciascun passo

# Funzione per controllare il motore con input da tastiera
def control_motor(stdscr):
    stdscr.clear()
    stdscr.addstr("Usa le frecce direzionali per controllare il motore. Premi 'q' per uscire.\n")
    stdscr.refresh()
    setup()

    while True:
        key = stdscr.getch()
        if key == curses.KEY_UP:
            step(1, 'clockwise')
        elif key == curses.KEY_DOWN:
            step(1, 'counterclockwise')
        elif key == ord('q'):
            GPIO.cleanup()  # Pulizia dei pin GPIO prima di uscire
            break

# Eseguire il programma principale utilizzando curses
if __name__ == "__main__":
    curses.wrapper(control_motor)`

aaronwmorris, 2024-05-06:
Thanks for the code snippet.  I am close to being able to start working on this.

Mooflotic, 2024-05-06:
Take your time, I'm here to help you if you need me.

aaronwmorris, 2024-05-17:
Merged #1266 to support controlling a 28BYJ-48 stepper motor in indi-allsky.

I swapped the RPi.GPIO module for the Adafruit-Blinka module which is better maintained and supports a wide variety of systems, not just Raspi.

When you configure the focuser in the Config -> Devices view, an interface will appear in the Focus view were you can directly control the stepper motor to adjust focus.

aaronwmorris, 2024-06-08:
@Mooflotic Have you had a chance to test the focuser support?

Mooflotic, 2024-06-09:
@aaronwmorris yup, it works flawlessly. I very like the ability to rotate various degrees in one shot.

Mooflotic, 2024-06-17:
Do you want me to close this issue? I'm rather content.

aaronwmorris, 2024-06-17:
I'll go ahead and close it for now.  Thanks for the feedback!

closed 2024-06-17 by aaronwmorris