aaronwmorris/indi-allsky #1256: Keep DNG raw files that have been processed for a user defined period of time for manual post processing

andylsun, 2024-05-11:
Hi! Thanks for indi-allsky, been running it for about six months and it works really well.

I use a Raspberry Pi HQ camera in DNG mode with no AWB, but do use SCNR average neutral processing. No other stretching or other settings.

After last night's Aurora, I found that the Aurora was quite washed our colour wise and believe that might be SCNR processing? Is there a way to get the software to keep (not necessarily accessible via the web interface) the raw DNG prior to processing, so that they can be processed by hand as well? Say the last 'n' days worth?

As an aside to the SCNR processing, I'm going to disable it tonight and try a green balance factor of 0.5 and see how that works out tonight.

Thanks

Andy

aaronwmorris, 2024-05-11:
Awesome!

There is an option to save FITS files which is the RAW data that was in the DNG files.  Once you have FITS files, you can use the FITS processor under the Tools and manually try different processing options.

andylsun, 2024-05-11:
Oh cool that's great. I forgot to mention I'm using libcamera with my HQ camera not INDI if that makes a difference. Will enable it right now!

If you want me to close this issue I can do that.

aaronwmorris, 2024-05-11:
It will work fine for the libcamera type cameras.  You can keep this issue open and let me know how it works for you.

If you select the FITS processor view directly in Tools, it will just open the last FITS file saved.  You can also use the Image Viewer to select individual FITS files for processing (under the image).

andylsun, 2024-05-12:
It's working fine, gathering FITS images now and playing around with the config! Will close.

closed 2024-05-12 by andylsun